thesofproject/sof #2020: [BUG] logger does not compile with GCC 9.2.1

plbossart, 2019-10-28:
**Describe the bug**

````
/data/pbossart/SOF-public/sof/sof/tools/logger/convert.c: In function ‘serial_read’:
/data/pbossart/SOF-public/sof/sof/tools/logger/convert.c:311:2: error: converting a packed ‘struct log_entry_header’ pointer (alignment 1) to a ‘uint32_t’ {aka ‘unsigned int’} pointer (alignment 4) may result in an unaligned pointer value [-Werror=address-of-packed-member]
  311 |  for (len = 0, n = (uint32_t *)&dma_log; len < sizeof(dma_log); n++) {
      |  ^~~
````

**To Reproduce**
install Ubuntu 19.10

**Reproduction Rate**
100%

**Expected behavior**
no compilation issues

**Environment**
firmware aa7c73c15635eedb53403bee11357b0b856c6537

plbossart, 2019-10-28:
suggested diff to remove the error

````diff
diff --git a/tools/logger/convert.c b/tools/logger/convert.c
index fd2f25cf..649c3d8a 100644
--- a/tools/logger/convert.c
+++ b/tools/logger/convert.c
@@ -305,17 +305,17 @@ static int serial_read(const struct convert_config *config,
 {
        struct log_entry_header dma_log;
        size_t len;
-       uint32_t *n;
+       uint8_t *n;
        int ret;
 
-       for (len = 0, n = (uint32_t *)&dma_log; len < sizeof(dma_log); n++) {
-               ret = read(config->serial_fd, n, sizeof(*n));
+       for (len = 0, n = (uint8_t *)&dma_log; len < sizeof(dma_log); n+= sizeof(uint32_t)) {
+               ret = read(config->serial_fd, n, sizeof(*n) * sizeof(uint32_t));
                if (ret < 0)
                        return -errno;
 
                /* In the beginning we read 1 spurious byte */
-               if (ret < sizeof(*n))
-                       n--;
+               if (ret < sizeof(*n) * sizeof(uint32_t))
+                       n-= sizeof(uint32_t);
                else
                        len += ret;
        }
````

but was it really intentional to do such unaligned reads?

lgirdwood, 2019-11-04:
@plbossart I've got latest ubuntu, let me fix.

lgirdwood, 2019-11-07:
PR is https://github.com/thesofproject/sof/pull/2073

closed 2020-02-20 by plbossart