thesofproject/sof #2041: [BUG][ICL] Sporadic stream stall on multicore timer driven pipelines

zrombel, 2019-10-31:
**Description**
When running simultaneous capture and playback on timer driven pipelines, where pipelines are run on same dsp core different than 0, or each on a different core, pipelines stop copping data after trigger_start and stream stalls. 
Caught on audio formats:
- 8kHz 16/16b 8ch
- 48kHz 32/32b 4ch
- 48kHz 24/32b 4ch
But bug seems to be audio format agnostic.

**To Reproduce**
Run slimdriver python tests:
tests/sof/fw_17_multi_core/test_00_multi_core_same_kernel.py
tests/sof/fw_17_multi_core/test_01_multi_core_multi_kernel.py
--playback_iterations > 500

**Reproduction Rate**
<1%

**Environment**
1) FW pull request https://github.com/thesofproject/sof/pull/2031 b41dd49

zrombel, 2019-12-11:
I was able to reproduce this issue on master branch commit 65de27d2cbfe7bdaaf5c45da2677e7c933dcc294.
WHL after 636 playback iterations.
ICL after 921 playback iterations.

Logs:
[Multicore Stream Stall Logs.zip](https://github.com/thesofproject/sof/files/3949922/Multicore.Stream.Stall.Logs.zip)

lgirdwood, 2019-12-17:
@zrombel these logs are big, can you paste the relevant sections showing the failure. Thanks.

emilchudzik, 2020-02-24:
There is no repro of stream stall on multicore tests on top branch from commit: c35e1aa6.
Exectued tests: each audio format x1000 iteration -> ~48k of tests without stream stall during the weekend

closed 2020-02-24 by emilchudzik