thesofproject/sof #2067: Replace macros with functions wherever possible

paulstelian97, 2019-11-06:
There are plenty of hidden issues that we're only starting to uncover because of macros not doing any type checking and the errors showing up on non-default configs.

The suggestion is to replace all macros that can be replaced with functions. We could start with `spinlock_init` which is the situation from where the original comment comes from, but there are many others. Type checking them removes a class of possible issues.

I had the idea to do it myself however I'm not sure I'm going to get the required time on my hands soon enough so I'll leave it here in case someone else wants to try it.

_Suggested by @jajanusz in https://github.com/thesofproject/sof/pull/2016/files_

paulstelian97, 2019-11-21:
I have looked into the headers and I have found the following places have macros which may be worth replacing with inline functions:

* `<sof/spinlock.h>` -- I consider this one needs to be refactored more seriously. `spin_lock_irq` will need an interface change to return flags instead of passing as parameter, although I can accept having a wrapper macro for this purpose (if there is a good reason, such aligning with Linux interface).
* `<sof/math/numbers.h>` -- The macros MIN, MAX, ABS, SGN might benefit from turning into inline functions (but I'm posting this here to see if you share my opinion on this; I'm not really pushing this one too hard)

paulstelian97, 2019-11-21:
Will bring a pull request with the changes in `<sof/spinlock.h>` soon.

paulstelian97, 2019-11-21:
To those interested, see #2122 (it's in draft for now because there are certain things that may need to be redone), for `<sof/spinlock.h>`. For the other one I'm waiting for feedback on whether they should change (it should be a much simple change though)

paulstelian97, 2019-12-13:
#2122 is merged now. @dbaluta and everyone else: Should I also consider the macros in `<sof/math/numbers.h>` for turning into inline functions or not?

jajanusz, 2019-12-13:
Macros in numbers.h are different, they are more like templates from C++. you have MIN for float,double,int etc, if you'd like to turn them into function in C and keep the same functionality you'd have to make it for every number type.

paulstelian97, 2019-12-13:
@jajanusz Weren't all uses in the SOF code for int?

I guess I'll close this then.

closed 2019-12-13 by paulstelian97

jajanusz, 2019-12-13:
> @jajanusz Weren't all uses in the SOF code for int of various sizes?
> 
> I guess I'll close this then.

I guess most uses now will be for ints 8-32 signed/unsiged, what already would give you ~6 almost copy-pasted functions, there are is float support incoming.