thesofproject/sof #2075: [Bug][CML] ipc timeout when aplay via speaker pipeline on Chromebook

aiChaoSONG, 2019-11-08:
**Bug Description**

***sof-dev + cml-008-drop-stable:***
Everytime do playback via speaker pipeline (hw:0,5), there will be io error in terminal, check dmesg, you will see ipc timeout. this seems to be the rt1011 machine driver is not ready to work with sof.
this phenomenon is observed both on ChromeOS and Ubuntu1804.

***keyword_dectect + cml-007-drop-stable:***
Do playback via speaker pipeline (hw:0,5) on Ubuntu1804, there is no sound output, and there is no error in dmesg.  there is sound output on ChromeOS.

**To Reproduce**
run `aplay -Dhw:0,5 -f dat -i -vvv -c 2 ./audio.wav`

**Reproduction Rate**
100%

**Environment**
On sof-dev + cml-008-drop-stable branch:
1. Kernel: 872e5e80
2. SOF: f7dfd05b

On keyword_dectect + cml-007-drop-stable branch
1. kernel: 169c6594
2. SOF: f5dff2940

Tplg: sof-cml-rt1011-rt5682
Platform: CML Chromebook 
OS: ChromeOS & Ubuntu 1804

[helios-speaker-sof-logs.txt](https://github.com/thesofproject/linux/files/3822690/helios-speaker-sof-logs.txt)
[hatch-speaker-sof-log.txt](https://github.com/thesofproject/linux/files/3822691/hatch-speaker-sof-log.txt)

wwittbrx, 2019-11-08:
I cannot reproduce this issue on Ubuntu Hatch with https://github.com/thesofproject/sof/commit/f7dfd05b9fa4f92264905a47c0d9b155a9ce6602, https://github.com/thesofproject/linux/commit/872e5e80c67a8805694529918621260bc506e0f5 and sof-cml-rt5682-max98357a (this is the topology we should use for Hatch, sof-cml-rt1011-rt5682 is for Helios).

Also, how did you use sof-dev branch for ChromeOS? It doesn't even have chromeos folder with configs. Cml-008-drop-stable branch was released for ChromeOS with https://github.com/thesofproject/linux/tree/chrome-4.19-backport and we should use it for ChromeOS tests.

If you have custom build kernel from sof-dev for ChromeOS, this was never supported or tested before, and shouldn't be used for bug reports.

aiChaoSONG, 2019-11-11:
@wwittbrx  I see this issue **mainly on Helios**, please test with helios:
for 007 release on Ubuntu, please retry with keyword_dectect + cml-007-drop-stable branch, and aplay via speaker(hw:0,5), see if there is sound output. 

for 008 release on Chrome , I use chrome-4.19-backport + cml-008-drop-stable, and see ipc timeout if run aplay via speaker pipeline(hw:0,5)

for 008 release on Ubuntu, I use topic/sof-dev + cml-008-drop-stable, and see ipc timeout if run aplay via speaker pipeline(hw:0,5).  and this issue can also be reproduced  with topic/sof-dev + master 
Sorry for didn't make this clear, after you tried, please leave a comment.

lgirdwood, 2019-11-11:
@aiChaoSONG what happens if KWD is disabled ? Does this only occur if KWD detect is either in topology or enabled ?

aiChaoSONG, 2019-11-12:
@lgirdwood I think this has nothing to do with KWD, Even if I disable KWD pipeline, I still got this error. this error also exist on sof-dev + master on Helios.

keyonjie, 2019-11-14:
Looks the dw-dma channel enable fails, see this:
"Channel enable = 0."

··· 
    0      2          DAI 7.38    182357041.875000        13.906250      src/audio/dai.c:494 	dai_comp_trigger(), command = 1
    0      2          DAI 7.38    182357054.791667        12.916667      src/audio/dai.c:505 	dai_comp_trigger(), START
    0      2          SSP         182357252.968750       198.177078 ers/intel/cavs/ssp.c:877 	ssp_trigger() cmd 1
    0      2          SSP         182357267.291667        14.322917 ers/intel/cavs/ssp.c:806 	ssp_start()
    0      2          DMA         182369944.947917     12677.656250 src/drivers/dw/dma.c:1044	dw_dma_free_data_size() size is 0! Channel enable = 0.
    0      2       BUFFER         182369967.135417        22.187500   src/audio/buffer.c:126 	comp_update_buffer_produce(), no bytes to produce, source->comp.id = 35, source->comp.type = 5, sink->comp.id = 38, sink->comp.type = 2
···

sathyap-chrome, 2019-11-19:
@keyonjie @aiChaoSONG  Please help prioritize the bug. our next drop to Google is blocked due to this issue.

sathyap-chrome, 2019-11-19:
@mengdonglin Can we make this a P1 issue please

aiChaoSONG, 2019-11-19:
didn't reproduce speaker issue on Chrome OS with 007 release, later speaker issue track will continue after 008 integration.

closed 2019-11-19 by aiChaoSONG

sathyap-chrome, 2019-11-19:
hi @aiChaoSONG  we need the speaker playback to work before integrating 008.
I think we need to reopen

aiChaoSONG, 2019-11-19:
@sathyap-chrome This speaker pipeline timeout issue exist on 008 release and daily but not 007 release, now we observe and try to fix this issue on daily #2107.

sathyap-chrome, 2019-11-19:
@aiChaoSONG  Thanks. We can track the issue in #2107