thesofproject/sof #2077: [BUG][TGL]Topology without support from master branch for TGL

Bin-QA, 2019-11-08:
**Describe the bug**
Update firmware & tplg file kernel report `error: tplg component load failed -22`

**To Reproduce**
copy sof-hda-generic-4ch.tplg file to /lib/firmware/intel/sof-tplg/
copy sof-tgl.ri to /lib/firmware/intel/sof
copy sof-tgl.ldc & sof-tgl.lst to /etc/sof
After setup firmeware & kernel reboot system

**Reproduction Rate**
100%

**Expected behavior**
kernel driver load tplg fiel success

**Impact**
Driver load process

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: intel-next tag: intel-5.4-rc5-2019-11-01 commit: 7b84827404
    * SOF: master 0b7fd03
2) Name of the topology file
    * Topology: sof-hda-generic.m4
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL HDA ALC711

** support tplg file version:
branch: tgl-002-drop-stable
commit: c8bc4b2

**Screenshots or console output**
```
[    8.479625] sof-audio-pci 0000:00:1f.3: loading topology:intel/sof-tplg/sof-hda-generic-4ch.tplg
[    8.480212] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:11:0
[    8.480216] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 0 pipe 0 type 7 name : iDisp3 Tx stream none
[    8.480217] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    8.480218] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 1 pipe 1 type 23 name : PCM0P stream Passthrough Playback 0
[    8.480220] sof-audio-pci 0000:00:1f.3: loaded host PCM0P
[    8.480220] sof-audio-pci 0000:00:1f.3:  config: periods snk 2 src 0 fmt 0
[    8.480225] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    8.480318] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    8.480320] sof-audio-pci 0000:00:1f.3: tplg: load control type 1 name : 1 Master Playback Volume
[    8.480323] sof-audio-pci 0000:00:1f.3: tplg: load kcontrol index 2 chans 2
[    8.480324] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 2 pipe 1 type 6 name : PGA1.0 stream none
[    8.480325] sof-audio-pci 0000:00:1f.3:  config: periods snk 2 src 2 fmt 1
[    8.480328] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    8.480393] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    8.480395] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 3 pipe 1 type 31 name : BUF1.0 stream none
[    8.480397] sof-audio-pci 0000:00:1f.3: buffer BUF1.0: size 768 caps 0x71
[    8.480399] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    8.480455] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    8.480457] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 4 pipe 1 type 31 name : BUF1.1 stream none
[    8.480457] sof-audio-pci 0000:00:1f.3: buffer BUF1.1: size 768 caps 0x71
[    8.480459] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    8.480529] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    8.480530] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 5 pipe 1 type 27 name : HDA0.OUT stream Analog Playback and Capture
[    8.480532] sof-audio-pci 0000:00:1f.3: dai HDA0.OUT: type 3 index 0
[    8.480532] sof-audio-pci 0000:00:1f.3:  config: periods snk 0 src 2 fmt 2
[    8.480535] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    8.480621] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    8.480622] sof-audio-pci 0000:00:1f.3: tplg: connected widget HDA0.OUT -> DAI link Analog Playback and Capture
[    8.480623] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 6 pipe 1 type 32 name : PIPELINE.1.HDA0.OUT stream HDA0.OUT
[    8.480625] sof-audio-pci 0000:00:1f.3: tplg: pipeline id 1 comp 6 scheduling comp id 5
[    8.480626] sof-audio-pci 0000:00:1f.3: pipeline PIPELINE.1.HDA0.OUT: period 1000 pri 0 mips 5000 core 0 frames 0
[    8.480629] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[    8.480681] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[    8.480683] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[    8.480684] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[    8.480729] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[... 253 lines cut ...]
[    8.483886] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    8.483891] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    8.483895] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    8.483896] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    8.483901] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    8.483905] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    8.483906] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    8.483911] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    8.483914] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    8.483916] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    8.483920] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    8.483924] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    8.483926] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    8.483927] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
[    8.483929] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[    8.483930] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[    8.483943] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -22
[    8.483956] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22
[   16.410328] systemd-udevd (278) used greatest stack depth: 12528 bytes left
[   16.411712] systemd-udevd (296) used greatest stack depth: 11432 bytes left
[   28.701693] systemd-journald[260]: File /var/log/journal/c4515fabaeee4c0cb289119c9fd2ec29/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   29.945409] random: crng init done
[   29.945410] random: 7 urandom warning(s) missed due to ratelimiting
[   37.913963] rfkill: input handler disabled
```

plbossart, 2019-11-08:
Wondering how a generic topology could fail here:

[    8.483852] sof-audio-pci 0000:00:1f.3: error: invalid kcontrol count 2 for volume
[    8.483853] sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 6 name : Dmic0 stream none reply 0
[    8.483854] sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget Dmic0

xiulipan, 2019-11-11:
@plbossart @Bin-QA 
check the md5sum of our binary and find the change happen on OCT.24 from https://github.com/thesofproject/sof/pull/1867 and https://github.com/thesofproject/linux/pull/1251

This change https://github.com/thesofproject/linux/pull/1251/commits/40bcfd01d72906c491f52db49ce7512806aa0ddb should have ABI change, but not.

@ClarexZhou Could you check the Linux without https://github.com/thesofproject/linux/pull/1251 can work with master FW and TPLG on other HDA based board.

dengyangchao, 2019-11-11:
Same issue occurred when use sof-dev:7b7f94cc(Oct 7th) and Master fb068e6a (Nov 8th ) on CFL-S RVP with onboard  codec ALC700 in HDA mode.
TPLG: sof-hda-generic-4ch.tplg

```
[   10.150704] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 61 pipe 10 type 6 name : Dmic0 stream none
[   10.150706] sof-audio-pci 0000:00:1f.3: error: invalid kcontrol count 2 for volume
[   10.150793] sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 6 name : Dmic0 stream none reply 0
[   10.150890] sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget Dmic0
[   10.150993] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[   10.151003] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[   10.151008] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[   10.151023] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151032] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151037] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[   10.151049] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151057] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151062] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[   10.151098] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151110] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151115] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[   10.151130] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151139] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151144] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[   10.151167] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151176] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151180] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[   10.151192] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151201] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   10.151206] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[   10.151212] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
[   10.151298] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[   10.151367] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
```

[dmesg.log](https://github.com/thesofproject/sof/files/3829868/dmesg.log)

juimonen, 2019-11-11:
so where is the kernel coming from? I can't find those hash and tag ?

juimonen, 2019-11-11:
So this is because dmic topology has 2 controls (volume and mute switch) and we have in kernel before a68c6b6cc77b841dc37c17a5d9a7074e26801af5 test that there can be only 1 control. If this is to be used with older kernel (?) we can create topology with only 1 switch?

plbossart, 2019-11-11:
> @plbossart @Bin-QA
> check the md5sum of our binary and find the change happen on OCT.24 from #1867 and [thesofproject/linux#1251](https://github.com/thesofproject/linux/pull/1251)
> 
> This change [thesofproject/linux@40bcfd0](https://github.com/thesofproject/linux/commit/40bcfd01d72906c491f52db49ce7512806aa0ddb) should have ABI change, but not.
> 
> @ClarexZhou Could you check the Linux without [thesofproject/linux#1251](https://github.com/thesofproject/linux/pull/1251) can work with master FW and TPLG on other HDA based board.

I don't understand a word of all this.
The kernel was changed 1.5 month ago to support 2 controls for volume,  you must have a configuration issue or a bad environment. Things work fine for me on HDaudio+dmic platforms.

dengyangchao, 2019-12-17:
As branch intel-next updated to 5.5 rc1,  issue can't reproduce, close.

Recipe:
kernel:  intel next  tag: intel-5.5-rc1-2019-12-09 
SOF: master commit: 98d7ab21

closed 2019-12-17 by dengyangchao