thesofproject/sof #2081: [BUG][CML-007] ipc error in pause-release stress test on Chrome OS

aiChaoSONG, 2019-11-11:
**Describe the bug**
run pause/release stress test on Chrome OS, first you should see i/o error in the terminal, check dmesg, you will see ipc error.

**To Reproduce**
run pause-release script: `bash ./sof-test-pause-release.sh -t pcm2p -r 48000 -f s16_le -c2 -d 5000 -i 0.5 ./llwav.wav`

**Reproduction Rate**
Headset pipeline(hw:0,0): 
> 1st try: ipc error at 70th  with -i 0.5
> 2nd try: ipc error at 300th with -i 0.5
> 3rd try: ipc error at  700th with -i 2

HDMI pipeline(hw:0,2)
> 1st try: ipc error at 800th with -i 2
> 2nd try: ipc error at 1888th with -i 0.5
> 3rd try: ipc error at 3605th with -i 0.2

**Expected behavior**
no error in pause-release

**Environment**
Kernel: Chrome OS 4.19.79-07473-g5a6eeaee512fe
SOF: cml-007-drop-stable
tplg: sof-cml-rt1011-rt5682.tplg

all above are from ChromeOS image 12607.8.0   v79

[dmesg_pause_release_ipc_error.log](https://github.com/thesofproject/sof/files/3829974/dmesg_pause_release_ipc_error.log)
[pause_release_ipc_error_hdmi.log](https://github.com/thesofproject/sof/files/3829975/pause_release_ipc_error_hdmi.log)

lgirdwood, 2019-11-11:
@aiChaoSONG looks like a pause during an xrun.
```
    0      2          DAI 4.20    652756126.614583         5.000000      src/audio/dai.c:483 	dai_comp_trigger(), command = 3
    0      2          DMA         652756132.135417         5.520833 intel/cavs/hda-dma.c:569 	hda-dmac: 7 channel 0 -> release
    0      2          DMA         652756137.239583         5.104167 intel/cavs/hda-dma.c:531 	hda-dmac: 7 channel 0 -> start
    0      2          DMA         652756141.875000         4.635417 intel/cavs/hda-dma.c:353 	hda-dmac: 7 channel 0 -> enable
    0      2          IPC         653395641.093750    639499.250000    src/ipc/handler.c:450 	ipc: comp 16 -> trigger cmd 0x60000
    0      2         PIPE 4.21    653395652.083333        10.989583 src/audio/pipeline.c:608 	pipeline_trigger()
    0      2         HOST         653395657.031250         4.947917     src/audio/host.c:291 	host_trigger()
    0      2       VOLUME         653395662.447917         5.416667 udio/volume/volume.c:533 	volume_trigger()
    0      2          DAI 4.20    653395667.500000         5.052083      src/audio/dai.c:483 	dai_comp_trigger(), command = 2
    0      2          DAI 4.20    653395672.187500         4.687500      src/audio/dai.c:547 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         653395677.031250         4.843750 intel/cavs/hda-dma.c:626 	hda-dmac: 7 channel 0 -> stop
    0      2          IPC         654006664.739583    610987.687500    src/ipc/handler.c:450 	ipc: comp 16 -> trigger cmd 0x70000
    0      2         PIPE 4.21    654006675.625000        10.885417 src/audio/pipeline.c:608 	pipeline_trigger()
    0      2         HOST         654006680.625000         5.000000     src/audio/host.c:291 	host_trigger()
    0      2       VOLUME         654006685.989583         5.364583 udio/volume/volume.c:533 	volume_trigger()
    0      2          DAI 4.20    654006690.989583         5.000000      src/audio/dai.c:483 	dai_comp_trigger(), command = 3
    0      2          DMA         654006696.302083         5.312500 intel/cavs/hda-dma.c:569 	hda-dmac: 7 channel 0 -> release
    0      2          DMA         654006701.562500         5.260417 intel/cavs/hda-dma.c:531 	hda-dmac: 7 channel 0 -> start
    0      2          DMA         654006706.197917         4.635417 intel/cavs/hda-dma.c:353 	hda-dmac: 7 channel 0 -> enable
    0      2       BUFFER         654006801.041667        94.843750   src/audio/buffer.c:146 	comp_update_buffer_produce(), no bytes to produce, source->comp.id = 16, source->comp.type = 1, sink->comp.id = 17, sink->comp.type = 5
    0      1          DAI 4.20    654008040.312500      1239.270874      src/audio/dai.c:575 	dai_check_for_xrun() error: underrun due to no data available
    0      1         COMP         654008047.343750         7.031250 of/audio/component.h:743 	comp_underrun() error: dev->comp.id = 20, source->avail = 768, copy_bytes = 0
    0      2         PIPE 4.21    654008054.687500         7.343750 src/audio/pipeline.c:608 	pipeline_trigger()
    0      2         HOST         654008059.635417         4.947917     src/audio/host.c:291 	host_trigger()
    0      2          DMA         654008064.739583         5.104167 intel/cavs/hda-dma.c:626 	hda-dmac: 5 channel 0 -> stop
    0      2       VOLUME         654008069.947917         5.208333 udio/volume/volume.c:533 	volume_trigger()
    0      2          DAI 4.20    654008075.104167         5.156250      src/audio/dai.c:483 	dai_comp_trigger(), command = 8
    0      2          DAI 4.20    654008079.843750         4.739583      src/audio/dai.c:541 	dai_comp_trigger(), XRUN
    0      2          DAI 4.20    654008084.062500         4.218750      src/audio/dai.c:547 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         654008088.697917         4.635417 intel/cavs/hda-dma.c:626 	hda-dmac: 7 channel 0 -> stop
    0      1         PIPE         654008097.343750         8.645833 src/audio/pipeline.c:764 	pipeline_copy() error: ret = -61
    0      1         PIPE 4.21    654008103.854167         6.510417 src/audio/pipeline.c:977 	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      2          IPC         654767715.937500    759612.062500    src/ipc/handler.c:450 	ipc: comp 16 -> trigger cmd 0x60000
    0      2         PIPE 4.21    654767727.135417        11.197917 src/audio/pipeline.c:608 	pipeline_trigger()
    0      2         HOST         654767732.500000         5.364583     src/audio/host.c:291 	host_trigger()
    0      1         COMP         654767737.395833         4.895833 rc/audio/component.c:179 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_PAUSE
    0      1          IPC         654767744.166667         6.770833 src/audio/pipeline.c:635 	pipeline_trigger() error: ret = -22, host->comp.id = 16, cmd = 2
    0      1          IPC         654767751.458333         7.291667    src/ipc/handler.c:484 	ipc: comp 16 trigger 0x60000 failed -22
    0      2          IPC         654769341.145833      1589.687500    src/ipc/handler.c:450 	ipc: comp 16 -> trigger cmd 0x70000
    0      2         PIPE 4.21    654769351.927083        10.781250 src/audio/pipeline.c:608 	pipeline_trigger()
    0      2         HOST         654769356.822917         4.895833     src/audio/host.c:291 	host_trigger()
    0      1         COMP         654769361.458333         4.635417 rc/audio/component.c:153 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_RELEASE
    0      1          IPC         654769368.125000         6.666667 src/audio/pipeline.c:635 	pipeline_trigger() error: ret = -22, host->comp.id = 16, cmd = 3
    0      1          IPC         654769375.520833         7.395833    src/ipc/handler.c:484 	ipc: comp 16 trigger 0x70000 failed -22
    0      2          IPC         655405063.802083    635688.250000    src/ipc/handler.c:450 	ipc: comp 16 -> trigger cmd 0x50000
```

ranj063, 2019-11-11:
@aiChaoSONG it is likely that this chromeos branch does not have the recent fixes for pause/release fixes we made recently: fcb9d23 and fd274c2. Can you please check?

aiChaoSONG, 2019-11-12:
@ranj063 the Chrome OS kernel repo(v4.19) has these two commits

wwittbrx, 2019-11-12:
Where did the topology come from for this issue? On branch cml-007-drop-stable the sof-cml-rt1011-rt5682.tplg does not exist. I understand that it was installed with ChromeOS image, but we need specific commit for debugging. Since there were many changes to topology since sof-cml-007-drop-stable, we need to check if the provided .tplg is compatible with this older FW.

aiChaoSONG, 2019-11-13:
@wwittbrx for cml-007-drop-stable, the firmware is from chromeos, and the commit of chromeos kernel is `5a6eeaee52fe`, if you are asking the fw commit, I think it may not be possible to find it, because chrome team get fw from us. is that right? @mengdonglin

mengdonglin, 2019-11-19:
@wwittbrx @aiChaoSONG 
@sathyap-chrome confirmed https://github.com/thesofproject/sof/pull/1948 contains the topology m4 for cml-007-drop-stable.  And https://github.com/thesofproject/sof/pull/1774 is the upstream version for master branch.

wwittbrx, 2019-11-21:
I cannot reproduce this issue with the following recipe (cml-008-drop):

PLATFORM: CML HATCH
OS: UBUNTU 19.04
FW: f7dfd05b9fa4 sof-cml-rt5682-max98357a.m4
KERNEL: 5d67c4b + PR1383 + 169c659

Could you check if this is true on your side as well? I would strongly recommend switching to CML 008 as it is much more stable, with multiple fixes.

aiChaoSONG, 2019-11-25:
yes, 008 is much stable, but google insist on doing release with 007, could you please try to test on Chrome, the 007 version is shipped with google image.

aiChaoSONG, 2019-11-28:
we upgrade only the  cml-008-drop-stable branch FW on Chrome OS and test, this issue also seen.
Recipe:
Kernel: ChromeOS R79 12607.28.0, commit id: `a5e05a63`
FW: cml-008-drop-stable (from 008 release)
Tplg: 007 version, shipped with ChromeOS

aiChaoSONG, 2019-12-03:
After picking Kernel SOF [PR#1277](https://github.com/thesofproject/linux/pull/1277) to Chrome Kernel, issue cannot reproduced. Close this.

closed 2019-12-03 by mengdonglin