thesofproject/sof #2082: [BUG][CML-007] KWD trigger failure on ChromeOS

aiChaoSONG, 2019-11-11:
**Describe the bug**
1. dmesg shows "failed to trigger DETECT9.0", even KWD is triggered and recording normally.
2. dmesg shows "failed to trigger DETECT9.0", but KWD is surely triggered, but after recording about 0.2s, recording is blocked.

**To Reproduce**
run `arecord -Dhw:0,8 -M -N -f s16_le -r 16000 -c2 -i -d5 -vvv  --period-size=400 wov.wav`

**Reproduction Rate**
can be reporduced with in 10 trys.

**Expected behavior**
KWD trigger successfuly and start to recording

**Environment**
Environment
Kernel: Chrome OS 4.19.79-07473-g5a6eeaee512fe
SOF: cml-007-drop-stable
tplg: sof-cml-rt1011-rt5682.tplg

all above are from ChromeOS image 12607.8.0 v79

[dmesg_kwd_trigger_failure.log](https://github.com/thesofproject/sof/files/3830720/dmesg_kwd_trigger_failure.log)
[sof_trace_kwd.log](https://github.com/thesofproject/sof/files/3830726/sof_trace_kwd.log)

lgirdwood, 2019-11-11:
@mrajwa I guess this is related to drain and needs your drain updates ?

mrajwa, 2019-11-12:
@lgirdwood, Yes could be, he uses old FW branch + new topology, so arecord hang due to overrun can happen. I will make it uniform soon.

@aiChaoSONG, please PM me with your test machine credentials I will use it as a test unit.

aiChaoSONG, 2019-11-13:
@mrajwa we are testing google pv release and see if there is any critical issue, and fw, tplg and kernel are all from chromeos

wwittbrx, 2019-11-21:
I cannot reproduce this issue with the following recipe (cml-008-drop):

PLATFORM: CML HATCH
OS: UBUNTU 19.04
FW: f7dfd05b9fa4 sof-cml-rt5682-max98357a.m4
KERNEL: 5d67c4b + PR1383 + 169c659

Could you check if this is true on your side as well? I would strongly recommend switching to CML 008 as it is much more stable, with multiple fixes.

mengdonglin, 2020-01-07:
Close this issue since KWD is working well with cml-008-drop. So we don't need to fix this in cml-007.

closed 2020-01-07 by mengdonglin