thesofproject/sof #2084: [BUG][TGL] SSP clocks generated are not as per topology configured

sathya-nujella, 2019-11-12:
**Describe the bug**
1) SSP clocks generated are not correct and not matching with the values set in topology.
2) In topology Backend setting is,
#SSP 1 (ID: 0)
DAI_CONFIG(SSP, 1, 0, SSP1-Codec,
	SSP_CONFIG(I2S, SSP_CLOCK(mclk, 19200000, codec_mclk_in),
		      SSP_CLOCK(bclk, 2304000, codec_slave),
		      SSP_CLOCK(fsync, 48000, codec_slave),
		      SSP_TDM(2, 24, 3, 3),
		      SSP_CONFIG_DATA(SSP, 1, 24)))
#SSP 0 (ID: 1)
DAI_CONFIG(SSP, 0, 1, SSP0-Codec,
        SSP_CONFIG(I2S, SSP_CLOCK(mclk, 19200000, codec_mclk_in),
                      SSP_CLOCK(bclk, 2400000, codec_slave),
                      SSP_CLOCK(fsync, 48000, codec_slave),
                      SSP_TDM(2, 25, 3, 3),
                      SSP_CONFIG_DATA(SSP, 0, 24)))
3) SSP clocks observed are:
SSP1:  BCLK=2.3MHz, LRCLK=48Khz  (this is correct)
SSP0:  BCLK=1.5Mhz, LRCLK=30KHz, MCLK=19.2Mhz (because of BCLK is wrong, playback sound is bad)

**To Reproduce**

**Reproduction Rate**
100%

**Expected behavior**
SSP clocks should be right

**Impact**
Playback sound is not good

**Environment**
TGL.
Firmware is: latest from master. Fw built with: 6984e86b1561c8bdad26463753fe87f5181aae6c

jairaj-arava, 2019-11-12:
Hi,
We have added the MCLK id 0 in both SSP configs as below, but still we see the clock issue.(BCLK=1.5MHz, LRCLK = 48KHz, MCLK = 19.2MHz)

#SSP 1 (ID: 0)
DAI_CONFIG(SSP, 1, 0, SSP1-Codec,
        SSP_CONFIG(I2S, SSP_CLOCK(mclk, 19200000, codec_mclk_in),
                      SSP_CLOCK(bclk, 2304000, codec_slave),
                      SSP_CLOCK(fsync, 48000, codec_slave),
                      SSP_TDM(2, 24, 3, 3),
                      **SSP_CONFIG_DATA(SSP, 1, 24,0)))**
#SSP 0 (ID: 1)
DAI_CONFIG(SSP, 0, 1, SSP0-Codec,
        SSP_CONFIG(I2S, SSP_CLOCK(mclk, 19200000, codec_mclk_in),
                      SSP_CLOCK(bclk, 2400000, codec_slave),
                      SSP_CLOCK(fsync, 48000, codec_slave),
                      SSP_TDM(2, 25, 3, 3),
                      **SSP_CONFIG_DATA(SSP, 0, 24,0)))**

lgirdwood, 2019-11-12:
@sathya-nujella @jairaj-arava anything in the trace logs that would indicate a problem within the SSP driver ?

jairaj-arava, 2019-11-12:
Attached the logger trace that has no changes in ssp driver:

[logs.txt](https://github.com/thesofproject/sof/files/3838587/logs.txt)

plbossart, 2019-11-12:
@jairaj-arava thanks, but where is the code that adds those traces? it's hard to follow a log without the matching line numbers.

ranj063, 2019-11-12:
@plbossart please give me 5 mins. we're working on getting the logs with hex values instead of uints. I will post the diff along with it.

ranj063, 2019-11-12:
[logs_2.txt](https://github.com/thesofproject/sof/files/3838722/logs_2.txt)
diff ```\diff --git a/src/drivers/intel/cavs/ssp.c b/src/drivers/intel/cavs/ssp.c
index 59e47531..a057f71e 100644
--- a/src/drivers/intel/cavs/ssp.c
+++ b/src/drivers/intel/cavs/ssp.c
@@ -389,6 +389,8 @@ static int ssp_set_config(struct dai *dai,
        if (clk_index >= 0) {
                mdivc |= MCDSS(ssp_freq[clk_index].enc);
                mdivr_val = ssp_freq[clk_index].freq / config->ssp.mclk_rate;
+               trace_ssp("ssp_set_config() clk_index %d MCDSS: 0x%x mdivr_val: %u\n",
+                         clk_index, MCDSS(ssp_freq[clk_index].enc), mdivr_val);
        } else {
                trace_ssp_error("ssp_set_config() error: MCLK %d",
                                config->ssp.mclk_rate);
@@ -416,6 +418,10 @@ static int ssp_set_config(struct dai *dai,
                 */
                if (ssp_freq[clk_index].enc != CLOCK_SSP_XTAL_OSCILLATOR)
                        sscr0 |= SSCR0_ECS;
+
+               trace_ssp("ssp_set_config() clk_index: %d MNDSS: 0x%x mdiv: %u sscr0 0x%x\n",
+                         clk_index, MNDSS(ssp_freq[clk_index].enc), mdiv, sscr0);
+
        } else {
                /* check if we can get target BCLK with M/N */
                for (i = 0; i <= MAX_SSP_FREQ_INDEX; i++) {
@@ -439,6 +445,11 @@ static int ssp_set_config(struct dai *dai,
 
                /* M/N requires external clock to be selected */
                sscr0 |= SSCR0_ECS;
+
+               trace_ssp("ssp_set_config() clk_index: %d MNDSS: 0x%x",
+                         clk_index, MNDSS(ssp_freq[clk_index].enc));
+               trace_ssp("ssp_set_config() mdiv: %u sscr0 0x%x mdivc %u\n",
+                         mdiv, sscr0, mdivc);
        }
 
        switch (mdivr_val) {
```

ranj063, 2019-11-12:
Here's the specific part in the logs that I feel might be the reason for the issue where the clock indices
chosen are different for MCLK and BCLK.
```
    0      2          SSP          92714493.541667        15.729167 ers/intel/cavs/ssp.c:393 	ssp_set_config() **clk_index 1 MCDSS: 0x0** mdivr_val: 2

    0      2          SSP          92714508.750000        15.208333 ers/intel/cavs/ssp.c:442 	ssp_set_config(), M = 15, N = 16
    0      2          SSP          92714521.875000        13.125000 ers/intel/cavs/ssp.c:450 	ssp_set_config() **clk_index: 0 MNDSS: 0x100000**
    0      2          SSP          92714534.895833        13.020833 ers/intel/cavs/ssp.c:452 	ssp_set_config() mdiv: 10 sscr0 0xc0c00070 mdivc 1048577
```

plbossart, 2019-11-13:
MDNSS is not an actual register, you'd need to trace mdivc in all cases

ranj063, 2019-11-13:
> MDNSS is not an actual register, you'd need to trace mdivc in all cases

@plbossart yes but MNDSS and MCDSS values printed are what are OR'ed into mdivc. Thats why I printed those individually.

plbossart, 2019-11-13:
mdivc is THE shared register so we need to trace its actual value, not nominal ones.

jairaj-arava, 2019-11-13:
@plbossart 
Attached are the logs that  has traces of MDIVC.
[logs_3.txt](https://github.com/thesofproject/sof/files/3838944/logs_3.txt)

plbossart, 2019-11-14:
@ranj063 and I talked and I think the topology file is wrong.
Since we have access to a 24.576 MHz we need to use it without M/N and use either a 1.536 MHz (16*2*48000) or 3.072 MHz (32*2*48000).

The value SSP_CLOCK(bclk, 2304000, codec_slave) was copy/paste from CML I guess, but that's not what we used on GLK.

In short, let's not use the M/N dividers since we have no reason to use them.

ranj063, 2019-11-15:
@plbossart Unfortunately we end up with the same problem even when we use 16-bits and BCLK of 1.536MHz. So even in this case, MCLK source will be 38.4MHz xtal and BCLK source will be 24.576MHz cardinal clock if we set the MCLK to be 19.2 MHz.

ranj063, 2019-11-15:
@plbossart I am wondering if instead of changing the logic in the SSP driver, we could just change the ssp freq table in https://github.com/thesofproject/sof/blob/f004d11f306e2d910e22e06953fc252c301161a9/src/platform/tigerlake/lib/clk.c#L23

to make the xtal the first item as it iscalled out as default anyway?

plbossart, 2019-11-15:
@ranj063 I don't get how you can have the 'same problem' since you are no longer using M/N.

Again the idea is 
for speaker, the MCLK is a don't care, BCLK is sourced from 24.576 and divided by 16
for headset, the MCLK is sourced from 38.4 and divided by 2, and the BCLK is sourced from 38.4 divided by 16

ranj063, 2019-11-15:
> @ranj063 I don't get how you can have the 'same problem' since you are no longer using M/N.
>
> Again the idea is
> for speaker, the MCLK is a don't care, BCLK is sourced from 24.576 and divided by 16 [...]

@plbossart what I meant by the same problem was that the clock sources chosen are exactly the same as before but without M/N this time.

plbossart, 2019-11-15:
what does 'same as before' mean? can you clarity the difference from the goal listed above?

ranj063, 2019-11-15:
> what does 'same as before' mean? can you clarity the difference from the goal listed above?

@plbossart it means, playback on the headset is still distorted and measured clocks are same as before:
SSP clocks observed are:
SSP1: BCLK=2.3MHz, LRCLK=48Khz (this is correct)
SSP0: BCLK=1.5Mhz, LRCLK=30KHz, MCLK=19.2Mhz (because of BCLK is wrong, playback sound is bad)

plbossart, 2019-11-15:
why is 2.3 correct? didn't we talk about using 1.536?

ranj063, 2019-11-15:
> why is 2.3 correct? didn't we talk about using 1.536?

@plbossart Sorry I never learn not to copy &paste. Yes, the SSP1 BCLK is 1.536 but so is the BCLK for SSP0 (for the headset).

plbossart, 2019-11-15:
So it looks like you are taking the BCLK from the PLL instead of the oscillator then, so is the ECS bit correct for SSP0?

ranj063, 2019-11-15:
> > why is 2.3 correct? didn't we talk about using 1.536?
> 
> @plbossart Sorry I never learn not to copy &paste. Yes, the SSP1 BCLK is 1.536 but so is the BCLK for SSP0 (for the headset).

@plbossart I'm still waiting for the logs from @jairaj-arava. Will confirm soon

ranj063, 2019-11-15:
[ssp0_logs3.txt](https://github.com/thesofproject/sof/files/3852958/ssp0_logs3.txt)
@plbossart here're the logs with the changed topology to use 1.536MHz bclk for the speaker

sathya-nujella, 2019-11-25:
Hi @plbossart , @ranj063 , 
Currently we are using the firmware with the changes proposed in PR related to this. 
Please suggest and do the changes so that we can release official firmware image to customer. Thanks!

plbossart, 2019-11-25:
> Hi @plbossart , @ranj063 ,
> Currently we are using the firmware with the changes proposed in PR related to this.
> Please suggest and do the changes so that we can release official firmware image to customer. Thanks!

there's a need for more experiments on this one, the proposal from @ranj063 is far from optimal (uses M/N dividers even when it's not needed). we'll probably need the firmware folks to look into this.

sathya-nujella, 2019-12-04:
Hi @jajanusz 
Can you please update on this?

jajanusz, 2019-12-05:
@sathya-nujella I'm debugging this, I have to test it on ICL that has the same clocks to ensure whatever it's just TGL restriction or general requirement that accidentally occurred because of given frequencies.

lgirdwood, 2019-12-17:
@sathya-nujella @jajanusz any updates ?

jajanusz, 2019-12-18:
Waiting for info from HW team.
Basically for the same values of SSP & M/N registers it worked on ICL (that has the same clocks) and doesn't work on TGL, maybe there are some difference in HW impl (maybe due to introduction of ssp sync) and we need to do some extra steps..

jajanusz, 2019-12-23:
@lgirdwood @plbossart 
We missed change register desc. `ECS` bit is not officialy supported since cavs 2.0, looks like it worked on ICL as `undefined behavior `, on cavs2+ ECS has to be always 1.

lgirdwood, 2019-12-24:
@jajanusz I guess this means we need an update to ignore this then on 2.0+, Is there a PR?

jajanusz, 2020-01-02:
> @jajanusz I guess this means we need an update to ignore this then on 2.0+, Is there a PR?

There is no PR. I'll make it. I think current implementation is wrong even pre-ICL. If I understand correctly you can have only 1 master clock source for M/N for all SSPs BCLK, so we should track if there is any active SSP already using M/N as BCLK source, if so then we cannot look for  BCLK source, but just have to use what is already there and just tune M/N divider for that port (because that's the only thing that is set per-port).

ranj063, 2020-01-02:
> > @jajanusz I guess this means we need an update to ignore this then on 2.0+, Is there a PR?
> 
> There is no PR. I'll make it. I think current implementation is wrong even pre-ICL. If I understand correctly you can have only 1 master clock source for M/N for all SSPs BCLK (even if M/N is 1/1 and we just use something that is not XTAL), so we should track if there is any active SSP already using M/N as BCLK source, if so then we cannot look for BCLK source, but just have to use what is already there and just tune M/N divider for that port (because that's the only thing that is set per-port).

@jajanusz just wnated to clarify one thing. The TGL topology under consideration no longer requires the M/N divider for any of the BCLK's

plbossart, 2020-01-02:
@jajanusz are you saying that the attached clock tree is no longer accurate? If ECS has to be zero always, it means we cannot bypass the M/N dividers, and thus have to use the same source clock for all SSPs?

[APL-clocks.pdf](https://github.com/thesofproject/sof/files/4017015/APL-clocks.pdf)

jajanusz, 2020-01-03:
@plbossart Yes, I don't have pdf for ICL/TGL like you have for APL, but it's wrote in registers description for cavs 2.x that ECS=1, so you can pretty much only take bclk from m/n.

@ranj063 It doesn't matter if TPLG actually needs M/N, even if it doesn't the clock is going to go by M/N (cos ECS=1, just M/N is 1/1)

plbossart, 2020-01-03:
> @ranj063 It doesn't matter if TPLG actually needs M/N, even if it doesn't the clock is going to go by M/N (cos ECS=1, just M/N is 1/1)

Right, but in that case we would want to use 32-bit slots for both the codecs and the speaker amplifier, instead of 25 and 32 respectively as we do today, instead of using M/N to generate a bclock.

ranj063, 2020-01-06:
> There is no PR. I'll make it. I think current implementation is wrong even pre-ICL. If I understand correctly you can have only 1 master clock source for M/N for all SSPs BCLK (even if M/N is 1/1 [...]

@jajanusz I can update my PR to align with this direction but just a quick confirmation for CAVS 2.0+, ECS bit should always be set indicating that we always use M/N (even if it mean 1/1) and the clock source for BCLK for all SSP ports have to be common? Is this correct? Also, is there any restriction for the MCLK at all? Does it have to be a common clock source for all SSP's?

plbossart, 2020-01-06:
@ranj063 the MCLK is orthogonal to SSPs, there are 2 MCLKs but a single source (you can only change the dividers for each MCLK).

ranj063, 2020-01-06:
> @ranj063 the MCLK is orthogonal to SSPs, there are 2 MCLKs but a single source (you can only change the dividers for each MCLK).

@plbossart I'm a bit confused by "MCLK is orthogonal to SSPs". In the SSP driver, we configure the MCLK by choosing the MCLK source for every SSP config IPC and we could in theory end up choosing 2 different MCLK sources while configuring 2 SSP's, isnt it?

plbossart, 2020-01-06:
> @plbossart I'm a bit confused by "MCLK is orthogonal to SSPs". In the SSP driver, we configure the MCLK by choosing the MCLK source for every SSP config IPC and we could in theory end up choosing 2 [...]

no, that's not supported. You can only select one MCLK source. see the diagram above.

plbossart, 2020-01-06:
> no, that's not supported. You can only select one MCLK source. see the diagram above.

And we have 2 MCLKs even when we can have up to 6 SSPs, so there is no 1:1 mapping

ranj063, 2020-01-06:
> no, that's not supported. You can only select one MCLK source. see the diagram above.

hmm then is the logic in the SSP driver today correct? What happens during this step https://github.com/thesofproject/sof/blob/c68605403c52ac43ac1b97dec780b9407026ec7f/src/drivers/intel/cavs/ssp.c#L367?
The last SSP to be configured is the one that gets to choose which clock source to use for the MCLK?

plbossart, 2020-01-06:
yes, the MCLK programming is not really good. we should really have a clock driver, as 'documented' by the  /* TODO: move this into M/N driver */

ranj063, 2020-01-06:
@plbossart then does it make sense to fix this now along with the changes required for this bug? In which case, I'm afraid it might be beyond my expertise and probably for suited for @jajanusz to handle it?

jajanusz, 2020-01-07:
@plbossart @ranj063 Yes it needs bigger refactor. MCLK and BCLK are independent, but all BCLKs will share the same source clock and then you have SCR+M/N to change freq per SSP, all (2) MCLKs also share the same source and can have 2 dividers configured.

plbossart, 2020-01-07:
just to be clear about the 'same'
- all MCLKs share the same source
- all BCLKs share the same source
- the MCLK and BCLK source can be different

agree?

jajanusz, 2020-01-07:
> just to be clear about the 'same'
> 
> * all MCLKs share the same source

Yes
> * all BCLKs share the same source

Depends. It's M/N that can have only 1 source and all BCLKs share the same M/N source (with different dividers). So on ICL+ yes they share the same source, on previous platforms where you can set ECS=0, then you can have 2 sources M/N & raw XTAL.
> * the MCLK and BCLK source can be different

Yes

mengdonglin, 2020-01-13:
Fixes #2276 is available on https://github.com/thesofproject/sof/tree/tgl-003-drop-stable

PR #2276 (2020-01-21): drivers: intel: M/N initial driver and fixes
Initial MN driver version that should fix 2 major problems:
1) BCLK & MCLK logic was written like there were only 1 SSP port in existence, topologies with more SSPs in use could result in incorrect settings (config 2nd overriding source clock for config 1st etc).
2) ECS bit support was dropped in cAVS 2.0 and M/N is only supported source for BCLKs.
   
Fixes #2084
closed 2020-01-21 by jajanusz