thesofproject/sof #2090: [BUG][CML-007] sound channel crosstalk on ChromeOS

aiChaoSONG, 2019-11-13:
**Describe the bug**
you can hear very slight sound on one channel when speaker-test via another sound channel. 

**To Reproduce**
1. tune volume to max.
2. run `speaker-test -Dhw:0,0 -t wav -c 2  -s 1`, play on left channel, and  hear on right channel.
3. run `speaker-test -Dhw:0,0 -t wav -c 2  -s 2`, play on right channel, and  hear on left channel.

Expected behavior
sound should only  on the channel which we played on.

Environment
Kernel: Chrome OS 4.19.79-07473-g5a6eeaee512fe
SOF: cml-007-drop-stable
tplg: sof-cml-rt1011-rt5682.tplg

all above are from ChromeOS image 12607.8.0 v79

no error messages seen

aiChaoSONG, 2019-11-13:
this issue exist on 007 ubuntu: keyword_detect + cml-007-drop-stable

singalsu, 2019-11-13:
I don't remember the audio HW arch for this platform but if the speaker amplifier is connected via analog input some cross-talk is normal. E.g. -40 dB attenuation could be still normal. A near field stereo recording from both speakers would be useful (attach two identical microphones as close to each speaker as possible) and repeat the test.

aiChaoSONG, 2019-11-14:
As Seppo stated, if the speaker amplifier is connected via analog input some cross-talk is normal, and I run test on sof-dev + master, there is slight cross-talk, too. This may not be a Kernel or FW bug, so close it.

closed 2019-11-14 by aiChaoSONG