thesofproject/sof #2092: [BUG][CML-007] ipc error in dmic pause-release test

aiChaoSONG, 2019-11-13:
**Describe the bug**
capture via dmic, then press SPACE to pause-release capture very quickly, first you will see io error in the terminal, check dmesg, you'll see ipc error.

**To Reproduce**
1. capture via dmic: `arecord -Dhw:0,1 -f s32_le -r 48000 -c4 -i -vvv ./dmic.wav`
2. press SPACE very quickly

**Reproduction Rate**
reproduced within less than 20 quick press on SPACE

**Environment**
Kernel: Chrome OS 4.19.79-07473-g5a6eeaee512fe
SOF: cml-007-drop-stable
tplg: sof-cml-rt1011-rt5682.tplg

all above are from ChromeOS image 12607.8.0 v79

[sof_trace_dmic_pause_release.log](https://github.com/thesofproject/sof/files/3840060/sof_trace_dmic_pause_release.log)
[dmesg_dmic_pause_release.log](https://github.com/thesofproject/sof/files/3840061/dmesg_dmic_pause_release.log)

aiChaoSONG, 2019-11-13:
this issue is the same as #2044 , we may need cherry-pick the fix from PR2078 to cml-007-drop-stable.

mengdonglin, 2019-11-18:
@aiChaoSONG Please confirm if the fix for SOF #2044 can also fix this issue.

aiChaoSONG, 2019-11-18:
@mengdonglin the fix for #2044 cannot be applied to cml-007-drop-stable,  I think it is because the 007 branch is too old.

lgirdwood, 2019-11-18:
@lbetlej should'n the cml007 branch now all be merged into master ? if so, we can cut another release with this #2044 fix.

wwittbrx, 2019-11-21:
I cannot reproduce this issue with the following recipe (cml-008-drop):

PLATFORM: CML HATCH
OS: UBUNTU 19.04
FW: f7dfd05b9fa4 sof-cml-rt5682-max98357a.m4
KERNEL: 5d67c4b + PR1383 + 169c659

Could you check if this is true on your side as well? I would strongly recommend switching to CML 008 as it is much more stable, with multiple fixes.

aiChaoSONG, 2019-11-25:
cml-007-drop is ship with google's official image, could you try directly on helios, and see if you can reproduce this issue? since hatch is the old prototype, we are now pay more attention on helios.

aiChaoSONG, 2019-11-27:
This issue has high impact on user experience.

aiChaoSONG, 2019-11-29:
kernel issue,  (Kernel PR #1070) https://github.com/thesofproject/linux/pull/1070 can fix this. close this

closed 2019-11-29 by aiChaoSONG