thesofproject/sof #2096: [BUG] IPC timeout on BDW

xiulipan, 2019-11-14:
**Describe the bug**
When try to playback on BDW platform, ipc timeout happen at TRIG_START

**To Reproduce**
aplay on BDW paltform and dmesg shwo ipc timeout

**Reproduction Rate**
4/4

**Expected behavior**
aplay should we with no ipc timeout.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
Test with https://github.com/thesofproject/sof/pull/2095 with sof-bdw-nocodec.tplg

or

Test with https://github.com/thesofproject/linux/pull/1484 and https://github.com/thesofproject/sof/pull/2095 with sof-bdw-rt286.tplg

**Screenshots or console output**
<details>
 <summary>dmesg</summary>
```
[   99.874778] sof-audio-acpi INT3438:00: pcm: open stream 0 dir 0
[   99.874784] sof-audio-acpi INT3438:00: period min 192 max 16384 bytes
[   99.874787] sof-audio-acpi INT3438:00: period count 2 max 16
[   99.874790] sof-audio-acpi INT3438:00: buffer max 65536 bytes
[  100.049301] sof-audio-acpi INT3438:00: rate_min: 48000 rate_max: 48000
[  100.049304] sof-audio-acpi INT3438:00: channels_min: 2 channels_max: 2
[  100.084524] sof-audio-acpi INT3438:00: rate_min: 48000 rate_max: 48000
[  100.084529] sof-audio-acpi INT3438:00: channels_min: 2 channels_max: 2
[  100.085179] sof-audio-acpi INT3438:00: rate_min: 48000 rate_max: 48000
[  100.085184] sof-audio-acpi INT3438:00: channels_min: 2 channels_max: 2
[  100.085193] sof-audio-acpi INT3438:00: pcm: hw params stream 0 dir 0
[  100.085200] sof-audio-acpi INT3438:00: generating page table for 000000000e0ae6ec size 0xffc0 pages 16
[  100.085204] sof-audio-acpi INT3438:00: stream_tag 0
[  100.085214] sof-audio-acpi INT3438:00: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  100.085666] sof-audio-acpi INT3438:00: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  100.085675] sof-audio-acpi INT3438:00: pcm: stream dir 0, posn mailbox offset is 649728
[  100.087212] sof-audio-acpi INT3438:00: pcm: trigger stream 0 dir 0 cmd 1
[  100.087218] sof-audio-acpi INT3438:00: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  100.591396] sof-audio-acpi INT3438:00: error: ipc timed out for 0x60040000 size 12
[  100.591401] sof-audio-acpi INT3438:00: info: preventing DSP entering D3 state to preserve context
[  100.591470] sof-audio-acpi INT3438:00: error: unexpected fault 0x00000000 trace 0x00000800
[  100.591474] sof-audio-acpi INT3438:00: error: ipc host -> DSP: pending no complete no raw 0x00000800
[  100.591476] sof-audio-acpi INT3438:00: error: mask host: pending no complete no raw 0x7fff0000
[  100.591478] sof-audio-acpi INT3438:00: error: ipc DSP -> host: pending no complete no raw 0x00000000
[  100.591481] sof-audio-acpi INT3438:00: error: mask DSP: pending yes complete no raw 0x7f9e0002
[  100.591483] sof-audio-acpi INT3438:00: error: waking up any trace sleepers
[  100.591487]  Low Latency: ASoC: trigger FE cmd: 1 failed: -110
```
</details>

<details>
 <summary>dma trace</summary>
```
$ sudo ./sof-logger -l sof-bdw.ldc -t
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      2  SCHEDULE_LL          61867304.166667  61867304.000000 edule/timer_domain.c:47     timer_domain_register()
    0      2          IPC          64610050.520833   2742746.250000    src/ipc/handler.c:918    ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC          64610083.802083        33.281250    src/ipc/handler.c:921    ipc: pipe 1 comp 0 -> new (type 1)
    0      2         HOST          64610135.677083        51.875000     src/audio/host.c:349    host_new()
    0      2          DMA          64610391.927083       256.250000        src/lib/dma.c:128    dma_get() ID 1 sref = 3 busy channels 1
    0      2          IPC          64635568.802083     25176.875000    src/ipc/handler.c:918    ipc: hdr 0x30010000 tx (84) > rx (28)
    0      2          IPC          64635602.083333        33.281250    src/ipc/handler.c:921    ipc: pipe 1 comp 1 -> new (type 5)
    0      2       VOLUME          64635654.062500        51.979168 udio/volume/volume.c:144    volume_new()
    0      2       VOLUME 1.1      64635947.864583       293.802094 udio/volume/volume.c:222    vol->initial_ramp = 250, vol->ramp = 0, vol->min_value = 0, vol->max_value = 65536
[... 156 lines cut ...]
    0      2       VOLUME 1.2      64739385.625000        34.010418 udio/volume/volume.c:498    volume_ctrl_get_cmd(), SOF_CTRL_CMD_VOLUME / SOF_CTRL_CMD_SWITCH, cdata->comp_id = 2
    0      2       VOLUME 1.2      64739418.854167        33.229168 udio/volume/volume.c:505    volume_ctrl_get_cmd(), channel = 0, value = 65536
    0      2       VOLUME 1.2      64739452.343750        33.489582 udio/volume/volume.c:505    volume_ctrl_get_cmd(), channel = 1, value = 65536
    0      2          IPC          64740561.302083      1108.958374    src/ipc/handler.c:857    ipc: hdr 0x50020000 tx (108) > rx (92)
    0      2          IPC          64740593.541667        32.239582    src/ipc/handler.c:859    ipc: comp 1 -> cmd 0
    0      2       VOLUME 1.1      64740631.822917        38.281250 udio/volume/volume.c:528    volume_cmd()
    0      2       VOLUME 1.1      64740665.833333        34.010418 udio/volume/volume.c:498    volume_ctrl_get_cmd(), SOF_CTRL_CMD_VOLUME / SOF_CTRL_CMD_SWITCH, cdata->comp_id = 1
    0      2       VOLUME 1.1      64740699.062500        33.229168 udio/volume/volume.c:505    volume_ctrl_get_cmd(), channel = 0, value = 65536
    0      2       VOLUME 1.1      64740732.552083        33.489582 udio/volume/volume.c:505    volume_ctrl_get_cmd(), channel = 1, value = 65536
    0      2          IPC          65707026.979167    966294.437500    src/ipc/handler.c:857    ipc: hdr 0x50010000 tx (108) > rx (92)
    0      2          IPC          65707060.156250        33.177082    src/ipc/handler.c:859    ipc: comp 1 -> cmd 0
    0      2       VOLUME 1.1      65707099.375000        39.218750 udio/volume/volume.c:528    volume_cmd()
```
</details>

<details>
 <summary>error trace</summary>
```
$ sudo ./sof-logger -l sof-bdw.ldc
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      2         PIPE 1.9     1668112435.312500              nan src/audio/pipeline.c:658   pipeline_trigger()
    0      2          IPC         1668112397.708333              nan    src/ipc/handler.c:459   ipc: comp 1 -> trigger cmd 0x40000

```
</details>

xiulipan, 2019-11-14:
With arecord the error log is different

```bash
$ sudo ./sof-logger -l sof-bdw.ldc
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      2         PIPE 2.16    23163051377.812500              nan src/audio/pipeline.c:658  pipeline_trigger()
    0      2          IPC         23163051309.427086              nan    src/ipc/handler.c:459  ipc: comp 1 -> trigger cmd 0x40000
    0      2          SSP         23163051708.229168       398.802094 /intel/haswell/ssp.c:417  ssp_start()
    0      2  SCHEDULE_LL         23163051805.625000        97.395836 _multi_chan_domain.c:105  dma_multi_chan_domain_register()
```

lgirdwood, 2019-11-15:
@xiulipan let me list some differences to check comapred to BYT.
1) BDW has no external timer based on SSP clock, it uses CCOUNT for the timing. CCOUNT should not be gated during WFI (driver sets bit for this already)
2) BDW SSP has different clock sources than BYT, are these clocks running ?
3) BDW DMACs are unidirectional OR host/link IIRC, i.e. DMAC0 may be for host only OR for playback only. @lbetlej @mmaka1 can you guys confirm BDW DMACs

It does look like IPC, scheduling, WFI, topology is all working based on above, so I'm suspecting it's SSP, DMAC.

xiulipan, 2019-11-22:
@lgirdwood Thanks for the hint, I will try to add some log to check if the BDW hits these issues.

IPC is working with TPLG load.
Volume change is always working.
DMA trace is working.
Only playback is not working, seems to be SSP DAI driver issue.

lgirdwood, 2019-11-22:
@xiulipan ok, I'll send you some SSP configs that might help.

closed 2019-12-16 by lgirdwood