thesofproject/sof #2098: [FEATURE] Tar balls

perexg, 2019-11-14:
It would be nice to provide tar balls with the ready to use firmware files. I am doing this for my purposes at https://www.alsa-project.org/files/pub/misc/sof/ or ftp://ftp.alsa-project.org/pub/misc/sof/ , but it's far from the ideal solution.

BTW: When will be available the signed files for 1.4?

jajanusz, 2019-11-15:
I'm not sure what do you want in tarball, but if you mean source dist that is in tar instead of git repo (that's what I usually see), then you have `make dist` target implemented here: https://github.com/thesofproject/sof/blob/f004d11f306e2d910e22e06953fc252c301161a9/scripts/cmake/dist.cmake#L1-L21

IRT attaching binaries to releases I'm not sure how it should be handled to be legal and practical - firmware (ast features) is just one thing in a big chain of dependencies on top of that you have f.e. also signing and then you may need f.e. to have keys in some other SoC matching what you have in FW or some other things that may depend on whatever manufacturer thinks of.

jajanusz, 2019-11-15:
@perexg @lgirdwood @plbossart  What do you suggest?

perexg, 2019-11-15:
I am basically concerned with the useble pre-compiled binary firmware files for the linux kernel. In my opinion, it should be mix of signed and unsigned firmware files usable in a generic way for the most of devices on the marked. My tar balls are just an example. Yes, the correct licence should be provided, too. I just used the SOF license. If Intel has another idea for the signed files, let me know, please.

Fedora is using the 1.3.1 signed tar ball from my URL right now until we have the signed firmware files.

The ideal goal is to push the firmware files to linux-firmware repository. But it did not happen.

plbossart, 2019-11-15:
We are still planning on pushing the files through linux firmware. There are delays as in all big companies but there is no plan B.
If you want tar files, we can use the linux-firmware files, install them to a local directory with all the links prepared, and then tar the intel/sof parts.

lgirdwood, 2019-11-15:
@perexg PR for FW happening today or Monday, we had a bug to squash that was blocking the PR. It's ready to go shortly after more review. v1.4 next week, again another issue cropped (from production testing) up so it's likely we will just do a v1.4.1

@jajanusz I agree with @perexg, we should provide the unsigned binaries, ldc files and community signed binaries for "make dist". Make dist would need to work with XCC and GCC though.

perexg, 2019-11-15:
Another option is to create a temporary git tree with the ready to use binary firmware files with appropriate tags (something like an extension to the linux-firmware git tree). The github will create tar balls / zip files for free for you. Thanks.

lgirdwood, 2019-11-15:
@perexg All are here atm, https://github.com/thesofproject/linux-firmware

perexg, 2019-11-15:
@lgirdwood : It's unusable for the package maintainers. We need to add only SOF files to linux-firmware or alsa-firmware packages. That's the reason why I am doing tar balls with the ready to use SOF firmware files and using this as a source for the alsa-firmware package for Fedora.

But it might be a good start if you can split SOF only files to a branch and add an annotated tag to create a release...

lgirdwood, 2019-11-18:
@perexg ok, I see you now need this as standalone versioned binaries. I've already asked @xiulipan to look at having the CI create signed binaries for every git tag. The logical next step is for CI to push those binaries to a repo here that can tar, compress etc  and be cloned. This repo would have all versions though ? Or would it be better for seperate branch per release ?

perexg, 2019-11-18:
Branches with the annotated tags might be fine. We don't need to do diff on binary files and the git fetch might be more optimized. I would propose to create a separate subdirectories with firmware types with a symlink to the preferred one (signed, unsigned, gcc)?

Example (for firmware/intel/sof):

<pre>
── gcc
│   ├── sof-apl.ri
│   ├── sof-cnl.ri
│   └── sof-icl.ri
── signed
│   ├── sof-apl-signed-intel.ri
│   ├── sof-cnl-signed-intel.ri
│   └── sof-icl-signed-intel.ri
├── sof-apl.ri -> signed/sof-apl-signed-intel.ri
├── sof-cnl.ri -> signed/sof-cnl-signed-intel.ri
</pre>

lgirdwood, 2019-11-18:
@xiulipan @mengdonglin @wenqingfu fyi - we should be able to pick these binaries from CI (when CI encounters a new git release tag).
@perexg fyi, we can either build with GCC (functionally correct) or with XCC (optimised SIMD, but needs $$$ licence) and either sign with Intel private key or SOF community key. I would expect we would be able to post only XCC binaries, that are unsigned, signed with Intel key and signed with community key.

perexg, 2019-11-18:
Could you also add ldc files for traces?

xiulipan, 2019-11-19:
@lgirdwood @perexg @plbossart 
Few question here:
- Can you give some example about what is an unsigned sof-apl.ri? If we compress the binary into ri format it will always be signed with some key.
- Could you give more detail about what files should be hold in the binary release? (ldc files, ri for what platforms, ri with what key signed, with what compiler)
- What is the frequency of the binary release? (every commit? every release tag?)

perexg, 2019-11-19:
>     * Can you give some example about what is an unsigned sof-apl.ri? If we compress the binary into ri format it will always be signed with some key.

It's probably the explanation why both signed/unsigned v1.3 fw binaries work on my Lenovo notebook, but the gcc version does not.

>     * Could you give more detail about what files should be hold in the binary release? (ldc files, ri for what platforms, ri with what key signed, with what compiler)

We need a generic support for the generic distributions like Fedora / Ubuntu etc. So please, generate firmware files for *ALL* platforms. We cannot predict, which platform our users have. Please, place ldc files to a separate directory (I suggest ++intel/sof/signed/debug++ etc...)

>     * What is the frequency of the binary release? (every commit? every release tag?)

It's just an idea: Every release tag and weekly an updated development branch.

lgirdwood, 2019-11-19:
@perexg @xiulipan my understanding of "unsigned" is a plain ELF or flat binary file, but it seems the naming is misleading as the files are really signed (since they boot on the laptop). I will fix naming .
@xiulipan binary release should be *.ri, *.ldc and *.tplg on every git tag (not commit) signed with both keys (jenkins can do OTC key @zrombel @lbetlej QB will need to do Intel key).

marc-hb, 2021-08-23:
@perexg , did the more recent https://github.com/thesofproject/sof-bin/issues/46 and a couple others achieve all this? If not could you isolate and summarize what's left to do?

(just found this old issue while searching for something else)

closed 2021-09-21 by marc-hb