thesofproject/sof #2101: [BUG][CML][HDA] fixed distortion pattern at the beginning of recording on headset pipeline

aiChaoSONG, 2019-11-18:
**Describe the bug**
run arecord on headset capture pipeline, you will see volume keeps high at the beginning, even there is no sound. check the recorded wave, you can see a fixed distortion pattern, and all headset recorded waves have the same distortion.

if no headset connected when run arecord, no distortion found.

[image]

**To Reproduce**
run `arecord -Dhw:0,0 -fs32_le -r 48000 -c 2  -i -vvv headset.wav`

**Reproduction Rate**
10/10

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {fd302303}
    * SOF: {655d077f}
2) Name of the topology file
    * Topology: {sof-hda-generic-2ch.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {Dell HDA platform with ALC 3204 codec}

[sof_logger_trace.log](https://github.com/thesofproject/sof/files/3856576/sof_logger_trace.log)

lgirdwood, 2019-11-18:
@singalsu @bardliao  what do you think, this could just be the mic powering UP.

plbossart, 2019-11-18:
likely something in the HDaudio codec driver and/or bad settings applied?

bardliao, 2019-11-19:
Looks like a power up pop. @aiChaoSONG How long will the max volume continue? Will you see the same issue on DMIC recording?

aiChaoSONG, 2019-11-19:
@bardliao only see this on headset capture pipeline, and the max volume duration is very short, about 0.35s.

bardliao, 2019-11-19:
> @bardliao only see this on headset capture pipeline, and the max volume duration is very short, about 0.35s.

So I think it is a codec issue (or limitation). I think we can close this issue since there is nothing we can do.

singalsu, 2019-11-19:
This would be fixed by inserting an IIR EQ into capture path. If someone with access to this device and test an enhanced topology can contact me I can make a patch to sof-hda-generic-2ch.tplg to test with.

singalsu, 2019-11-19:
@bardliao @aiChaoSONG Can you please check if SOF PR #2109 helps for this.

lgirdwood, 2019-11-20:
@singalsu good work, btw this is worthy of a technical note on sof-decs i.e. a small howto.

singalsu, 2019-11-20:
I was able to repeat this issue with a device that I have. There's no volume ramp happening in capture, I'll try to  make a fix for this and related problems https://github.com/thesofproject/linux/issues/1463 and https://github.com/thesofproject/linux/issues/1480.

plbossart, 2019-11-20:
thanks @singalsu for looking into this.

singalsu, 2019-11-20:
I have a fix under development that is done in firmware to launch a volume ramp in beginning of copy() operations if the driver control sequence has not initiated it. I don't think it can be fixed in driver in simple way because the PM controls cache hides the unmute transition from firmware. 

Combination of ramp and DC block works well but I need to eliminate the volume setting dependent ramp length variation for the start ramp (use the whole topology defined time even if the gain request is much lower than max. volume).

lgirdwood, 2019-11-21:
@singalsu can we have topology tuple flags for ramp up/down at stream start, this way we can turn it off/on as needed.

singalsu, 2019-11-22:
We have a single token for ramp length but the same length is used for both start and during streaming. The ramp is turned off by setting length to zero.

singalsu, 2020-01-24:
@aiChaoSONG The PR #2109 is merged so I re-tested this on a WHL HDA platform with headset disconnected. The start waveform is now clean so I think this can be closed.

[image]

singalsu, 2020-01-24:
Closing this issue.

closed 2020-01-24 by singalsu