thesofproject/sof #2102: [BUG] DSP panic with Cannon Point-LP

perexg, 2019-11-18:
**Describe the bug**
I get a DSP panic with the current code in sound-next.

**To Reproduce**
Run pulseaudio with ucm2 profiles.

**Firmware**
278528 Jul 23 16:09 sof-cnl-signed-intel.ri

[   15.815177] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-5dd9a
[   15.815182] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:7:0 Kernel ABI 3:11:0
[   15.816373] sof-audio-pci 0000:00:1f.3: firmware boot complete

**Panic dump**
<pre>
[   25.969049] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[   25.972839] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[   25.973373] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[   25.973379] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   25.974167] sof-audio-pci 0000:00:1f.3: error: runtime exception
[   25.974169] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[   25.974171] sof-audio-pci 0000:00:1f.3: error: panic at :0
[   25.974174] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[   25.974176] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   25.974178] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x0000001c PS       0x00060325 SAR     0x00000000
[   25.974180] sof-audio-pci 0000:00:1f.3: EPC1     0xbe041efb EPC2     0xbe05da64 EPC3     0xbe05aaec EPC4    0x00000000
[   25.974183] sof-audio-pci 0000:00:1f.3: EPC5     0xbe05da64 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   25.974185] sof-audio-pci 0000:00:1f.3: EPS2     0x00060720 EPS3     0x00060b22 EPS4     0x00000000 EPS5    0x00060720
[   25.974187] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[   25.974188] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe2ec9d0
[   25.974191] sof-audio-pci 0000:00:1f.3: 0xbe2ec9d0: be041efb 00400000 000003e8 00040005
[   25.974194] sof-audio-pci 0000:00:1f.3: 0xbe2ec9d4: 9e0a32d4 00000000 00000000 00000000
[   25.974196] sof-audio-pci 0000:00:1f.3: 0xbe2ec9d8: b9b5bd00 8b3e2b4d a4152028 ffff9f34
[   25.974199] sof-audio-pci 0000:00:1f.3: 0xbe2ec9dc: a9089bc0 ffff9f34 a9089b80 ffff9f34
[   25.974201] sof-audio-pci 0000:00:1f.3: 0xbe2ec9e0: a9089b80 ffff9f34 92174b29 ffffffff
[   25.974203] sof-audio-pci 0000:00:1f.3: 0xbe2ec9e4: a43d8000 ffff9f34 aaa29800 ffff9f34
[   25.974206] sof-audio-pci 0000:00:1f.3: 0xbe2ec9e8: 921142e0 ffffffff 00000000 00000000
[   25.974208] sof-audio-pci 0000:00:1f.3: 0xbe2ec9ec: 00000000 00000000 00000000 00000000
[   28.666133] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[   28.666162] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   28.666379] sof-audio-pci 0000:00:1f.3: error: runtime exception
[   28.666383] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[   28.666388] sof-audio-pci 0000:00:1f.3: error: panic at :0
[   28.666392] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[   28.666397] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   28.666402] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x0000001c PS       0x00060325 SAR     0x00000000
[   28.666406] sof-audio-pci 0000:00:1f.3: EPC1     0xbe041efb EPC2     0xbe05da64 EPC3     0xbe05aaec EPC4    0x00000000
[   28.666411] sof-audio-pci 0000:00:1f.3: EPC5     0xbe05da64 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   28.666415] sof-audio-pci 0000:00:1f.3: EPS2     0x00060720 EPS3     0x00060b22 EPS4     0x00000000 EPS5    0x00060720
[   28.666420] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[   28.666424] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe2ec9d0
[   28.666430] sof-audio-pci 0000:00:1f.3: 0xbe2ec9d0: be041efb 00400000 000003e8 00040005
[   28.666435] sof-audio-pci 0000:00:1f.3: 0xbe2ec9d4: 9e0a32d4 00000000 00000000 00000000
[   28.666440] sof-audio-pci 0000:00:1f.3: 0xbe2ec9d8: a7b49a00 15d85a4b a4152028 ffff9f34
[   28.666445] sof-audio-pci 0000:00:1f.3: 0xbe2ec9dc: c047fcc0 ffffc30e 00000000 ffffffff
[   28.666450] sof-audio-pci 0000:00:1f.3: 0xbe2ec9e0: a2b46828 ffff9f34 0000004c 00000000
[   28.666455] sof-audio-pci 0000:00:1f.3: 0xbe2ec9e4: 00000001 00000000 aa1e6000 ffff9f34
[   28.666460] sof-audio-pci 0000:00:1f.3: 0xbe2ec9e8: 40010000 00000000 00000000 00000000
[   28.666465] sof-audio-pci 0000:00:1f.3: 0xbe2ec9ec: 00000000 00000000 a7b49a00 15d85a4b
[   28.666481] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[   28.666485] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   28.666501] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
</pre>

mrajwa, 2019-11-18:
@perexg, is there any chance to get firmware log or even exact FW binary you used?

perexg, 2019-11-18:
The used firmware is the signed firmware from the releases page on github (v1.3 - sof-cnl-signed-intel.ri).

My suspect is that the improved PM code added recently to the SOF driver code causes this. I'm compiling the kernel without this code right now.

I will try gather DSP logs, too.

Another issue: I cannot test gcc firmwares on this hw (Lenovo Carbon X1 7th):

<pre>
[   18.738215] sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[   18.738260] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[   18.738747] sof-audio-pci 0000:00:1f.3: error: status = 0x0000002c panic = 0x00000000
[   18.738792] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[   18.738794] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[   18.738800] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
</pre>

I tried firmware files from github and self-compiled (v1.3, v1.3.1, v1.4).

perexg, 2019-11-18:
The cnl_ipc_send_msg() with cnl_compact_ipc_compress() looks really suspicious in the SOF linux driver update. Is this code really supposed to work with the older fw binaries?

lgirdwood, 2019-11-18:
@perexg running binaries depends on the platform (as some platforms use the Intel private key) and other platforms like Chromebooks use the public signing key. It seems like you have a non Chromebook ?
ABI changes should not crash the FW or hang either, but I would expect an IPC error for unsupported  IPCs (and subsequent IPC should work).
I do agree PM changes in the driver could hang the DSP FW, since host can control some DSP clocks, power domains etc.

perexg, 2019-11-18:
> @perexg running binaries depends on the platform (as some platforms use the Intel private key) and other platforms like Chromebooks use the public signing key. It seems like you have a non Chromebook ?

Yes, it's the new Lenovo notebook... The unsigned fw binary also works but not the gcc one.

> ABI changes should not crash the FW or hang either, but I would expect an IPC error for unsupported IPCs (and subsequent IPC should work).
> I do agree PM changes in the driver could hang the DSP FW, since host can control some DSP clocks, power domains etc.

I am slowly bisecting this. 52 patches left. It seems that it's not PM.

ranj063, 2019-11-18:
> The cnl_ipc_send_msg() with cnl_compact_ipc_compress() looks really suspicious in the SOF linux driver update. Is this code really supposed to work with the older fw binaries?

@perexg This change is likely untested with older binaries but the update to add cnl_compact_ipc_compress() should only take effect when you suspend to S0ix. Would you be able to try a newer topology binary instead of the 1.3 tplg?

plbossart, 2019-11-18:
No, we need to use the topology released at the time of 1.3, there is a related bug on this https://github.com/thesofproject/linux/issues/1417

I agree with @perexg we should not have any kernel oops, the code is too flaky and I generated multiple errors when I played with the topology last week.

juimonen, 2019-11-18:
@plbossart @perexg please specify here are you using topology from the same release or has it been changed also. AFAIK we should be able to run newer kernel with older fw and tplg but if you start to mix and match we are in trouble... anyway it should not crash and give some reasonable error message.

perexg, 2019-11-18:
@plbossart : It's not a kernel oops, but the DSP oops.

Anyway, the fw / topology works with the 5.4 kernel but not with the linux-sound kernel (for 5.5). I am just bisecting what patch causes this DSP hung. The topology and fw is 1.3 (signed) for CNL. I don't have any other working firmware files for this hw platform.

perexg, 2019-11-18:
It seems that the commit 5eee2b3f60065a2530d13f28e771be48b989eb4c ("ASoC: SOF: topology: set trigger order for FE DAI link") causes this malfunction.

perexg, 2019-11-18:
https://mailman.alsa-project.org/pipermail/alsa-devel/2019-November/thread.html#158761

plbossart, 2019-11-19:
I just tried with the latest SOF kernel, the signed 1.3 firmware with this patch on a WHL device with HDaudio and dmic. No issues.

The commit above was actually generated at the request of the firmware team, so maybe it exposes a timing issue but it's not fundamentally flawed.

what I also don't get is your note that "The unsigned fw binary also works but not the gcc one."
That's not really possible to run with both the unsigned and signed binary, you'd get a boot error if you don't use the one that matches what Lenovo used for the CSME authentication.

@tlauda @jajanusz can we try and generate an updated signed firmware, or better yet, speed up the release of a 1.4 release. It's just awful to deal with such problems and break the device of Jaroslav (a VIP in audio :-))

ranj063, 2019-11-19:
@perexg thanks for the bisect. We encountered this DSP panic with the change in the trigger order and the fix has already been merged to SOF master: https://github.com/thesofproject/sof/commit/c623e9246325dbee615a5cad0c8e4b0c29976056

perexg, 2019-11-19:
https://github.com/thesofproject/sof/releases/download/v1.3/sof-cnl-signed-intel.ri

> I just tried with the latest SOF kernel (0cf356c64793058c8e43fe8a4c081ac6b3f2c86a), the signed 1.3 firmware with this patch on a WHL device with HDaudio and dmic. No issues.
> 
> The commit above was actually generated at the request of the firmware team, so maybe it exposes a timing issue but it's not fundamentally flawed.
> 
> what I also don't get is your note that "The unsigned fw binary also works but not the gcc one."
> That's not really possible to run with both the unsigned and signed binary, you'd get a boot error if you don't use the one that matches what Lenovo used for the CSME authentication.

I'm using those two files from github which both works on my Lenovo notebook:

https://github.com/thesofproject/sof/releases/download/v1.3/sof-cnl-signed-intel.ri
https://github.com/thesofproject/sof/releases/download/v1.3/sof-cnl-unsigned.ri

So perhaps sof-cnl-unsigned.ri is signed with the  community key and Lenovo accepts it? Could I generate gcc firmware files signed with the community key somehow?

Note that even gcc file from the 1.3 release does not work on my notebook:

https://github.com/thesofproject/sof/releases/download/v1.3/sof-cnl-gcc.ri

> @tlauda @jajanusz can we try and generate an updated signed firmware, or better yet, speed up the release of a 1.4 release. It's just awful to deal with such problems and break the device of Jaroslav (a VIP in audio :-))

It's my testing machine, but I'd like finally prepare the ready-to-use DMIC solution for Fedora users. They are waiting on it (and we want to deploy DMIC notebooks in our company, too).

> We encountered this DSP panic with the change in the trigger order and the fix has already been merged to SOF master: [c623e92](https://github.com/thesofproject/sof/commit/c623e9246325dbee615a5cad0c8e4b0c29976056)

Okay, but it does not help us when we have no other usable firmware. Perhaps, 1.3.2  binaries with this fix or 1.4 (signed) binaries will help a lot. Thanks.

plbossart, 2019-11-19:
> @perexg thanks for the bisect. We encountered this DSP panic with the change in the trigger order and the fix has already been merged to SOF master: [c623e92](https://github.com/thesofproject/sof/commit/c623e9246325dbee615a5cad0c8e4b0c29976056)

Well the conclusion is pretty clear. We cannot keep using sof-1.3 firmware, an update needs to be provided!

perexg, 2019-11-22:
I am trying to fix this in the linux driver using the fw ABI version check:

https://mailman.alsa-project.org/pipermail/alsa-devel/2019-November/159055.html

perexg, 2019-11-22:
Another issue: The capture rate is twice than requested for dmic input on this platform. Do prefer to create a new issue?

perexg, 2019-11-22:
Ok, it seems that if I use the sof-hda-generic-2ch.tplg for sof-hda-generic-4ch.tplg then the stereo rate is fine (the signal is really weak). What's the reason to not refuse the stereo for 4ch topology when it does not work? Or the 4ch should be converted to 2ch.

It's really difficult to work with this :-(

plbossart, 2019-11-22:
@perexg do you mind starting another issue and add @singalsu in the loop
I wonder if the NHLT reports are incorrect if you have 2 mics but the platform advertises 4.
We did add a volume boost for dmic, so maybe the topology is also old?

perexg, 2019-11-22:
The problem is not if the laptop has 2 or 4 mics. The problem is that if the user space application opens the PCM device with 4ch topology and asks for 2 channels, the fw/driver ignores that and sends 4ch stream to it (probably - the rate is doubled in my test environment). The driver is broken (it should not offer/accept the 2 channel PCM configuration in this case).

Also, we should pass 2ch/4ch configuration description to the user space, so UCM can set the right channel count value for apps.

perexg, 2019-11-22:
Created https://github.com/thesofproject/sof/issues/2131 .

plbossart, 2019-11-22:
> The problem is not if the laptop has 2 or 4 mics. The problem is that if the user space application opens the PCM device with 4ch topology and asks for 2 channels, the fw/driver ignores that and [...]

ah ok, yes, that's looks like a bug but on the driver side then. It's not a firmware bug.

> Also, we should pass 2ch/4ch configuration description to the user space, so UCM can set the right channel count value for apps.

The Chrome folks use a channel mask IIRC, but I can't recall if it's static or not.

perexg, 2020-02-21:
Another round: https://mailman.alsa-project.org/pipermail/alsa-devel/2020-February/163649.html

lgirdwood, 2020-02-21:
@perexg @plbossart can we close once fix is merged upstream.

tlauda, 2020-03-12:
@perexg Can be closed?

perexg, 2020-03-12:
Yes, it seems that this patch is not necessary. The panic is gone with the latest driver / firmware.

closed 2020-03-12 by perexg

azrdev, 2020-03-20:
I can confirm a booting Thinkpad X1 Carbon Gen.7 (no kernel or DSP crashes) of Fedora31 with package versions
kernel-5.5.9-200.fc31.x86_64
alsa-ucm-1.2.2-1.fc31.noarch
alsa-firmware-1.2.1-4.fc31.noarch

and the following kernel boot log in dmesg:
~~~

[  +0,192363] snd_soc_skl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[  +0,003582] snd_soc_skl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[  +0,079622] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[  +0,002503] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[  +0,000949] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[  +0,000793] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[  +0,001039] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[  +0,005095] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[  +0,006201] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[  +0,001249] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[  +0,055355] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[  +0,000794] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  +0,086999] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[  +0,001297] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[  +0,003388] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  +0,005487] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp1
[  +0,003264] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp2
[  +0,002034] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp3
[  +0,003339] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link Analog Playback and Capture
[  +0,002090] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link Digital Playback and Capture
[  +0,002747] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link dmic01
[  +0,002102] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link dmic16k
[  +0,005041] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:11:0
[  +0,000751] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[  +0,001698] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[  +0,000735] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[  +0,001764] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[  +0,000725] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[  +0,001415] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[  +0,000701] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[  +0,002085] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[  +0,001390] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[  +0,003297] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[  +0,001429] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[  +0,000699] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[  +0,000697] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[  +0,004298] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[  +0,001221] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[  +0,005816] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[  +0,000729] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[  +0,001730] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[  +0,013045] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[  +0,031529] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC285: line_outs=2 (0x14/0x17/0x0/0x0/0x0) type:speaker
[  +0,001304] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[  +0,000609] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[  +0,000606] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[  +0,000608] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[  +0,001169] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[  +0,044683] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[  +0,002423] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi1 <-> iDisp1 Pin mapping ok
[  +0,002735] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi2 <-> iDisp2 Pin mapping ok
[  +0,003501] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi3 <-> iDisp3 Pin mapping ok
[  +0,003387] skl_hda_dsp_generic skl_hda_dsp_generic: Analog Codec DAI <-> Analog CPU DAI mapping ok
[  +0,003277] skl_hda_dsp_generic skl_hda_dsp_generic: Digital Codec DAI <-> Digital CPU DAI mapping ok
[  +0,002329] skl_hda_dsp_generic skl_hda_dsp_generic: dmic-hifi <-> DMIC01 Pin mapping ok
[  +0,002295] skl_hda_dsp_generic skl_hda_dsp_generic: dmic-hifi <-> DMIC16k Pin mapping ok
[  +0,000141] mc: Linux media interface: v0.10
[... cut ...]
[  +0,001964] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16
[  +0,001531] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[  +0,001394] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[  +0,001720] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
[  +0,001369] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input20