thesofproject/sof #2113: [BUG][CML][KWD] KWD trigger failure under S16_LE

aiChaoSONG, 2019-11-20:
**Describe the bug**
Use S16_LE in KWD pipeline, recording cannot be triggered, but with S24_LE, recording can be triggered.

**To Reproduce**
1. reboot
2. run `arecord -Dhw:0,8 -M -N -f S16_LE -r 16000 -c2 -i -vvv -d5 ./wov.wav`

**Reproduction Rate**
10/10

**Expected behavior**
recording should be triggered

**Impact**
function abnormal

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel:  topic/sof-dev {ecc776ae}
    * SOF:  master  {63a55299}
2) Name of the topology file
    * Topology: {sof-cml-rt5682-max98357a.tplg, sof-cml-rt1011-rt5682.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {CML Chromebook with rt5682 and max98357a codec, CML Chromebook with rt5682 and rt1011codec}

[wov_no_trigger_dmesg.log](https://github.com/thesofproject/sof/files/3867143/wov_no_trigger_dmesg.log)
[wov_no_trigger_sof_error.log](https://github.com/thesofproject/sof/files/3867144/wov_no_trigger_sof_error.log)
[wov_no_trigger_sof_trace.log](https://github.com/thesofproject/sof/files/3867145/wov_no_trigger_sof_trace.log)

wwittbrx, 2019-11-21:
I cannot reproduce this issue with the following recipe (cml-008-drop):

PLATFORM: CML HATCH
OS: UBUNTU 19.04
FW: f7dfd05b9fa4 sof-cml-rt5682-max98357a.m4
KERNEL: 5d67c4b + PR1383 + 169c659

Could you check if this is true on your side as well? I would strongly recommend switching to CML 008 as it is much more stable, with multiple fixes.

aiChaoSONG, 2019-11-25:
@wwittbrx did you try sof master? please give it a try.

mengdonglin, 2019-12-26:
@mrajwa Can you please check this issue? This issue cannot be reproduced with cml-008 branch, so we suspect the master branch may missing something.

keyonjie, 2020-01-13:
Hi @aiChaoSONG can you try if reverting the below commit will fix the issue:

https://github.com/thesofproject/sof/commit/078c3c1a71390c8c69ae22d3854956891e8fb5fd

aiChaoSONG, 2020-01-13:
@keyonjie after reverting 078c3c1, still cannot trigger in s16_le.

aiChaoSONG, 2020-01-13:
After bisect, this issue is caused by 7003c1f,  after reverting 7003c1f, issue not reproduced on master.
@ktrzcinx Please take a look, and fix this.

mengdonglin, 2020-01-14:
@ktrzcinx can you please check this regression?

aiChaoSONG, 2020-01-14:
@mengdonglin  I have sent a pr to fix this

ktrzcinx, 2020-01-14:
And I approve it

ktrzcinx, 2020-01-14:
* @aiChaoSONG fix PR

closed 2020-01-14 by lgirdwood