thesofproject/sof #2114: [BUG][BYT][RT5682]Can not adjust volume  in alsamixer when aplay 44100Hz audio

Liviali155, 2019-11-20:
**Describe the bug**
Sudo reboot
aplay a 44100Hz audio  via  media playback(0,1),then adjust the volume to maximum/Minimum in alsamixer,continue to adjust the volume ,can not adjust,after that aplay via 0,0 can not adjust volume too

aplay  audio of other frequencies can not reproduce this issue

**To Reproduce**
1.sudo reboot with headset insert
2."aplay -Dhw:0,1 -r 44100 -c2 wav/44California_Dreaming.wav -vv -i" to aplay a 44100Hz audio
3.open another terminal to  open alsamixer
4.adjust the volume to maximum/Minimum
5.continue to adjust volume

```ubuntu@sh-byt-mb-rt5682-1111:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofrt5682 [sof-rt5682], device 0: Low Latency (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 1: Media Playback 1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
**Reproduction Rate**
5/5

**Expected behavior**
Volume can be adjusted normally in alsamixer when aplay

**Impact**
Can not adjust volume in alsamixer,after that aplay via 0,0 can not adjust volume too

**Environment**
platform: BYT MB with  RT5682 in I2S mode
Firmware: 63a55299 https://github.com/thesofproject/sof/commits/master
Kernel: ecc776ae https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-byt-rt5682.tplg same as firmware

<details>
 <summary>sof_errortrace</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1           SA         114486625.052083 114486624.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34167
    0      1           SA         114490636.250000      4011.197998      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34383
    0      1           SA         114494636.354167      4000.104248      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34385
    0      1           SA         114498651.875000      4015.520752      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34683
    0      1           SA         114502656.302083      4004.427002      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34767
    0      1       VOLUME 3.22    114671941.197917    169284.890625 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    118275329.062500   3603387.750000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    118275558.020833       228.958328 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    118275567.552083         9.531250 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    118728606.770833    453039.218750 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    121945741.145833   3217134.500000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    121946679.479167       938.333313 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    121946689.010417         9.531250 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    122334688.750000    387999.750000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     126824496.875000   4489808.000000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     126824718.020833       221.145828 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     126824727.916667         9.895833 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     127077393.489583    252665.578125 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     138107571.041667  11030178.000000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     138107760.104167       189.062500 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     138107770.000000         9.895833 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     138408265.572917    300495.562500 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    142351756.406250   3943490.750000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    142351906.927083       150.520828 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    142351916.458333         9.531250 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    142590224.895833    238308.437500 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.3     146137429.843750   3547205.000000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.3     146137588.125000       158.281250 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.3     146137598.020833         9.895833 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.3     146523037.552083    385439.531250 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1           SA         114430632.187500              nan      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34305
    0      1           SA         114434641.197917      4009.010498      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34477
    0      1           SA         114438631.354167      3990.156250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34289
    0      1           SA         114442667.395833      4036.041748      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34981
    0      1           SA         114446634.583333      3967.187500      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34351
    0      1           SA         114450650.000000      4015.416748      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34647
    0      1           SA         114454619.583333      3969.583252      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34063
    0      1           SA         114458652.447917      4032.864502      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34694
    0      1           SA         114462655.729167      4003.281250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34757
    0      1           SA         114466632.656250      3976.927002      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34314
    0      1           SA         114470645.156250      4012.500000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34554
    0      1           SA         114474648.125000      4002.968750      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34611
    0      1           SA         114478633.229167      3985.104248      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34325
````

</details>

[dmesgvolume.log](https://github.com/thesofproject/sof/files/3867521/dmesgvolume.log)
[sof_trace.log](https://github.com/thesofproject/sof/files/3867522/sof_trace.log)

lgirdwood, 2019-11-20:
@Liviali155 which volume control in alsamixer are you adjusting ? Do any other controls change the volume ?

Liviali155, 2019-11-21:
Bi-sect result: Caused by firmwire commit [63a55299](https://github.com/thesofproject/sof/commit/63a552992caa833ef9ac106037ca0106eab75fb8)
@tlauda  please check it

Liviali155, 2019-11-26:
On BYT MB nocodec can reproduce this issue
platform: BYT MB with nocodec
Firmware: db7671a https://github.com/thesofproject/sof/commits/master
Kernel: 58ed3272 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-byt-nocodec.tplg same as firmware

lgirdwood, 2019-11-26:
@singalsu it looks like volume not working after it's adjusted to max or min value.

Liviali155, 2019-11-29:
Not need adjusted the volume to max or min,just adjusted it ,It works at the beginning,but after adjusted sometimes,volume did not work

Liviali155, 2019-11-29:
When #2135  occurred ,error not seen on BYT RT5682,,but can be reproduced on BYT nocodec,
After #2135  is fixed，isssue  can be reproduced on BYT-RT5682&BYT nocodec

Liviali155, 2019-12-17:
Use ”amixer cget ..." to get the value of PGA3.0 3,the value will change along with adjusting in alsamixer,but the volume of output do not change,as capture

[image]

juimonen, 2019-12-17:
@Liviali155 @singalsu seems that message is again flowing correctly up to dsp component, but nothing is happening.... so in logger we see messages going to comp id 22. @Liviali155 can you attach the logs all the way from the dsp boot that we could verify that comp id 22 is actually correct volume component in correct pipeline?

Liviali155, 2019-12-18:
> @Liviali155 @singalsu seems that message is again flowing correctly up to dsp component, but nothing is happening.... so in logger we see messages going to comp id 22. @Liviali155 can you attach the [...]

I catch the log from bootup,as attachment
[dmesg1218.log](https://github.com/thesofproject/sof/files/3976871/dmesg1218.log)
[sof_trace.txt](https://github.com/thesofproject/sof/files/3976872/sof_trace.txt)

juimonen, 2019-12-18:
@Liviali155 thanks, looks like the commands are going to the correct component, nothing is just happening... btw is this related to 44.1kHz somehow (as the title says) or also happening with 48kHz files, or whatever format you play to pipeline 1?

Liviali155, 2019-12-18:
> @Liviali155 thanks, looks like the commands are going to the correct component, nothing is just happening... btw is this related to 44.1kHz somehow (as the title says) or also happening with 48kHz files, or whatever format you play to pipeline 1?

Error only seen when play 44100Hz audio via pipeline 1
Error not seen when play 48000Hz audio via pipeline 1

juimonen, 2019-12-18:
@Liviali155 ok, have you tested this before and it was working or is this regression? we had some pipeline parameter handling changes in firmware lately...

Liviali155, 2019-12-18:
> @Liviali155 ok, have you tested this before and it was working or is this regression? we had some pipeline parameter handling changes in firmware lately...

Yes, It is a regression caused by firmwire commit 63a55299

juimonen, 2019-12-18:
@tlauda fyi... @Liviali155 kind of having trouble to understand how that commit can affect volume processing though...

lgirdwood, 2019-12-18:
@singalsu any hints here ?
@juimonen volume updates are also scheduled and ramped. It could be that they are not being run by scheduler (esp if 44.1kHz conversion is utilising DSP)

slawblauciak, 2020-01-27:
@tlauda any updates on this issue?

dengyangchao, 2020-02-17:
Issue can't be reproduced with below recipe, close

Platform: BYT with nocodec  & BYT with RT5682 in I2S mode
Firmware: b206b7a https://github.com/thesofproject/sof/commits/master
Kernel: b2d98237 https://github.com/thesofproject/linux/commits/topic/sof-dev

closed 2020-02-17 by dengyangchao