thesofproject/sof #2117: [BUG] CML : suspend failures with ipc timed out for 0x40010000 size 76

sathyap-chrome, 2019-11-20:
**Describe the bug**
During Soix cycling , the suspend failure is seen multiple times resulting in eventual shutdown

**To Reproduce**
WIP. its mostly from customer reports. we are trying to find a way to reproduce the same.

**Reproduction Rate**
Not sure

**Expected behavior**
Suspend failures should not be seen while running soix.

**Impact**
This is blocking FSI of Customer and hence high impact.

**Environment**
CML 007 drop of firmware
2) Name of the topology file
    * Topology: sof-rt5682-rt1011.m4 and sof-da7219-max98357a.m4
3) Name of the platform(s) on which the bug is observed.
    * Platform:CML

**Screenshots or console output**
2019-11-14T19:40:40.842510-08:00 ERR kernel: [ 7764.368615] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
2019-11-14T19:40:40.842510-08:00 ERR kernel: [ 7764.368636] sof-audio-pci 0000:00:1f.3: error: status = 0x001d003c panic = 0x00000000
2019-11-14T19:40:40.842511-08:00 ERR kernel: [ 7764.368651] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
2019-11-14T19:40:40.842511-08:00 ERR kernel: [ 7764.368652] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
2019-11-14T19:40:40.842512-08:00 ERR kernel: [ 7764.368662] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00040001 mask 0x00000000
2019-11-14T19:40:40.842512-08:00 WARNING kernel: [ 7764.368665] sof-audio-pci 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
2019-11-14T19:40:40.842513-08:00 ERR kernel: [ 7764.419271] sof-audio-pci 0000:00:1f.3: error: reset enter failed: core_mask f adspcs 0x1d003c
2019-11-14T19:40:40.842513-08:00 ERR kernel: [ 7764.419273] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask f
2019-11-14T19:40:40.842514-08:00 ERR kernel: [ 7764.419274] sof-audio-pci 0000:00:1f.3: error: failed to power down core during suspend
2019-11-14T19:40:40.842514-08:00 ERR kernel: [ 7764.419275] sof-audio-pci 0000:00:1f.3: error: suspending dsp
2019-11-14T19:40:40.842515-08:00 ERR kernel: [ 7764.419276] sof-audio-pci 0000:00:1f.3: error: failed to power down DSP during suspend -5
2019-11-14T19:40:40.842516-08:00 ERR kernel: [ 7764.419283] pci_pm_suspend(): snd_sof_suspend+0x0/0x11 [snd_sof] returns -5
2019-11-14T19:40:40.842516-08:00 ERR kernel: [ 7764.419288] dpm_run_callback(): pci_pm_suspend+0x0/0x1a7 returns -5
2019-11-14T19:40:40.842517-08:00 INFO kernel: [ 7764.419292] sof-audio-pci 0000:00:1f.3: pci_pm_suspend+0x0/0x1a7 returned -5 after 347093 usecs
2019-11-14T19:40:40.842517-08:00 ERR kernel: [ 7764.419294] PM: Device 0000:00:1f.3 failed to suspend: error -5
2019-11-14T19:40:40.842518-08:00 DEBUG kernel: [ 7764.419297] PM: suspend of devices aborted after 893.658 msecs
[power_down.log](https://github.com/thesofproject/sof/files/3870561/power_down.log)

sathyap-chrome, 2019-11-21:
we have found a way to reproduce the issue. 

please find steps to reproduce the issue on chrome environment.

echo 0 > /var/lib/power_manager/disable_dark_resume
echo 0 >/usr/share/power_manager/disable_idle_suspend
set_power_policy --battery_idle_action=suspend --ac_idle_action=suspend 
set_power_policy --battery_idle_delay=10 --ac_idle_delay=10 
restart powerd

After doing this, allow the system idle for 10 seconds, the system will try to enter soix and fails.
I think the idle suspend path is broken.

Will capture required traces & logs tomorrow. 
I would also try to  collect more logs / sof traces in the failure scenario.
Please try on chrome environment.

Moving to CML 008 might be long term solution ( if this fix is present in cml 008), but we need a hotfix as Google FSI is with CML 007.

lgirdwood, 2019-11-21:
@tlauda I think we need to improve this as we may have some races here with external IP and D3/S0ix entry. @ranj063 also seeing similar issues on CHT. I think we should be disabling all external timers, DMICs, SSPs, DMAC and waiting for thier completion prior to returing IPC sucess to the driver. DMA trace and external timers are obvious users that could race ?

```c
static int ipc_pm_context_save(uint32_t header)
{
	//struct sof_ipc_pm_ctx *pm_ctx = _ipc->comp_data;

	trace_ipc("ipc: pm -> save");

	/* TODO: check we are inactive - all streams are suspended */

	/* TODO: mask ALL platform interrupts except DMA */

	/* TODO now save the context - create SG buffer config using */
	//mm_pm_context_save(struct dma_sg_config *sg);

	/* mask all DSP interrupts */
	arch_interrupt_disable_mask(0xffffffff);

	/* TODO: mask ALL platform interrupts inc DMA */

	/* TODO: clear any outstanding platform IRQs - TODO refine */

	/* TODO: stop ALL timers */
	platform_timer_stop(platform_timer);

	/* TODO: disable SSP and DMA HW */

	/* TODO: save the context */
	//reply.entries_no = 0;

	/* write the context to the host driver */
	//mailbox_hostbox_write(0, pm_ctx, sizeof(*pm_ctx));

	_ipc->pm_prepare_D3 = 1;

	return 0;
}
```

mengdonglin, 2019-11-22:
Progress shared by @sathyap-chrome：
"please find steps to reproduce the issue on chrome environment.

echo 0 > /var/lib/power_manager/disable_dark_resume
echo 0 >/usr/share/power_manager/disable_idle_suspend
set_power_policy --battery_idle_action=suspend --ac_idle_action=suspend 
set_power_policy --battery_idle_delay=10 --ac_idle_delay=10 
restart powerd

After doing this, allow the system idle for 10 seconds, the system will try to enter soix and fails.
I think the idle suspend path is broken.

Will capture required traces & logs tomorrow. 
I would also try to  collect more logs / sof traces in the failure scenario.
Please try on chrome environment"

sinahuang, 2019-11-22:
Issue still can be reproduced on TGL-RVP with MSI mode when doing suspeand & resume without playback stress test.

Test for 300 times, failed at 200th time.

**To reproduce:**
Run "sudo  ./stress_s3.sh  300 1 0"

Scripts:
[stress-s3.txt](https://github.com/thesofproject/sof/files/3878101/stress-s3.txt)

Dmesg:
```
[ 3647.483664] sof-audio-pci 0000:00:1f.3: snd_sof_suspend
[ 3647.483667] sof-audio-pci 0000:00:1f.3: sof_suspend
[ 3647.483669] sof-audio-pci 0000:00:1f.3: sof_send_pm_ctx_ipc
[ 3647.483674] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 3647.984013] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[ 3647.984016] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[ 3647.984034] sof-audio-pci 0000:00:1f.3: unknown ROM status value 01009701
[ 3647.984358] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x001d003c trace 0x00000000
[ 3647.984389] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0x00000000 rirb 00
[ 3647.984390] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 3647.984423] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00040000 mask 0x00000000
[ 3647.984425] sof-audio-pci 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
[ 3648.004049] ACPI: EC: interrupt blocked
[ 3652.946527] ACPI: EC: interrupt unblocked
[ 3653.006767] sof-audio-pci 0000:00:1f.3: snd_sof_resume
[ 3653.006769] sof-audio-pci 0000:00:1f.3: sof_resume
[ 3653.006771] sof-audio-pci 0000:00:1f.3: loading firmware
[ 3653.006780] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
[ 3653.006785] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[ 3653.006791] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[ 3653.006796] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
[ 3653.006796] sof-audio-pci 0000:00:1f.3: booting DSP firmware
[ 3653.006839] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[ 3653.007651] sof-audio-pci 0000:00:1f.3: error: stream reset failed
[ 3653.007651] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
[ 3653.007659] sof-audio-pci 0000:00:1f.3: error: dma prepare for fw loading err: ffffff92
[ 3653.007660] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[ 3653.007660] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware after resume -110
[ 3653.007664] PM: dpm_run_callback(): pci_pm_resume+0x0/0xa0 returns -110
[ 3653.007665] PM: Device 0000:00:1f.3 failed to resume: error -110
[ 3653.012641] serial 00:03: activated
```

**Test recipe**

Kernel:https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 68cadd8
FW: https://github.com/thesofproject/sof/commits/master commit: 9b5dc8c
Tplg: Same with TGL FW branch commit: sof-tgl-rt711-rt1308-nohdmi.tplg rename to sof-tgl-rt711-rt1308.tplg
Platform:TGL RVP with onboard ACL711 in SDW mod and ALC1308 in I2S mode.

[s3_test_200_logger_all.log](https://github.com/thesofproject/sof/files/3878102/s3_test_200_logger_all.log)
[dmesg_200.log](https://github.com/thesofproject/sof/files/3878103/dmesg_200.log)

keyonjie, 2019-11-22:
when you can reproduce the issue, can you try if https://github.com/thesofproject/sof/pull/2125 help?

sinahuang, 2019-11-22:
Issue still can be reproduced with master(commit: 9b5dc8c+PR #2125) on TGL-RVP when doing suspeand & resume without playback stress test.

Test for 300 times, failed at 130th time.

**Test recipe**

Kernel:https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 68cadd8
FW: https://github.com/thesofproject/sof/commits/master commit: 9b5dc8c+PR #2125 
Tplg: Same with TGL FW branch commit: sof-tgl-rt711-rt1308-nohdmi.tplg rename to sof-tgl-rt711-rt1308.tplg
Platform:TGL RVP with onboard ACL711 in SDW mod and ALC1308 in I2S mode.

[s3_test_130_logger_all.log](https://github.com/thesofproject/sof/files/3879288/s3_test_130_logger_all.log)
[s3_test_135_logger_all.log](https://github.com/thesofproject/sof/files/3879289/s3_test_135_logger_all.log)

mmaka1, 2019-11-22:
Please retest with #2132. Disassembly clearly shows that address of literal block passed to the dcache lock instruction in power_down() routine was computed incorrectly. And it might cause rare issues when trying to load the control bits for IPC ctx_save ipc response. This patch fixes the issue. No guarantee this was the only one, so the issue may be still reproducible.

lgirdwood, 2019-11-23:
@sathyap-chrome @sinahuang #2132 now merged in master. Please test.

sathyap-chrome, 2019-11-24:
Thanks liam, will retest.
unfotunately, the issue itself is hard to reproduce, but as soon as we hit, will test with this.

sinahuang, 2019-11-25:
Issue still can be reproduced when doing suspeand & resume without playback stress test after PR #2132 merged in master . 

Test for 300 times, failed at 205th time.

**Test recipe**

Kernel:https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 68cadd8
FW: https://github.com/thesofproject/sof/commits/master commit: d6fe2b1
Tplg: Same with TGL FW branch commit: sof-tgl-rt711-rt1308-nohdmi.tplg rename to sof-tgl-rt711-rt1308.tplg
Platform:TGL RVP with onboard ACL711 in SDW mod and ALC1308 in I2S mode.

[s3_test_205_logger_all.log](https://github.com/thesofproject/sof/files/3885428/s3_test_205_logger_all.log)
[dmesg _205.log](https://github.com/thesofproject/sof/files/3885440/dmesg._205.log)

sathyap-chrome, 2019-11-25:
Hi liam, fw team,
 since customer PV is on CML 007, we tried to make this change on top of cml_007 branch and the configs failed - error attached.
 Seems there is couple of dependent patches we need to take.
 Kindly help porting back this fix - assuming we are having high confidence to fix the issue.
[cml_007_plus_pr2132_build_fail.txt](https://github.com/thesofproject/sof/files/3886119/cml_007_plus_pr2132_build_fail.txt)

wwittbrx, 2019-11-25:
I am unable to reproduce the issue with the provided steps on any of CML platforms available to me right now. I will try TGL next and update if there are results.

wenqingfu, 2019-11-25:
> I am unable to reproduce the issue with the provided steps on any of CML platforms available to me right now. I will try TGL next and update if there are results.

can you please elaborate what environment you used that you cannot reproduced the issue?

ranj063, 2019-11-25:
> can you please elaborate what environment you used that you cannot reproduced the issue?

@wenqingfu it looks like the kernel used by @sinahuang is the SDW dev tree. We should be using the sof-dev branch for this test or the cml-007 release branch

plbossart, 2019-11-25:
> > can you please elaborate what environment you used that you cannot reproduced the issue?
> 
> @wenqingfu it looks like the kernel used by @sinahuang is the SDW dev tree. We should be using the sof-dev branch for this test or the cml-007 release branch

there's no real difference, the sdw is rebased on topic/sof-dev every time we have a new PR.

wenqingfu, 2019-11-26:
Um... my question is actually to @wwittbrx ...

wwittbrx, 2019-11-26:
@wenqingfu 
I tried reproducing the issue with env provided in ChromeOS-12607.8.0 image (no changes on my part beyond installing the OS; I later upgraded to 12607.38.0 version with same result) on Helios platform - this is the main recipe for this issue for me. I also ran tests on Hatch with Ubuntu on cml-007-drop-stable recipe and on CML RVP with recent sof-dev and master branches. The tests were running for 10+ hours.

mengdonglin, 2019-11-26:
@sathyap-chrome 

Maybe we should solve https://github.com/thesofproject/sof/issues/2136 at first?

Because in your attached kernel log https://github.com/thesofproject/sof/files/3870561/power_down.log, there is the same FW loading failure as issue https://github.com/thesofproject/sof/issues/2136, before ipc timeout happens:

[   24.947381] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[   24.954809] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[   24.964376] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[   24.971292] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[   24.979470] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
...

[   82.769785] sof-audio-pci 0000:00:1f.3: calling pci_pm_suspend+0x0/0x1a7 @ 5957, parent: pci0000:00
[   83.076718] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[   83.076741] sof-audio-pci 0000:00:1f.3: error: status = 0x001d003c panic = 0x00000000
[   83.076756] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[   83.076758] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00040000
[   83.076768] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00040001 mask 0x00000000
[   83.076770] sof-audio-pci 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
[   83.127098] sof-audio-pci 0000:00:1f.3: error: reset enter failed: core_mask f adspcs 0x1d003c
[   83.127100] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask f
[   83.127101] sof-audio-pci 0000:00:1f.3: error: failed to power down core during suspend
[   83.127103] sof-audio-pci 0000:00:1f.3: error: suspending dsp
[   83.127104] sof-audio-pci 0000:00:1f.3: error: failed to power down DSP during suspend -5

sathyap-chrome, 2019-11-26:
@mengdonglin The logs i attached are indeed from the DSP load fail scenario. But the customer reports suggest they were able to use Audio until a potential suspend / idle suspend resulted in fw errors and eventual shutdown due to continuous suspend entry failures.

Unfortunately we are able to reproduce only the DSP load failure issue and not the the customer reported ones.

keyonjie, 2019-11-27:
Hi @sathyap-chrome from the dmesg you attached in the bug description, you didn't blacklist the snd_soc_skl driver? I think this is why we are seeing the probing fail, the SST driver was loading.

sathyap-chrome, 2019-11-27:
@keyonjie  the fix to blacklist SST driver , i'll do. But i feel this is not realted as this is there in Chrome tree for long time. we should have hit more issues if SST driver was interfering with SOF driver.

lgirdwood, 2019-11-27:
@sathyap-chrome can you blacklist it anyway (just to rule it out), there could be recent and unrelated kernel/FS/coreboot changes that could effect the probing order (and any IP programming flow).

sathyap-chrome, 2019-11-27:
@lgirdwood yes. i'll be doing the same.

lancedkhou, 2019-11-29:
I tried to disable SST by adding "blacklist snd_soc_skl" on CML 007 device but it can  not fix  DSP load fail scenario.

mengdonglin, 2019-12-02:
@lancedkhou  DSP load fail scenario has the same root cause as #2136

keqiaozhang, 2019-12-04:
Since the log attached here are incorrect and the steps to reproduce the issue are not clear, so I filed a new one. Let's track this issue in https://github.com/thesofproject/sof/issues/2164.

closed 2019-12-04 by keqiaozhang