thesofproject/sof #2118: Signed binaries for v1.4?

hsanjuan, 2019-11-20:
Should signed binaries be available for v1.4 like they were for v1.3?

lgirdwood, 2019-11-21:
@hsanjuan Yes, we are working on this, it's a manual process. We also think we have an issue with v1.4 which we are stress testing and may just go straight to v1.4.1 in next few days. + @fkaczmar @lbetlej @wenqingfu

hsanjuan, 2019-11-21:
Thanks for the info!

perexg, 2019-11-25:
Any update? We are a bit stuck, because v1.3 binaries/topologies are not so much usable. Thanks.

lgirdwood, 2019-11-25:
@perexg starting to trickle through now, I have CNL and ICLonly atm will be posting shortly. Others to follow.

lgirdwood, 2019-11-25:
@perexg https://github.com/thesofproject/sof/releases/tag/v1.4.1 has APL, CNL and ICL signed bianries attached. Others to follow.

perexg, 2019-11-25:
@lgirdwood : Does not work on CNL, could you check, if those files are really signed with the Intel key?

<pre>
[   15.090406] sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[   15.090865] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[   15.091408] sof-audio-pci 0000:00:1f.3: error: status = 0x0000002c panic = 0x00000000
[   15.091632] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[   15.091633] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[   15.091634] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
</pre>

v1.3 signed are fine...

plbossart, 2019-11-25:
the 1.4.1 firmware works for me on a WHL device (same as CNL).

perexg, 2019-11-25:
@plbossart : Do you run this with the kernel which is in the linux-sound-next tree (tiwai)?

plbossart, 2019-11-25:
> @plbossart : Do you run this with the kernel which is in the linux-sound-next tree (tiwai)? I mean the public 5.5 code.
> 
> Tried 2nd time, no way...

i run our development tree, which follows both Mark and Takashi trees weekly.
https://github.com/thesofproject/linux/commits/topic/sof-dev

can you share the PCI ID, the release for CML is missing but will be available tomorrow

perexg, 2019-11-25:
Lenovo ThinkPad Carbon X1 7th gen:

<pre>
00:1f.3 0403: 8086:9dc8 (rev 11) (prog-if 80)
	Subsystem: 17aa:2292
</pre>

perexg, 2019-11-25:
It seems that the problem is on ICL, too: https://github.com/thesofproject/sof/issues/2126

plbossart, 2019-11-25:
> Lenovo ThinkPad Carbon X1 7th gen:
> 
> 00:1f.3 0403: 8086:9dc8 (rev 11) (prog-if 80)
> 	Subsystem: 17aa:2292

that's the same PCI ID I tried. were you able to use the 1.3 binaries on this one?

perexg, 2019-11-25:
> > Lenovo ThinkPad Carbon X1 7th gen:
> > 00:1f.3 0403: 8086:9dc8 (rev 11) (prog-if 80)
> > Subsystem: 17aa:2292
> 
> that's the same PCI ID I tried. were you able to use the 1.3 binaries on this one?

Yes, the signed 1.3 binaries can be loaded to DSP here.

hsanjuan, 2019-11-25:
> @lgirdwood : Does not work on CNL, could you check, if those files are really signed with the Intel key?
> 
> [   15.090406] sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
> [   15.090865] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
> [   15.091408] sof-audio-pci 0000:00:1f.3: error: status = 0x0000002c panic = 0x00000000
> [   15.091632] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
> [   15.091633] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
> [   15.091634] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
> 
> v1.3 signed are fine...

I also have this on Kernel 5.4.

lgirdwood, 2019-11-26:
Ok, so I've checked and confim the signing key is the same for both v1.3 and v1.4.1, however I can see some differences with some of the signing flags. @lbetlej @fkaczmar can you guys check. Thanks.

perexg, 2019-11-27:
Ping. We really need the updated signed firmware. v1.3 is not sufficient.

lgirdwood, 2019-11-27:
@perexg apologies, we may have an incorrect validation flag setting in the binary that some CSE firmwares use to verify the binary. @lbetlej is discussing with the CSE folks to make sure we use correct flags per target device.

lbetlej, 2019-11-27:
@perexg Could you run https://downloadcenter.intel.com/download/28632/Intel-Converged-Security-and-Management-Engine-Intel-CSME-Detection-Tool and provide info on : 
*** Host Computer Information *** and *** Intel(R) ME Information *** on your Lenovo ThinkPad Carbon X1 7th gen ?

perexg, 2019-11-27:
<pre>
*** Host Computer Information ***
Name: localhost.localdomain
Manufacturer: LENOVO
Model: 20QE000VMC
Processor Name: Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz
OS Version: Fedora 31 Thirty One (5.4.0-0.rc8.git0.1.alsa6.fc31.x86_64)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 12.0.40.1433
</pre>

perexg, 2019-11-29:
Ping.

lgirdwood, 2019-11-29:
@perexg unfortunately not yet, I guess we should this be unblocked early next week as some required folks are on ThanksGiving holidays.

perexg, 2019-12-02:
Ok, another Lenovo X1 Carbon 7th (the first one was a test platform from Lenovo for partners, this one is the final):

<pre>
*** Host Computer Information ***
Name: localhost.localdomain
Manufacturer: LENOVO
Model: 20QES2P407
Processor Name: Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz
OS Version: Fedora 31 Thirty One (5.4.0-2.alsa8.fc31.x86_64)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 12.0.47.1524
</pre>

Same errors for the firmware download.

moreka, 2019-12-04:
Same for me here on Lenovo X1 Carbon 7th
```
*** Host Computer Information ***
Name: carbon
Manufacturer: LENOVO
Model: 20QES16700
Processor Name: Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz
OS Version:    (5.4.1-arch1-1)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 12.0.47.1524
```

perexg, 2019-12-04:
Ping? 9 days....

lbetlej, 2019-12-05:
@perexg @hsanjuan The updated binaries (CNL and ICL) for SOF 1.4.1 release are available for download. Could you test them out and report if issue can be closed ? Thank you.

moreka, 2019-12-05:
For me the binary loads without problems now. But still, there is only noise when I try to record audio (by using `pavucontrol` I see that the microphone is always having some noise, even when everything is quite). Here's the related dmesg log for sof:
```
sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
sof-audio-pci 0000:00:1f.3: use msi interrupt mode
sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:10:0
sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
sof-audio-pci 0000:00:1f.3: firmware boot complete
sof-audio-pci 0000:00:1f.3: Topology: ABI 3:7:0 Kernel ABI 3:10:0
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
input: sof-skl_hda_card Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input14
input: sof-skl_hda_card Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input15
input: sof-skl_hda_card HDMI/DP, pcm=11 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16
input: sof-skl_hda_card HDMI/DP, pcm=12 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
input: sof-skl_hda_card HDMI/DP, pcm=13 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
```

plbossart, 2019-12-05:
Thanks @moreka for the update

> sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:10:0
> sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
> sof-audio-pci 0000:00:1f.3: firmware boot complete [...]

this looks like a new topology is needed? are you using the topology file that came with the device?

> Also after booting is complete, I'm constantly observing the following lines in dmesg (they're constantly printed):
> 
> ```
> [  250.655905] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
> [  250.754114] sof-audio-pci 0000:00:1f.3: firmware boot complete
> ```

yes, it's a known issue, we are working on this. It's harmless, please ignore this misleading log.

moreka, 2019-12-05:
Thanks @plbossart. It was always confusing for me how to obtain the topology files. (I am using the one that comes with the Ubuntu's linux-firmware package... this was supposed to be for sof1.3, though). Is there a place I can download the most recent topology file for my machine?

hsanjuan, 2019-12-05:
@moreka you can build them (I follow instructions at https://github.com/thesofproject/sof/tree/master/tools)

perexg, 2019-12-05:
The tar ball with signed / unsigned fw and topology files is here: ftp://ftp.alsa-project.org/pub/misc/sof/sof-firmware-1.4.1.tar.bz2 .

I did a quick test and this fw can be finally loaded to DSP. There are 'Dmic0 Capture Volume / Switch' controls and only 4 channels are accepted for DMIC capture.

@plbossart : I think that it's time to push this fw to linux-firmware repo ASAP with the topology files.

plbossart, 2019-12-05:
> The tar ball with signed / unsigned fw and topology files is here: https://www.alsa-project.org/files/pub/misc/sof/sof-firmware-1.4.1.tar.bz2 .
> 
> I did a quick test and this fw can be finally loaded to DSP. There are 'Dmic0 Capture Volume / Switch' controls and only 4 channels are accepted for DMIC capture.
> 
> @plbossart : I think that it's time to push this fw to linux-firmware repo ASAP with the topology files.

the topology files should go with UCM files IMHO. That's where we add/remove controls.

moreka, 2019-12-05:
> The tar ball with signed / unsigned fw and topology files is here: https://www.alsa-project.org/files/pub/misc/sof/sof-firmware-1.4.1.tar.bz2 .
>
> I did a quick test and this fw can be finally loaded [...]

I have used the provided files by @perexg and still the following is shown:
```
[   10.788457] sof-audio-pci 0000:00:1f.3: error: invalid kcontrol count 2 for volume
[   10.788565] sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 6 name : Dmic0 stream none reply 0
[   10.788702] sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget Dmic0
[   10.788839] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
[   10.788934] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[   10.789029] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[   10.789130] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -22
```

ranj063, 2019-12-05:
[   10.788457] sof-audio-pci 0000:00:1f.3: error: invalid kcontrol count 2 for volume
@moreka This seems to suggest the kernel is older? Previously, we limited the number of kcontrols to 1 but not it can be > 1.

perexg, 2019-12-05:
> > @plbossart : I think that it's time to push this fw to linux-firmware repo ASAP with the topology files.
> 
> the topology files should go with UCM files IMHO. That's where we add/remove controls.

I don't think so. It affects how the driver works, thus it's part of the firmware not the user space. We can handle the differences using UCM2 conditions.

plbossart, 2019-12-05:
> I don't think so. It affects how the driver works, thus it's part of the firmware not the user space. We can handle the differences using UCM2 conditions.

well, the topology exposes PCM and kcontrols to applications. the kernel doesn't care about the contents, it just parses the file and does whatever is asked of it.
We've always said we want UCM and topology to be aligned.

plbossart, 2019-12-05:
There is al

> well, the topology exposes PCM and kcontrols to applications. the kernel doesn't care about the contents, it just parses the file and does whatever is asked of it.
> We've always said we want UCM and [...]

not to mention that the topology files can be regenerated at will, unlike firmware which has to be signed and cannot be compiled unless you have the proper non-public compiler tools.

plbossart, 2019-12-05:
> not to mention that the topology files can be regenerated at will, unlike firmware which has to be signed and cannot be compiled unless you have the proper non-public compiler tools.

and I forgot that the initial topologies were in alsa-lib, not the linux-firmware tree.

pfps, 2019-12-05:
TLDR: Things are better, but nowhere near perfect on a Lenovo Yoga C940.  There are glitches that require reboot.  Also, how do I figure out how to enable other speakers or get a topology file?  

Well, the signed 1.4 firmware certainly did something me on my Lenovo Yoga C940 (Ice Lake).  It loaded the firmware and I could play through pulseaudio

idefix ~> journalctl -b -1 | grep "sof-\|alsa-\|pulseaudio"
Dec 05 11:19:43 idefix kernel: sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
Dec 05 11:19:43 idefix kernel: sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
Dec 05 11:19:43 idefix kernel: sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
Dec 05 11:19:43 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
Dec 05 11:19:43 idefix kernel: sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
Dec 05 11:19:43 idefix kernel: sof-audio-pci 0000:00:1f.3: use msi interrupt mode
Dec 05 11:19:43 idefix kernel: sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Dec 05 11:19:43 idefix kernel: sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
Dec 05 11:19:43 idefix kernel: sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Dec 05 11:19:43 idefix kernel: sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
Dec 05 11:19:43 idefix kernel: sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: Topology: ABI 3:7:0 Kernel ABI 3:11:0
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
Dec 05 11:19:44 idefix kernel: sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
Dec 05 11:19:44 idefix kernel: input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input23
Dec 05 11:19:44 idefix kernel: input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input24
Dec 05 11:19:44 idefix kernel: input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input25
Dec 05 11:19:44 idefix kernel: input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input26
Dec 05 11:19:44 idefix kernel: input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input27
Dec 05 11:19:46 idefix audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=alsa-state comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Dec 05 11:19:46 idefix alsactl[1365]: Found hardware: "sof-hda-dsp" "Realtek ALC298" "HDA:8086280f,80860101,00100000 HDA:10ec0298,17aa3818,00100103 cfg-dmics:2" "" ""
Dec 05 11:19:56 idefix rtkit-daemon[1383]: Successfully made thread 2081 of process 2081 (/usr/bin/pulseaudio) owned by '1316' high priority at nice level -11.
Dec 05 11:19:56 idefix rtkit-daemon[1383]: Successfully made thread 2507 of process 2081 (/usr/bin/pulseaudio) owned by '1316' RT at priority 5.
[... cut ...]
Now switching around with pulseaudio is not causing problems.

I played some music and a sound sweep.  It looks as if only two of the five speakers in my Lenovo Yoga C940 are active, which is the same as with the hda setup.  What's a good way of figuring out what is going on and what can be done to try out different combinations?  Do I need a topology file (and can I get that out of Windows)?

Also, using pavucontrol to switch from internal speakers to HDMI and back results in no sound.  What might be going on here?

perexg, 2019-12-05:
> and I forgot that the initial topologies were in alsa-lib, not the linux-firmware tree.

Only source topology files were in alsa-lib (now moved to a separate repo). From my point of view, instead of the hard coded things in the driver (the usual way), you introduced this description to reduce the work in the driver (like the device tree does) and I think that the original idea was that the hardware vendors create own topology / UCM combos. The topology files are loaded just like another firmware files at the driver initialization and then parsed to create the whole picture in the driver. Thus they define the driver behaviour.

Looking to the code, only skylake based devices are using this in the whole ASoC tree. The generic topology names are used and slightly modified by the hardware configuration (dmics, chipset etc.) for SOF. The topology names are generic otherwise.  For the old skl drivers, the skl_nhlt_update_topology_bin() function is used, which **(by surprise)** creates the firmware name using the NHLT OEM identifiers.

The driver requires those files to be activated, thus they should belong to the firmware package.

perexg, 2019-12-05:
> TLDR: Things are better, but nowhere near perfect on a Lenovo Yoga C940. There are glitches that require reboot. Also, how do I figure out how to enable other speakers or get a topology file?

I'm trying to get additional information from Lenovo for your hardware. If speakers are connected to HDA, the SOF firmware won't help you, but it's just a step forward. Also, I will update Fedora bits because the HDMI issue is resolved in pulseaudio now: https://gitlab.freedesktop.org/pulseaudio/pulseaudio/merge_requests/213 .

pfps, 2019-12-05:
@pxreg Getting information from Lenovo would be awesome.  (I tried and got the usual "ask Linux" response.)  Getting information about the speakers would be useful even if they are connected to HDA as I could then check out how to make them work.  However, it looks to me as if the other speakers are not connected via HDA - at least I was unable to get them to emit sounds using hda-analyzer, and I tried to make all the available output pin complexes match the one that was working.  Let me know when the updated pulseaudio is available and I'll give it a whirl.

perexg, 2019-12-05:
> @pxreg Getting information from Lenovo would be awesome. (I tried and got the usual "ask Linux" response.) Getting information about the speakers would be useful even if they are connected to HDA as [...]

The other tricks are to play with the EAPD settings and with GPIO settings in the HDA codec, but it would be better to get more info from the hardware vendor (I communicate directly with some people from Lenovo - no promises, but we may get something).

pfps, 2019-12-05:
@pxreg I admit I don't understand EAPD and all the GPIO settings. I tried fiddling with EAPD but all it did was turn off all sound.

plbossart, 2019-12-05:
> Only source topology files were in alsa-lib (now moved to a separate repo). From my point of view, instead of the hard coded things in the driver (the usual way), you introduced this description to [...]

The point is that those binary topology files should be handled by a package, not part of linux-firmware.

> Looking to the code, only skylake based devices are using this in the whole ASoC tree. The generic topology names are used and slightly modified by the hardware configuration (dmics, chipset etc.) [...]

no, we use the topology on Baytrail, Cherrytrail and all platforms supported by SOF. this is not HDAudio specific.

> The driver requires those files to be activated, thus they should belong to the firmware package.

I maintain they should be a separate package since they are not provided in binary form.

hsanjuan, 2019-12-07:
> @perexg @hsanjuan The updated binaries (CNL and ICL) for SOF 1.4.1 release are available for download. Could you test them out and report if issue can be closed ? Thank you.

I confirm the cnl binary loads fine for me.

closed 2019-12-07 by hsanjuan