thesofproject/sof #2123: [BUG] WFI Panic when processing CTX_SAVE IPC

ranj063, 2019-11-21:
**Describe the bug**
Panic when processing CTX_SAVE IPC while trying to wait_for_interrupt with level 0 from the IPC task.

**To Reproduce**
Boot FW with PM enabled (https://github.com/thesofproject/linux/pull/1523)

**Reproduction Rate**
100%

**Expected behavior**
No panic with WFI

**Impact**
When the panic happens, the FW sends the panic message to the host
and goes into an infinite loop while(1);
This doesnt seem to impact any usecase and the resume flow thereafter but the DSP is not really in D3 either. 

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev + (https://github.com/thesofproject/linux/pull/1523)
    * SOF: https://github.com/thesofproject/sof/commit/9b5dc8c6ef2f53799db451c68e86308b238f5678
2) Name of the topology file
    * Topology: sof-byt-*.tplg/sof-cht-*.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: BYT/CHT

ranj063, 2019-11-21:
@tlauda FYI.

Moving the WFI to the audio task with run level 0 helps fix the panic but then the FW fails to boot randomly upon resuming from D3.

plbossart, 2019-12-04:
> @tlauda FYI.
> 
> Moving the WFI to the audio task with run level 0 helps fix the panic but then the FW fails to boot randomly upon resuming from D3.

@ranj063 can you clarify if PR #2137 fully solves this issue or not? It not clear from your write-up if additional changes are needed or not.

slawblauciak, 2020-01-21:
Is this issue still reproducible on master?

tlauda, 2020-03-13:
@ranj063 Ping.

closed 2020-03-19 by tlauda