thesofproject/sof #2124: [BUG] GLK Chromebooks with da7219 have 400ms delay waiting for SRM lock

fletcherw, 2019-11-21:
cc @cujomalainey

**Describe the bug**
There is a 400ms delay when calling snd_pcm_prepare on glk chromebooks using the da7219 since the code to wait for the SRM lock times out after 400ms. I spoke with the author of the codec driver at Dialog, and their response was:

"I'm almost certain this is because the platforms in question are not enabling the WCLK before the codec's PLL is being enabled. [...] So on the Apollo lake and Kaby lake platforms (and probably other Intel ones using DA7219) what would happen is that the codec's PLL would be enabled before WCLK, you see the timeout in the kernel logs after the large delay, and then subsequently (at some point) the Intel I2S related code would then enable the WCLK and the PLL would then lock with SRM."

**To Reproduce**
Just starting a stream using "aplay /dev/zero" is sufficient to trigger this bug. There will be a noticeable delay of about 700ms before audio starts.

The message "da7219 i2c-DLGS7219:00: SRM failed to lock" will also appear in dmesg.

**Reproduction Rate**
Every time a stream is started with the codec powered down.

**Expected behavior**
The SRM lock bit should be flipped before the 400ms is elapsed and we should not see the delay/warning message in dmesg.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://chromium.git.corp.google.com/chromiumos/third_party/kernel/+/19618df1bca52fcd9f3196005ec9505868c90287
    * SOF: unsure
2) Name of the topology file
    * Topology: presumably sof-glk-da7219.m4
3) Name of the platform(s) on which the bug is observed.
    * Platform: GLK Chromebooks

plbossart, 2019-11-21:
Humm... All machine drivers based on da7219 pretty much do the same thing, so not sure why this would be specific to GLK? 
besides, the MCLK is provided, so not sure why the codec would lock on the WCLK? The PLL reference should the be the MCLK, not WCLK.

cujomalainey, 2019-11-22:
This is not specific to GLK, we are just logging this here as a GLK bug since this is the only SOF based board that we have the da7219 on that we are seeing this issue with currently and figured since SOF is under active development we should report it here first.

lgirdwood, 2019-11-22:
@sathyap-chrome fyi

plbossart, 2019-11-22:
> This is not specific to GLK, we are just logging this here as a GLK bug since this is the only SOF based board that we have the da7219 on that we are seeing this issue with currently and figured since SOF is under active development we should report it here first.

ok, thanks for the precision.
I think at some point we discussed turning on the bclock earlier, if it was needed, but if we'd need to turn the fsync/WCLK earlier that's a completely different story, we need need to start dummy transfers.
@lbetlej please correct me if I am saying something silly here.

cujomalainey, 2019-11-22:
Jimmy pointed out there appears to be something similarly done for rt5664 on kbl. Maybe something similar needs to be done for all da7219 boards? https://chromium-review.googlesource.com/c/chromiumos/third_party/kernel/+/764232/

plbossart, 2019-11-22:
one of the difference with KBL is that the SOF firmware does not provide (yet) the ability to start BCLK/FSYNC on demand from the machine driver.
It's something we know we'll need but it's not there just yet.

lgirdwood, 2019-11-23:
@plbossart thinking aloud here would could have a tuple flag for this in DAI topology (to enable certain clocks immediately after hw_params are known) e.g.

1) hw_params IPC is sent from host to FW. hw_params propagates through pipeline.
2) DAI driver gets hw_params and checks it's topology clock flags.
3) DAI Enables MCKL/BCLK.
4) Back on host, the machine driver prepare() stalls for clock lock time.
5) Host then calls trigger(START)

plbossart, 2019-11-25:
> @plbossart thinking aloud here would could have a tuple flag for this in DAI topology (to enable certain clocks immediately after hw_params are known) e.g.
>
> 1. hw_params IPC is sent from host to FW. [...]

Yes, and I remember discussing this with @lbetlej. We can enable MCLK and BCLK as soon as the config is known, and we can use the existing SSP quirk mask to indicate the desire behavior.
The issue is for the WCLK/FSYNC, you have to enable it with dummy transfers so it's a completely different solution that would be needed.

lgirdwood, 2019-11-25:
@plbossart ouch, I can see why codec vendor has done this but from a system integration perspective it's undesirable (pretty much for every serial DAI IP). Gah.

plbossart, 2019-11-25:
@lgirdwood see https://github.com/thesofproject/linux/blob/58ed327217ba03698a14191fe17c540551bb956e/sound/soc/intel/boards/kbl_rt5663_rt5514_max98927.c#L102 for examples of what was done in the past to toggle the clocks on/off from the machine driver.

tlauda, 2020-03-19:
@lgirdwood @plbossart Keeping this here or maybe turning into feature?

lgirdwood, 2020-03-19:
@plbossart machine driver infrastructure in ASoC  ? or something similar ? My only other though is putting this 400ms delay in codec trigger() (not used atm IIRC)

plbossart, 2020-03-19:
we really need to expose MCLK, BCLK, FSYNC with the clock framework and let the machine driver turn them on/off at will (either on startup/shutdown or on DAPM event).

it's been done before by the Skylake driver. The main issue for us is the FSYNC which can only work if we start dummy DMA transfers with zero-ed data. MCLK and BCLK can be turned on without audio transfers.

lgirdwood, 2020-03-20:
@plbossart we could hook this into hw_params IPC. i.e. have a topology flag that just says "keep playing silence until trigger(start)". This way FW can play silence, we schedule pipeline normally but we manually disconnect the DAI source component from the DAI buffer (until we get trigger(start) where we can then reconnect the DAI buffer source component and process the full pipeline).
@tlauda any take on this ?

plbossart, 2020-03-20:
We already have parameters for this in hw_params, but we need this even outside of hw_params, when there is no streaming.
E.g. if you want to turn the clocks on at boot and leave them on, it has to be a separate IPC.
In the cAVS days, the IPC was something related to DMA configs.

lgirdwood, 2020-03-20:
@plbossart so you mean for the DAI topology load part ? I though we had that part, its should be simple enough from FW side. i.e. we append the "enabled DAI playback of 0s" check when we do the pipeline complete IPC, this would switch ON the DAI (we would need to de-couple the source component like before though).

plbossart, 2020-03-20:
@lgirdwood I meant than the machine driver could control when the clock are turned off/on, for example when the dailinks are initialized, or on a DAPM event. This is really machine specific and doesn't belong in the topology btw, it's tied to external codec requirements, not to the firmware topology

macchian, 2020-04-10:
@plbossart , @lgirdwood, @tlauda ,  could we have your comments on 

he main issue for us is the FSYNC which can only work if we start dummy DMA transfers with zero-ed data. MCLK and BCLK can be turned on without audio transfers.the fsync/WCLK turn on earlier in machine driver? Is it possible to

lgirdwood, 2020-04-10:
@macchian yes this is all doable. 
1) FW needs some new IPCs to turn on/off SSP clocks when asked by machine driver (it can send the zerod DMA buffer).
2) machine driver needs to create some DAPM events to send the new IPCs. @plbossart IIRC, there is already CLOCK DAPM types, so perhaps we could define these in machine driver and have sof-core do the IPC (meaning any changes to machine driver are purely DAPM related).

plbossart, 2020-04-10:
@lgirdwood it was my understanding that folks wanted to use  devm_clk_get("sspN_fsync"), clk_prepare_enable(). That's how it was done in the Skylake driver.

lgirdwood, 2020-04-10:
@plbossart fine by me, the FW change is the same either way.

plbossart, 2020-04-10:
right, but the clock framework is a bit of a pain to implement.

lgirdwood, 2020-04-10:
The path of least resistance is maybe DAPM then ?

plbossart, 2020-04-10:
No I am thinking that maybe the right thing to do is have the SOF core create a clock client - but only when an I2S-based machine driver is used and not for HDaudio. Then the client sends IPCs as required. Since we can play with pm_runtime in the prepare_enable/disable_deprepare, the DSP would remain on as long as the clock is needed.
We could even prototype this with a small change to @ranj063 's PC test client, instead of flooding the DSP we'd only send specific clock-related IPCs.

bardliao, 2020-08-11:
@lgirdwood @plbossart Won't MCLK be turned on in or right after snd_sof_rintime_resume()? 
@macchian Could you measure the delay in each process? 
For example DSP boot: 20ms, waiting for MCLK ready: 100ms, codec power up: 150ms...
So we can focus on the critical part to shorten the delay time.

lgirdwood, 2020-08-11:
@macchian is this still an issue or has it been solved hence the lack of discussion ?

cujomalainey, 2020-08-11:
I am still seeing activity on internal bug

brentlu, 2020-08-12:
Turning on mclk does not help because da7219 tries to lock the fsync in prepare stage. Another possible fix is using it's "PLL bypass mode" but it requires 11.288/24.576/49.152MHz mclk. This frequency could be achieved by using cardinal clock but KBL does not support it.

brentlu, 2020-08-18:
Another issue is there is 220ms delay which is caused by codec driver itself (30ms x 2 in da7219_mixout_event(), 40ms x 4 in da7219_mixout_event()). Google is about to check the 220ms delay with DA.

plbossart, 2021-07-14:
@brentlu @cujomalainey @fletcherw @bzhg does the recent work with the early BCLK enabling solve this SRM lock delay? I vaguely recall seeing such a statement, but the explanations above only talk about FSYNC which is still enabled in the trigger phase when data transfers start (the actual data come with a delay due to SOF pipelining).

Also adding @lgirdwood and @sathya-nujella on the Intel side.

cujomalainey, 2021-07-15:
I haven't tested, I am working remote till aug ~10 (currently at the mercy of westjet and they are doing their best to mess with me) so I can't pull ToT and test. I didn't see any work land in the machine driver for the da7219, we would need to enable the clock early there wouldn't we?

plbossart, 2021-07-15:
we use the same topology as for the CS42L42 and the changes in the kernel are related to the SSP in general, not the machine driver.