thesofproject/sof #2126: [BUG] Cannot load firmware on Acer Switch SF314-57G

perexg, 2019-11-22:
**Describe the bug**
The user cannot load signed firmware v1.3 on Acer Swift SF314-57G:

<pre>
Manufacturer:      Acer
Product Name:      Swift SF314-57G
Product Version:   V1.07
Firmware Version:  V1.07
Board Vendor:      IL
Board Name:        Floris_IL
</pre>

https://bugzilla.redhat.com/show_bug.cgi?id=1772498#c16

Discussion:

What is the expected behaviour, if the DSP has own (vendor) authentication key? Basically, how we can determine the right firmware for the given DSP in the generic Linux distribution? We need detect this automatically somehow. Can DSP give any hints to the driver before the fw upload?

plbossart, 2019-11-22:
@perexg thanks for bringing our attention on this

The executive summary is:

All platforms use the firmware signed with the production Intel key.
The exceptions are: Up2 and GLK+ chromebooks which are signed with the community key.
I am not aware of OEMs signing the base firmware, only extensions which are currently not supported by SOF.

we cannot query the key used, that's unfortunate but that's the reality. When the firmware is provided through linux-firmware, I'll add a quirk for these exceptions to point to the firmware signed with the community key.

I have been able to use the signed firmware on Acer devices since 1.3 

In this case, I would think it's a kernel issue (either a missing patch or a bad commit from Intel).

can you share the git tree used?

perexg, 2019-11-22:
Yep, its 5.4-rc8 with extra backports from 5.5: https://github.com/perexg/linux/tree/perex/fedora-test

plbossart, 2019-11-22:
Could you try removing all S0ix patches, e.g. starting with c13c962816ca038d8d8b71e8a51e8058ba78c39b

That series seems to introduce problems and there is on-going work to fix it.

perexg, 2019-11-22:
Per this kernel log, I don't think that fw is active, it appears like that the upload fails. The PM is not active (I think) in this case.

<pre>
[    5.793409] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    5.793702] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    5.793718] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    5.794122] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    5.794148] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    5.794416] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    5.794775] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    5.799105] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    5.799109] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    6.804322] sof-audio-pci 0000:00:1f.3: error: codec #2 probe error, ret: -5
[    6.808731] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    6.808735] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    9.877648] sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    9.878144] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[    9.878161] sof-audio-pci 0000:00:1f.3: error: status = 0x0000002c panic = 0x00000000
[    9.878166] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[    9.878167] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[    9.878167] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
</pre>

plbossart, 2019-11-22:
The first error appears related to the HDaudio codec (or hdmi), can you check what happens if you select the FORCE_NOCODEC mode and use a nocodec topology? That would remove dependencies on the codecs and check if the firmware load works or not.

plbossart, 2019-11-23:
@perexg another point to verify is the PCI ID of the device. We recently split CFL, CNL and CML firmwares instead of using sof-cnl.ri across the board due to additional firmware manifest/signature differences. If you are using the CNL file on CML, it's likely not going to work.

IIRC correctly CFL and CNL are still identical, but CML has a different manifest. @jajanusz please correct me if I am wrong.

perexg, 2019-11-23:
The complete alsa-info output is in the Fedora bugzilla (URL provided in the bug description):

<pre>
00:1f.3 0401: 8086:34c8 (rev 30)
	Subsystem: 1025:1407
</pre>

lgirdwood, 2019-11-23:
@perexg I've tagged v1.4.1, @lbetlej can you sign.

plbossart, 2019-11-25:
@perexg @lgirdwood PCI ID 34c8 is IceLake so that's a new one.

plbossart, 2019-11-25:
@perexg can you please with the file attached  (to be copied as /lib/firmware/intel/sof/sof-icl.ri)
I'd like to verify this is solved with the upcoming release, and if not to figure out the issue.

[sof-icl-signed.ri.gz](https://github.com/thesofproject/sof/files/3887252/sof-icl-signed.ri.gz)

perexg, 2019-11-25:
I am just a "proxy" for this from the Fedora's bugzilla. I redirected the user here, so hopefully, he will test this.

lgirdwood, 2019-11-25:
new signed binaries now on v1.4.1 release page.

omg, 2019-11-25:
Original reporter here. Unfortunately new firmware makes no difference at all

lbetlej, 2019-12-05:
@omg The updated binaries (CNL and ICL) for SOF 1.4.1 release are available for download. Could you test them and report if issue can be closed ? Thank you.

omg, 2019-12-06:
Unfortunately I've lost ability to boot into Linux after BIOS update. I'll try to do something about it

lgirdwood, 2019-12-17:
@omg any updateon your BIOS/Linux installation. Are you able to test ?

paulmenzel, 2020-01-18:
There is the same issue in the Ubuntu bug tracker: [
PCI/internal sound card not detected on a Swift SF314-57](https://bugs.launchpad.net/ubuntu/+source/alsa-driver/+bug/1856366/). Release 1.4.1 did not solve the issue.

```
 sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
 sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
 sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
 sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
 sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
 sof-audio-pci 0000:00:1f.3: use msi interrupt mode
 sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915
```

paulmenzel, 2020-01-18:
Clear Linux: Acer Swift 5: https://community.clearlinux.org/t/new-installation-no-sound-swift-5-i7-1065gt-iris-plus/2407

paulmenzel, 2020-01-18:
The issue title should have *Swift* instead of *Switch* I believe.

lgirdwood, 2020-01-27:
@perexg @omg @paulmenzel fwiw v1.42 has been released and signed now.
https://github.com/thesofproject/sof/releases/tag/v1.4.2
Apologies, it's not packaged atm as discussions still ongoing on how to achive this.

ltune, 2020-01-28:
Have still the problem with v1.4.2:
`[   11.316829] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   11.316981] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   11.341573] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   11.341736] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   11.341762] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[   11.341987] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   11.341992] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   11.342191] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   11.342359] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   11.346610] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   11.346611] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   11.348580] sof-audio-pci 0000:00:1f.3: error: codec #0 probe error, ret: -2
[   11.352572] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   11.352573] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   11.408362] iwlwifi 0000:00:14.3: base HW address: 0c:dd:24:cc:70:cc
[   11.422048] thermal thermal_zone9: failed to read out thermal zone (-61)
[   11.423850] iwlwifi 0000:00:14.3 wlp0s20f3: renamed from wlan0
[   11.434988] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
[   11.434990] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[   11.435368] sof-audio-pci 0000:00:1f.3: firmware boot complete`

lgirdwood, 2020-01-28:
@ltune thanks, your log is showing that the firmware has booted. Is there any subsequent data in dmesg ?
The unexpected IRQ sometimes shows but is harmless. However the codec probe error usually is more serious and IIRC there have been some HDA updates in sof-dev recently to address similar issues. @kv2019i any comment on the codec probe ?

kv2019i, 2020-01-28:
@ltune wrote:
> [   11.346611] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
> [   11.348580] sof-audio-pci 0000:00:1f.3: error: codec #0 probe error, ret: -2

Hmm, #0 is your external HDA codec (speaker/mics/headset) and -2/NOENT tells that matching codec driver was not found. Can you make sure your kernel is compiled with all the HDA codec drivers included, e.g:

CONFIG_SND_HDA_CODEC_REALTEK=m
CONFIG_SND_HDA_CODEC_ANALOG=m
CONFIG_SND_HDA_CODEC_SIGMATEL=m
CONFIG_SND_HDA_CODEC_VIA=m
CONFIG_SND_HDA_CODEC_HDMI=m
CONFIG_SND_HDA_CODEC_CIRRUS=m
CONFIG_SND_HDA_CODEC_CONEXANT=m
CONFIG_SND_HDA_CODEC_CA0110=m
CONFIG_SND_HDA_CODEC_CA0132=m
CONFIG_SND_HDA_CODEC_CMEDIA=m
CONFIG_SND_HDA_CODEC_SI3054=m

If not still working, can you share the .config file.

slawblauciak, 2020-02-18:
The original error code visible in the dmesg is a result of CSME security error. As others have mentioned, a problem with the key signed binary.

kv2019i, 2020-02-18:
@slawblauciak wrote:
> The original error code visible in the dmesg is a result of CSME security error. As others have mentioned, a problem with the key signed binary.

I went through the older comments and it seems @ltune confirmed 1.4.2 did boot on the Acer DUT in question (https://github.com/thesofproject/sof/issues/2126#issuecomment-579099179). I checked the linked distro bugs, and I see no conflicting reports on this, so I think we can conclude the remaining issue is not related to firmware loading anymore. So I'll remove the bios tag -- let's keep that for verified cases.

Ok for @ltune @perexg @lgirdwood ?

perexg, 2020-02-18:
Yes, the 1.4.2 firmware boots fine.

lgirdwood, 2020-02-18:
@kv2019i ack, I guess this is kernel related now with HDA integration. @ltune have you had any success with the kernel config suggestion or have you tried a recent kernel (last update was 21 days ago) ?

ltune, 2020-02-19:
Hi guys, sorry for late response.

@ kv2019i wrote:
> Can you make sure your kernel is compiled with all the HDA codec drivers included, e.g:
>
> CONFIG_SND_HDA_CODEC_REALTEK=m
> CONFIG_SND_HDA_CODEC_ANALOG=m
> CONFIG_SND_HDA_CODEC_SIGMATEL=m [...]

I took the ubuntu mainline kernel, in the[ build log ](https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.5/BUILD.LOG.amd64) you see that those options are set:
`  LD [M]  sound/pci/hda/snd-hda-codec-realtek.o
  LD [M]  sound/pci/hda/snd-hda-codec-cmedia.o
  LD [M]  sound/pci/hda/snd-hda-codec-analog.o
  LD [M]  sound/pci/hda/snd-hda-codec-idt.o
  LD [M]  sound/pci/hda/snd-hda-codec-si3054.o
  LD [M]  sound/pci/hda/snd-hda-codec-cirrus.o
  LD [M]  sound/pci/hda/snd-hda-codec-ca0110.o
  LD [M]  sound/pci/hda/snd-hda-codec-ca0132.o
  LD [M]  sound/pci/hda/snd-hda-codec-conexant.o
  LD [M]  sound/pci/hda/snd-hda-codec-via.o
  LD [M]  sound/pci/hda/snd-hda-codec-hdmi.o
  LD [M]  sound/pci/hda/snd-hda-intel.o`

@lgirdwood wrote:
> your log is showing that the firmware has booted. Is there any subsequent data in dmesg ?
I just booted without the snd_hda_intel.dmic_detect=0 and have the same output. I cannot see anything more, but here is the full dmesg
[dmesg_20200219.log](https://github.com/thesofproject/sof/files/4224556/dmesg_20200219.log)

@kv2019i wrote:
>  @ltune confirmed 1.4.2 did boot on the Acer DUT in question (#2126 (comment)). I checked the linked distro bugs, and I see no conflicting reports on this, so I think we can conclude the remaining issue is not related to firmware loading anymore.

Right I see also "firmware boot complete" in dmesg, but the result is I have no sound cards available. So for now I continue with the snd_hda_intel.dmic_detect=0  option.

ltune, 2020-02-19:
@paulmenzel  wrote
> The issue title should have _Swift_ instead of _Switch_ I believe.
right it is acer swift-314-57-77MU

perexg, 2020-02-19:
@ltune - show output from 'alsa-info.sh' . The SOF driver requires also updates in pulseaudio configs etc, so it's difficult to tell you what does not work, if we don't have additional information.

ltune, 2020-02-19:
Hi @perexg , find my alsa-info attached.
[alsa-info.txt.vlfiHPlolr.log](https://github.com/thesofproject/sof/files/4224917/alsa-info.txt.vlfiHPlolr.log)
let me know if there is anything else I can do in this issue.

perexg, 2020-02-19:
Ok, it's another issue with the driver (I overlooked this in the initial report).

<pre>
sof-audio-pci 0000:00:1f.3: error: codec #0 probe error, ret: -2
</pre>

So 5.5 kernel does not work for this hw.

lgirdwood, 2020-02-19:
@kv2019i do you know if this is HDA analog or HDMI HDA codec ? We've seen this before but my understanding this was fixed ?

kv2019i, 2020-02-19:
Thanks @ltune . This is definitely still a problem, but the remaining problem is no longer a SOF firmware issue.

@perexg @lgirdwood This is a error attaching the codec driver for the external HDA codec, so nothing to do with HDMI.

As return code is -2/ENOENT, the failing code is in:

sound/soc/sof/intel/hda-codec.c:hda_codec_load_module()

Call to:

device_attach(hda_codec_dev(codec))

is returning zero (later converted to -ENOENT), so this means:

```
device_attach()
 * Returns 1 if the device was bound to a driver;
 * 0 if no matching driver was found;
 * -ENODEV if the device is not registered.
```

So for some reason no matching codec driver is found.

kv2019i, 2020-02-19:
I'll move this to kernel driver side, let's continue there.