thesofproject/sof #2131: [BUG] Linux driver - wrong PCM channel constraints for dmic

perexg, 2019-11-22:
The 4ch topology does not work correctly with the ALSA's PCM API. There is no constraint to not accept the 2 channel stream configuration in this case, which results with the wrong captured stream (probably 4 channel stream which looks like 2 channel stream with the doubled rate for apps).

If I use the sof-hda-generic-2ch.tplg as sof-hda-generic-4ch.tplg then the stereo rate is fine.

Also, we should pass 2ch/4ch configuration description to the user space (in snd_soc_card->components), so UCM can set the right channel count value for apps.

perexg, 2019-11-22:
@plbossart : As requested, adding @singalsu to this bug...

plbossart, 2019-11-22:
@perexg could you trace the value returned by this function

````
static int check_nhlt_dmic(struct snd_sof_dev *sdev)
{
	struct nhlt_acpi_table *nhlt;
	int dmic_num;

	nhlt = intel_nhlt_init(sdev->dev);
	if (nhlt) {
		dmic_num = intel_nhlt_get_dmic_geo(sdev->dev, nhlt);
		intel_nhlt_free(nhlt);
		if (dmic_num == 2 || dmic_num == 4)
			return dmic_num;
	}

	return 0;
}
````

just to check that the detection is correct.

Also which topology are you using?

perexg, 2019-11-22:
It's really straight that dmic_num is 4 here, looking to hda_init_caps() in sound/soc/sof/intel/hda.c , default module parameters (no dmic_num override). I believe that there are 4 mics in hw.

The capture PCM device should not allow 2 channel operation or there should be a possibility to reduce the number of channels in DSP (from 4 to 2) when the app asks for it.

Topology is from the 1.3.1 tag.

singalsu, 2019-11-25:
The topology should should set the the PCM channels min and max to four and attempt to do "arecord -c 2" should be rejected. I'll check here if I can repeat the issue. Thanks for reporting it!

singalsu, 2019-11-25:
I get with topology sof-hda-generic-4ch.tplg this:

```
$ arecord -Dhw:0,6 -c 4 -r 48000 -f S16_LE -d 10 rec06_4ch.wav
Recording WAVE 'rec06_4ch.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Channels 4
$ arecord -Dhw:0,6 -c 2 -r 48000 -f S16_LE -d 10 rec06_2ch.wav
Recording WAVE 'rec06_2ch.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
arecord: set_params:1305: Channels count non available

```
Is the same topology that you used @perexg ?

perexg, 2019-11-25:
I am using the topology files from v1.3.1 tag.

singalsu, 2019-11-25:
I did the check with git master of today. With topology from v1.3.1 I get this error instead:

```
$ arecord -Dhw:0,6 -c 2 -r 48000 -f S16_LE -d 10 rec06_2ch.wav
Recording WAVE 'rec06_2ch.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
arecord: pcm_read:2103: read error: Input/output error
```

With FW build from the same v1.3.1 also the same error.  Illegal channels count is not rejected cleanly. Is a more recent version possible for you?

perexg, 2019-11-25:
It seems that @lgirdwood finally release v1.4.1. I will test it ASAP.

perexg, 2019-12-12:
Fixed with 1.4.1 topology files.

closed 2019-12-12 by perexg