thesofproject/sof #2134: [BUG] Cannot load firmware on Lenovo Yoga C940 with cpu IceLake i7-1065G7

perexg, 2019-11-24:
The user cannot load signed firmware v1.3:

<pre>
[    4.286091] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    4.286213] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    4.286225] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    4.286412] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    4.286418] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.286546] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    4.286737] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.290985] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    4.290986] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    5.328957] sof-audio-pci 0000:00:1f.3: error: codec #2 probe error, ret: -5
[    5.331010] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    5.331014] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    8.399958] sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    8.400470] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[    8.400490] sof-audio-pci 0000:00:1f.3: error: status = 0x0000002c panic = 0x00000000
[    8.400496] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[    8.400497] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[    8.400498] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110

00:1f.3 Multimedia audio controller: Intel Corporation Device 34c8 (rev 30)
	Subsystem: Lenovo Device 382e
</pre>

https://bugzilla.redhat.com/show_bug.cgi?id=1772498

pfps, 2019-11-25:
I'm the original user for this bug.  What can be done to diagnose the problem?

perexg, 2019-11-25:
@pfps: We have finally new firmware from Intel. Please, wait few hours (or to tomorrow). I'm updating Fedora packages right now.

joshk0, 2019-12-10:
I'm experiencing a similar issue. Please let me know if I should file a separate bug.

```
$ dmesg | grep sof-audio-pci
[   28.997977] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   28.997980] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   28.998098] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   29.002307] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   29.006849] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   29.006850] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   29.034759] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   29.034761] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   32.101032] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[   32.101047] sof-audio-pci 0000:00:1f.3: error: status = 0x0000002c panic = 0x00000000
[   32.101051] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[   32.101052] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[   32.101053] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
```

I am running Ubuntu 19.10 Eoan on 5.3.0-24-generic. Hardware is a Lenovo X1 Carbon 7th Gen. The sound card is as follows (It's different from that of the OP but the error seems very similar)

```
00:1f.3 Audio device: Intel Corporation Device 02c8 (prog-if 80)
	Subsystem: Lenovo Device 2292
	Flags: bus master, fast devsel, latency 64, IRQ 177
	Memory at ea23c000 (64-bit, non-prefetchable) [size=16K]
	Memory at ea000000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: <access denied>
	Kernel driver in use: sof-audio-pci
	Kernel modules: snd_hda_intel, snd_soc_skl, sof_pci_dev
```

I can provide more hardware info on request. Thank you!

pfps, 2019-12-10:
The most recent versions load for me, but I haven't had a chance to put everything through its paces because of other issues.   You need special kernels and not-yet-generally-released versions of firmware, alsa, and pulseaudio, though.  Ubuntu tends to lag quite a bit behind the bleeding-edge versions of software so it may be a while until everything is picked up there.

The X1 V7 is different from the Yoga C940 and may be better supported.

joshk0, 2019-12-10:
Yeah, worse yet i seem to have a different build of the V7. Most people with V7s have [this sound card](https://certification.ubuntu.com/catalog/component/8086:9dc8) (8086:9dc8 instead of 8086:02c8) and it seems like Ubuntu 19.10 does work with those, but 02c8 seems to be slightly different.

plbossart, 2019-12-10:
02c8 == CometLake LP
9dc8 == CannonLake

@joshk0 which firmware version were you trying to use?

@lbetlej FYI

joshk0, 2019-12-13:
Hi, thanks for your answer. I tried using the [1.3 signed one](https://github.com/thesofproject/sof/releases/download/v1.3/sof-cnl-signed-intel.ri). I wasn't sure if i could go to the very latest one on the kernel version that I'm on (5.3.0-24-generic in Ubuntu); can i?

Also, can you confirm my mental model that I'm not even getting to the point where the topology files are a factor? As long as I'm overwriting /lib/firmware/intel/sof/sof-cnl.ri with whatever i'm trying to use, that should be getting picked up when the kernel module loads, right?

BTW -- if i try to rmmod and modprobe again, it causes a kernel BUG (i'm not on that computer now and can't paste it), not sure if that matters. LMK!

kv2019i, 2019-12-13:
@joshk0 @pfps There are multiple issues in the logs. First HDMI codec initializing fails
[    5.328957] sof-audio-pci 0000:00:1f.3: error: codec #2 probe error, ret: -5
... and then FW load fails:
[   32.101032] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110

I may have idea how to solve #1. A fix in i915 driver is needed on some ICL platforms (0x34C8, sof-icl.ri). The patches were now included in v.5.5-rc1. The patches are these:
 drm/i915: save AUD_FREQ_CNTRL state at audio domain suspend                                                                                                   
 drm/i915: Fix audio power up sequence for gen10+ display                                                                                                      
 drm/i915: extend audio CDCLK>=2*BCLK constraint to more platforms         

In latest sound master, we've also added a feature to pass "codec_mask" parameter to snd_sof_intel_hda_common (e.g. with codec_mask=1 you can ignore the HDMI codec and probe should pass). Unfortunately this not in v5.5-rc1 and earlier and is only very recently merged to sound master.

For the other issue of FW loading, it seems that at least for some people the issue is solved. @perexg as the original submitted, is the issue on C940 now solved?

kv2019i, 2019-12-13:
@joshk0 wrote:
> I'm experiencing a similar issue. Please let me know if I should file a separate bug.

I think so yes. Your device has a different generation audio-DSP (0x02c8 -> cAVS1.8 -> sof-cnl.ri, see:
https://thesofproject.github.io/latest/platforms/intel-cavs/index.html ), so the issue potentially different than what is seen on IceLake (0x34c8, sof-icl.ri).

> Hi, thanks for your answer. I tried using the [1.3 signed one](https://github.com/thesofproject/sof/releases/download/v1.3/sof-cnl-signed-intel.ri). I wasn't sure if i could go to the very latest one on the kernel version that I'm on (5.3.0-24-generic in Ubuntu); can i?

In general we have backwards compability for minor-version changes in ABI. The old-kernel new-fw combo is in practise less tested, so some issues may be seen.

> Also, can you confirm my mental model that I'm not even getting to the point where the topology > files are a factor? As long as I'm overwriting /lib/firmware/intel/sof/sof-cnl.ri with whatever i'm trying

It is highly recommended to use topology files that match the FW release (both topology files and FW are versioned). So you if change sof-cnl.ri, you should take topology files from same release. There's ongoing work to improve how topology files are distributed, see e.g. https://github.com/thesofproject/sof/issues/2200

> BTW -- if i try to rmmod and modprobe again, it causes a kernel BUG (i'm not on that computer now and can't paste it), not sure if that matters. LMK!

That should work and the test is covered by SOF CI, so it if happens with latest upstream kernel, a separate bug should be filed for this as well.

joshk0, 2019-12-15:
Update: The latest sof-cnl.ri file works with the original 5.3.0-24-generic kernel so I won't file an issue. BTW, I also tried that on 5.5-rc1, which now wants sof-cml.ri. On that kernel version, I got it to work by symlinking sof-cnl.ri to sof-cml.ri. Could that be because of https://patchwork.kernel.org/patch/11262555/ ? I find it strange that this is the only place on the internet which mentions that filename.

perexg, 2019-12-15:
I believe that Intel should release signed sof-cml.ri, too. The changes are probably queued to 5.6 kernel - https://git.kernel.org/pub/scm/linux/kernel/git/broonie/sound.git/log/?h=for-5.6 .

kv2019i, 2019-12-17:
@joshk0  wrote:
> Update: The latest sof-cnl.ri file works with the original 5.3.0-24-generic kernel so I won't file an  issue.

Ok, good! 

> BTW, I also tried that on 5.5-rc1, which now wants sof-cml.ri. On that kernel version, I got it to work by symlinking sof-cnl.ri to sof-cml.ri. Could that be because of [...]

Yes, this is very recent change (in upstream the main patch is "ASoC: Intel: acpi-match: split CNL tables in three", part of 5.5-rc1 already). Matching FW release has not been made, but it will be included (FYI @lgirdwood and @xiulipan )

slawblauciak, 2020-01-21:
Any updates on this issue? Is it still relevant in the light of the aforementioned commit?

lgirdwood, 2020-01-21:
@slawblauciak I've tagged v1.4.2 now as it passes all validation on the development boards, I think we just need @lbetlej to arrange the signing now.

pfps, 2020-01-21:
I think that this can be closed.  I have been able to load the software on my Yoga C940.  (There are other issues with this laptop, but those are reported elsewhere.)

paulmenzel, 2020-01-21:
@pfps, could you please paste the relevant log lines (including the Linux version, and firmware version) for the archive and reference?

danielcherubini, 2020-01-21:
@pfps could you update this page https://wiki.archlinux.org/index.php/Lenovo_Yoga_c940 with the update to SOF, and it would be nice if you could add any other issues you have?

pfps, 2020-01-21:
@paulmenzel OK.  
I've attached information from a successful load and semi-successful startup.  Sometimes the load succeeds but the startup fails.  I think that this has been addressed recently, but the current Fedora 31 kernel doesn't have SOF so I'm waiting for everything to be available in Fedora Rawhide.
@danielcherubini I did some minor changes to the Arch page.
[Lenovo-C940-SOF.txt](https://github.com/thesofproject/sof/files/4092350/Lenovo-C940-SOF.txt)

paulmenzel, 2020-01-21:
Thanks. The gist should be.

```
Jan 21 11:48:40 idefix kernel: sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
Jan 21 11:48:40 idefix kernel: sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Jan 21 11:48:40 idefix kernel: sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
Jan 21 11:48:40 idefix kernel: sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
Jan 21 11:48:40 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
Jan 21 11:48:40 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
Jan 21 11:48:40 idefix kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
```

No idea what version 1:1:0-fcf6c is.

perexg, 2020-01-21:
It's wrong tag interpretation in the released binary firmware from Intel. Use 'fcf6c' git hash prefix only. It should be fixed with the recent firmware files. It's 1.4.1 firmware. https://github.com/thesofproject/sof/commit/fcf6c8c416f790ad3d9f76bf75b5b528c11896c7

pfps, 2020-02-02:
I'm running with a version of Fedora that I think should have good SOF support - kernel 5.4.14-200.fc31.x86_64 

Sometimes everything starts fine and the mic is fine and the speakers are OK.  Sometimes the firmware loads but I see an immediate failure, as in
Feb 01 15:02:07 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
Feb 01 15:02:07 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
Feb 01 15:02:07 idefix kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
Feb 01 15:02:07 idefix kernel: skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to init link iDisp1: -517

Even when sound works initially I sometimes experience later failures.  The worst one is thousands (maybe even hundreds of thousands) or message like
Feb 01 16:35:07 idefix kernel: sof-audio-pci 0000:00:1f.3: ASoC: can't set Analog CPU DAI hw params: -16
Feb 01 16:35:07 idefix kernel:  Analog Playback and Capture: ASoC: hw_params BE failed -16
Feb 01 16:35:07 idefix kernel:  HDA Analog: ASoC: hw_params BE failed -16
This may only be happening after suspending and resuming.

danielcherubini, 2020-02-02:
It’s working fantastic on Arch 5.5 on the C940

kv2019i, 2020-02-03:
@pfps wrote:
> Sometimes everything starts fine and the mic is fine and the speakers are OK. Sometimes the firmware loads but I see an immediate failure, as in
> Feb 01 15:02:07 idefix kernel: sof-audio-pci [...]

If you can reproduce this, can you file a new bug on this and provide full kernel logs? Firmware is loading here, so we can close this bug. This could be a case where communication with i915 is failing and that is blocking audio driver load. 

> Even when sound works initially I sometimes experience later failures. The worst one is thousands (maybe even hundreds of thousands) or message like
> Feb 01 16:35:07 idefix kernel: sof-audio-pci [...]

And ditto here.

jlaham, 2020-02-08:
@danielcherubini is it working for you with PulseAudio on your C940?

I'm running GNOME on Arch (package versions below) with no custom patching. On a normal boot all sound devices (connected at 0000:00:1f.3) are detected, however I'd get no sound. Digging in a little further, aplay and arecord will list devices detected and alsamixer will run, but they will all hang before completing. Further digging led me to find that PulseAudio remains in a state of "Starting" and never actually starts.

If I mask pulseaudio.service, then aplay, arecord, and alsamixer will list devices detected without hanging (i.e. they operate normally), and I can get sound since ALSA loads the drivers normally. At that point, if I run pulseaudio manually after booting into GNOME, most of the time it will run fine and I'll get speakers and the mic working. I've also tried blacklisting `snd_hda_intel` and `snd_soc_skl` but that hasn't led to any noticeable changes.

At first this sounds like it may be an issue with PulseAudio, but examining `dmesg` I see this notice showing up on boot, which makes me wonder if there is some sort of race condition happening somewhere:
```
[    4.018569] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
```

Boot Logs (`$ dmesg | grep 1f.3`) => https://pastebin.com/raw/uFdkQVD9

**Package versions:**
- linux 5.5.2.arch1-1
- linux-firmware 20200122.1eb2408-1
- pulseaudio 13.0-3
- pulseaudio-alsa 2-5
- alsa-firmware 1.2.1-2
- alsa-lib 1.2.1.2-3
- alsa-plugins 1.2.1-1
- alsa-topology-conf 1.2.1-1
- alsa-ucm-conf 1.2.1.2-1
- alsa-utils 1.2.1-2
- sof-firmware 1.4.2-1

poscat0x04, 2020-02-09:
Can confirm it's working on Arch with kernel 5.5+

lgirdwood, 2020-02-10:
@jlaham Thanks, I've pasted here below too. We have seen some issues with Linux, Pulseaudio and UCM integration that have caused similar issues, I think now that @kv2019i and @juimonen have recently upstreamed all remaining UCM/Pulseaudio fixes. Any out standing guys ?
I also see you have FW version v1.4.1 (based on the incorrect version string, version v1.4.2 has been released recently so I'd recommend this too). I think it's likely that some of your ingredients (either kernel, FW, UCM, Pulseaudio) are missing some recent updates. 
```
[    2.614893] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.615075] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.030904] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.031083] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.031096] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.031323] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    3.031328] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.031484] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.903102] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.907578] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    3.907582] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.934726] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    3.934728] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    4.020258] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[    4.020259] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[    4.020483] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    4.022898] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp1
[    4.022900] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp2
[    4.022900] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp3
[    4.022901] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link Analog Playback and Capture
[    4.022901] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link Digital Playback and Capture
[    4.022902] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link dmic01
[    4.022902] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link dmic16k
[    4.023506] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:11:0
[    4.023509] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    4.024416] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    4.024417] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    4.025218] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    4.025219] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    4.026037] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    4.026039] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    4.026852] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    4.026854] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    4.026855] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    4.026856] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    4.026856] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    4.026857] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    4.027775] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    4.027779] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    4.028650] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    4.028652] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    4.029506] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    4.037538] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    4.054691] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC298: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:speaker
[    4.054693] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    4.054694] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    4.054694] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    4.054695] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    4.054696] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x18
[    4.096124] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[    4.096128] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[    4.096159] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi1 <-> iDisp1 Pin mapping ok
[    4.096161] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi2 <-> iDisp2 Pin mapping ok
[... cut ...]
[    4.101330] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input24
[    4.101356] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input25
[    4.101379] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input26
[  181.602985] snd_hda_codec_hdmi ehdaudio0D2: Unable to sync register 0x2f8100. -5
[  396.402424] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  396.492809] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

kv2019i, 2020-02-10:
@jlaham Based on this trace:
[  181.602985] snd_hda_codec_hdmi ehdaudio0D2: Unable to sync register 0x2f8100. -5

I think you are hitting same bug as:
https://github.com/thesofproject/linux/pull/1764
Fixed with following patch series sent upstream:
https://mailman.alsa-project.org/pipermail/alsa-devel/2020-February/162509.html

Did you have HDMI/DP monitor connected during boot?

PS As stated earlier in the bug, these errors are no longer related to firmware load, but are rather Linux driver issues.

jlaham, 2020-02-11:
Thanks both for your feedback.

@kv2019i I did not have anything plugged in; trying to keep my machine as "vanilla" as can be when debugging these sorts of issues.

I was to make some significant progress on this from my end, and now have a functioning system; following are my findings:
1. I can confirm that firmware is loading fine now
2. Only caveat is that every few reboots there will be one where no devices are detected, but another reboot solves the issue; not the end of the world, plus I rarely reboot unless debugging startup issues
3. This is possible now a driver issue or a udevd issue, as I was able to narrow it down to some funky interaction between PulseAudio and its reliance on udevd; commenting out use of `load-module module-udev-detect` and using static hardware detection `load-module module-detect` fixed all the weird hanging issues I was experiencing previously.

lgirdwood, 2020-02-11:
@jlaham can you attach the dmesg for 2 next time it happens.

kv2019i, 2020-02-11:
@jlaham wrote.
> 2. Only caveat is that every few reboots there will be one where no devices are detected, but another reboot solves the issue; not the end of the world, plus I rarely reboot unless debugging startup issues

Ack, and based on your previous logs, I believe this is fixed with in upstream kernel (see link to patches above). If rootcause is the same, reloading the SOF kernels driver can also help to detect the devices.  Thank you for the report!

kv2019i, 2020-02-11:
@lgirdwood @perexg Can we close this now?

lgirdwood, 2020-02-11:
@kv2019i I think so. It can be re-opened if needed.

closed 2020-02-11 by lgirdwood

Aurelien-, 2020-02-19:
Hello, I am commenting here as I have some issues with the sof driver with my C940 under manjaro gnome with kernel 5.5.4-1 or 5.6-rc2. I have never been able to make the speakers working (showing  fake output in gnome sound manager) or the mic except partially with kernel 5.4. I moved to the unstable branch to test with latest packages.

The main error in dmesg is the following:
> [    0.688205] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.688206] software IO TLB: mapped [mem 0x1ea78000-0x22a78000] (64MB)
[    2.476564] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.483455] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.483469] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    2.483688] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    2.483695] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.483886] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.290712] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.295722] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    3.295725] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.386039] sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof/sof-icl.ri failed with error -2
[    3.386043] sof-audio-pci 0000:00:1f.3: error: request firmware intel/sof/sof-icl.ri failed err: -2
[    3.386044] sof-audio-pci 0000:00:1f.3: error: failed to load DSP firmware -2
[    3.386477] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -2

I joined the full dmesg logs for kernel 5.5 and 5.6 together with the packages version installed on my machine. Hope in can help you.

Thanks for your efforts !
[dmesg-5.5.4-1.log](https://github.com/thesofproject/sof/files/4227655/dmesg-5.5.4-1.log)
[dmesg-5.6.0-1-rc2.log](https://github.com/thesofproject/sof/files/4227658/dmesg-5.6.0-1-rc2.log)
[pacman.txt](https://github.com/thesofproject/sof/files/4227660/pacman.txt)

plbossart, 2020-02-19:
> [    3.295725] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
> [    3.386039] sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof/sof-icl.ri failed with error [...]

This looks like you don't have any firmware installed on your device?

can you do a 

ls -l /lib/firmware/intel/sof*

and paste the results?

Aurelien-, 2020-02-19:
Thanks for your very quick answer ! Indeed, no sof installed, I thought it was by default... No now it seems ok from dmesg but still no speakers/mic detected by gnome...

 

> ls -l /lib/firmware/intel/sof*
> /lib/firmware/intel/sof:
> total 1992
> drwxr-xr-x 2 root root   4096 20 fév 00:20 debug
> -rw-r--r-- 1 root root 282624 23 jan 00:40 sof-apl.ri
> -rw-r--r-- 1 root root 109616 23 jan 00:40 sof-bdw.ri
> -rw-r--r-- 1 root root  88164 23 jan 00:40 sof-byt.ri
> lrwxrwxrwx 1 root root     10 23 jan 00:40 sof-cfl.ri -> sof-cnl.ri
> -rw-r--r-- 1 root root  88972 23 jan 00:40 sof-cht.ri
> lrwxrwxrwx 1 root root     10 23 jan 00:40 sof-cml.ri -> sof-cnl.ri
> -rw-r--r-- 1 root root 286720 23 jan 00:40 sof-cnl.ri
> -rw-r--r-- 1 root root 109616 23 jan 00:40 sof-hsw.ri
> -rw-r--r-- 1 root root 286720 23 jan 00:40 sof-icl.ri
> -rw-r--r-- 1 root root 115472 23 jan 00:40 sof-imx8.ri
> -rw-r--r-- 1 root root 176128 23 jan 00:40 sof-jsl.ri
> -rw-r--r-- 1 root root 151552 23 jan 00:40 sof-kbl.ri
> -rw-r--r-- 1 root root 151552 23 jan 00:40 sof-skl.ri
> -rw-r--r-- 1 root root 180224 23 jan 00:40 sof-sue.ri
> 
> /lib/firmware/intel/sof-tplg:
> total 1668
> -rw-r--r-- 1 root root 33227 23 jan 00:40 sof-apl-da7219.tplg
> -rw-r--r-- 1 root root 23311 23 jan 00:40 sof-apl-demux-pcm512x.tplg
> -rw-r--r-- 1 root root 10295 23 jan 00:40 sof-apl-dmic-2ch.tplg
> -rw-r--r-- 1 root root 10431 23 jan 00:40 sof-apl-dmic-4ch.tplg
> -rw-r--r-- 1 root root 10295 23 jan 00:40 sof-apl-dmic-a2ch-b2ch.tplg
> -rw-r--r-- 1 root root 10363 23 jan 00:40 sof-apl-dmic-a2ch-b4ch.tplg
> -rw-r--r-- 1 root root  5223 23 jan 00:40 sof-apl-dmic-a2ch.tplg
> -rw-r--r-- 1 root root 10363 23 jan 00:40 sof-apl-dmic-a4ch-b2ch.tplg
> -rw-r--r-- 1 root root  5223 23 jan 00:40 sof-apl-dmic-b2ch.tplg
> -rw-r--r-- 1 root root  6743 23 jan 00:40 sof-apl-eq-dmic.tplg
> -rw-r--r-- 1 root root  7223 23 jan 00:40 sof-apl-eq-pcm512x.tplg
> -rw-r--r-- 1 root root  9019 23 jan 00:40 sof-apl-keyword-detect.tplg
> -rw-r--r-- 1 root root 38003 23 jan 00:40 sof-apl-nocodec.tplg
> -rw-r--r-- 1 root root  5127 23 jan 00:40 sof-apl-pcm512x-nohdmi.tplg
> -rw-r--r-- 1 root root 25031 23 jan 00:40 sof-apl-pcm512x.tplg
> -rw-r--r-- 1 root root 19815 23 jan 00:40 sof-apl-rt298.tplg
> -rw-r--r-- 1 root root  5791 23 jan 00:40 sof-apl-src-50khz-pcm512x.tplg
> -rw-r--r-- 1 root root 11623 23 jan 00:40 sof-apl-src-dmic.tplg
> -rw-r--r-- 1 root root  5791 23 jan 00:40 sof-apl-src-pcm512x.tplg
> -rw-r--r-- 1 root root 43619 23 jan 00:40 sof-apl-tdf8532.tplg
> -rw-r--r-- 1 root root  5127 23 jan 00:40 sof-apl-wm8804.tplg
> -rw-r--r-- 1 root root 12579 23 jan 00:40 sof-bdw-rt286.tplg
> -rw-r--r-- 1 root root 12579 23 jan 00:40 sof-bdw-rt5640.tplg
> -rw-r--r-- 1 root root 13371 23 jan 00:40 sof-byt-da7213.tplg
> -rw-r--r-- 1 root root 13107 23 jan 00:40 sof-byt-nocodec.tplg
> -rw-r--r-- 1 root root 13371 23 jan 00:40 sof-byt-rt5640.tplg
> -rw-r--r-- 1 root root 13371 23 jan 00:40 sof-byt-rt5645.tplg
> -rw-r--r-- 1 root root 13371 23 jan 00:40 sof-byt-rt5651.tplg
> -rw-r--r-- 1 root root 13371 23 jan 00:40 sof-byt-rt5682.tplg
> -rw-r--r-- 1 root root 13371 23 jan 00:40 sof-cht-max98090.tplg
> -rw-r--r-- 1 root root 13107 23 jan 00:40 sof-cht-nocodec.tplg
> -rw-r--r-- 1 root root  7927 23 jan 00:40 sof-cht-rt5682.tplg
> -rw-r--r-- 1 root root  5555 23 jan 00:40 sof-cht-src-50khz-pcm512x.tplg
> -rw-r--r-- 1 root root 34187 23 jan 00:40 sof-cml-demux-rt5682-max98357a.tplg
> -rw-r--r-- 1 root root 29283 23 jan 00:40 sof-cml-demux-rt5682.tplg
> -rw-r--r-- 1 root root 39787 23 jan 00:40 sof-cml-rt1011-rt5682.tplg
> -rw-r--r-- 1 root root 34883 23 jan 00:40 sof-cml-rt5682-kwd.tplg
> -rw-r--r-- 1 root root 39787 23 jan 00:40 sof-cml-rt5682-max98357a.tplg
> -rw-r--r-- 1 root root 31123 23 jan 00:40 sof-cml-rt5682.tplg
> -rw-r--r-- 1 root root 32719 23 jan 00:40 sof-cml-rt700.tplg
> -rw-r--r-- 1 root root 34411 23 jan 00:40 sof-cml-rt711-rt1308-mono-rt715.tplg
> -rw-r--r-- 1 root root 39295 23 jan 00:40 sof-cml-rt711-rt1308-rt715.tplg
> -rw-r--r-- 1 root root 26751 23 jan 00:40 sof-cml-src-rt5682.tplg
> -rw-r--r-- 1 root root 25715 23 jan 00:40 sof-cnl-nocodec.tplg
> -rw-r--r-- 1 root root 11335 23 jan 00:40 sof-cnl-rt274.tplg
[... 21 lines cut ...]
> -rw-r--r-- 1 root root 25647 23 jan 00:40 sof-tgl-nocodec.tplg
> -rw-r--r-- 1 root root 25071 23 jan 00:40 sof-tgl-rt711-rt1308-nohdmi.tplg
> -rw-r--r-- 1 root root 44607 23 jan 00:40 sof-tgl-rt711-rt1308.tplg
> -rw-r--r-- 1 root root 29283 23 jan 00:40 sof-whl-demux-rt5682.tplg
> -rw-r--r-- 1 root root 34883 23 jan 00:40 sof-whl-rt5682-kwd.tplg
> -rw-r--r-- 1 root root 31123 23 jan 00:40 sof-whl-rt5682.tplg

> [    0.687162] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.687163] software IO TLB: mapped [mem 0x1ea78000-0x22a78000] (64MB)
[    2.807355] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.807732] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.807745] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    2.808024] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    2.808032] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.808214] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.575842] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.580196] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[... cut ...]
[    4.617112] sof-audio-pci 0000:00:1f.3: error: codec #2 probe error, ret: -5
[    4.618286] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    4.618288] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    4.717911] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[    4.717912] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[    4.718129] sof-audio-pci 0000:00:1f.3: firmware boot complete
[dmesg-5.5.4-1-sof.log](https://github.com/thesofproject/sof/files/4227799/dmesg-5.5.4-1-sof.log)

Aurelien-, 2020-02-19:
Ok I had to reboot a second time as hinted in a previous comment and it is now working ! Sorry for having polluted the issue... Thanks for the help !

Enerccio, 2020-04-22:
It happened again. Unfortunately, I didn't know what to do with sof-logger and hardlocked myself with save state options so I lost dmesg. But it seems it happens every time I leave computer idle for long time (night sleep). Will check if it happens again today, but can you tell me what exactly I need to do with that sof-logger?

buzuddha, 2020-06-04:
I'm having issues here too. I'm new to arch. As of now I have the left speaker (weak at max volume) and the mic appears to work. See attached for how far I've gotten. 

I'd be very grateful for any guidance. Thanks for your time and consideration!

[sof-troubleshoot.txt](https://github.com/thesofproject/sof/files/4731155/sof-troubleshoot.txt)

danielcherubini, 2020-06-04:
https://wiki.archlinux.org/index.php/Lenovo_Yoga_c940#Linux_5.5_and_later

buzuddha, 2020-06-04:
So as far as I can tell I've done what is described in the wiki. 
1) installed sof-firmware using pacman
2) blacklisted snd_hda_intel and snd_soc_skl
3) added the lines to load alsa modules

paulmenzel, 2020-06-04:
@buzuddha, this issue is closed, and your issues is probably something else. Please create a new issue.

aleksfadini, 2020-06-21:
> So as far as I can tell I've done what is described in the wiki.
> 
> 1. installed sof-firmware using pacman
> 2. blacklisted snd_hda_intel and snd_soc_skl
> 3. added the lines to load alsa modules

I'm having the same issues, with the Cannonlake C940. Could it have to do with our version of the C940 that is different?

aleksfadini, 2020-06-21:
> @buzuddha, this issue is closed, and your issues is probably something else. Please create a new issue.

I am another user having the same issue, I think our Lenovo Yoga c940 being the CannonLake version might have different hardware.
I opened this issue
https://github.com/thesofproject/linux/issues/2020#issuecomment-638265496