thesofproject/sof #2135: [BUG][BYT][RT5682]The output frequency is very slow when aplay via media palyback(0,1)

Liviali155, 2019-11-25:
**Describe the bug**
The output frequency is very slow when aplay via media palyback(0,1)
It is normally when aplay via (0,0)
It seems a firmware regression issue

**To Reproduce**
1.sudo reboot with headset insert
2."aplay -Dhw:0,1 -r 48000 -c2 wav/44California_Dreaming.wav -vv -i" to aplay  audio

```ubuntu@sh-byt-mb-rt5682-1111:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofrt5682 [sof-rt5682], device 0: Low Latency (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 1: Media Playback 1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
**Reproduction Rate**
5/5

**Expected behavior**
Output normally when aplay via media playback(0,1)

**Impact**
The output is very slow when aplay via media playback
It is normally when aplay via (0,0)

**Environment**
platform: BYT MB with  RT5682 in I2S mode
Firmware: d6fe2b13 https://github.com/thesofproject/sof/commits/master
Kernel: b090cc8a https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-byt-rt5682.tplg same as firmware

There is no dmesg error log
<details>
 <summary>sof_errortrace</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1           SA         184477491.093750 184477488.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 32881
    0      1           SA         184482507.708333      5016.614746      src/lib/agent.c:55  	validate(), ll drift detected, delta = 29200
    0      1           SA         184497484.010417     14976.301758      src/lib/agent.c:55  	validate(), ll drift detected, delta = 41745
    0      1           SA         184503518.541667      6034.531250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 32607
    0      1           SA         184508486.041667      4967.500000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 27984
    0      1           SA         184518507.239583     10021.198242      src/lib/agent.c:55  	validate(), ll drift detected, delta = 45391
    0      1           SA         184523511.614583      5004.375000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 41475
    0      1           SA         184529514.114583      6002.500000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 31723
    0      1           SA         184534442.031250      4927.916504      src/lib/agent.c:55  	validate(), ll drift detected, delta = 26338
    0      1           SA         184544518.020833     10075.989258      src/lib/agent.c:55  	validate(), ll drift detected, delta = 44798
    0      1           SA         184550476.927083      5958.906250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34208
    0      1           SA         184555500.520833      5023.593750      src/lib/agent.c:55  	validate(), ll drift detected, delta = 30662
    0      1           SA         184570426.250000     14925.729492      src/lib/agent.c:55  	validate(), ll drift detected, delta = 42236
    0      1           SA         184576513.281250      6087.031250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34107
    0      1           SA         184581488.593750      4975.312500      src/lib/agent.c:55  	validate(), ll drift detected, delta = 29632
    0      1           SA         184596497.083333     15008.489258      src/lib/agent.c:55  	validate(), ll drift detected, delta = 42795
    0      1           SA         184602441.354167      5944.270996      src/lib/agent.c:55  	validate(), ll drift detected, delta = 31925
    0      1           SA         184607519.218750      5077.864746      src/lib/agent.c:55  	validate(), ll drift detected, delta = 29421
    0      1           SA         184622406.666667     14887.448242      src/lib/agent.c:55  	validate(), ll drift detected, delta = 40259
    0      1           SA         184628536.093750      6129.427246      src/lib/agent.c:55  	validate(), ll drift detected, delta = 32944
    0      1           SA         184633499.739583      4963.645996      src/lib/agent.c:55  	validate(), ll drift detected, delta = 28247
    0      1           SA         184643434.895833      9935.156250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 44002
    0      1           SA         184648504.791667      5069.895996      src/lib/agent.c:55  	validate(), ll drift detected, delta = 41344
    0      1           SA         184654506.666667      6001.875000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 31580
    0      1           SA         184659536.614583      5029.947754      src/lib/agent.c:55  	validate(), ll drift detected, delta = 28154
    0      1           SA         184669508.958333      9972.343750      src/lib/agent.c:55  	validate(), ll drift detected, delta = 44624
    0      1           SA         184675506.302083      5997.343750      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34772
    0      1           SA         184680418.125000      4911.822754      src/lib/agent.c:55  	validate(), ll drift detected, delta = 29080
    0      1           SA         184685510.781250      5092.656250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 26859
    0      1           SA         184695507.031250      9996.250000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 43787
    0      1           SA         184700644.062500      5137.031250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 42416
    0      1           SA         184706567.343750      5923.281250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 31144
    0      1           SA         184711624.322917      5056.979004      src/lib/agent.c:55  	validate(), ll drift detected, delta = 28239
    0      1           SA         184721608.437500      9984.114258      src/lib/agent.c:55  	validate(), ll drift detected, delta = 44934
    0      1           SA         184726635.416667      5026.979004      src/lib/agent.c:55  	validate(), ll drift detected, delta = 41452
    0      1           SA         184732529.166667      5893.750000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 29611
    0      1           SA         184747655.885417     15126.718750      src/lib/agent.c:55  	validate(), ll drift detected, delta = 45045
    0      1           SA         184753606.770833      5950.885254      src/lib/agent.c:55  	validate(), ll drift detected, delta = 34301
    0      1           SA         184430499.843750              nan      src/lib/agent.c:55  	validate(), ll drift detected, delta = 30649
    0      1           SA         184435520.625000      5020.781250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 27048
    0      1           SA         184445514.791667      9994.166992      src/lib/agent.c:55  	validate(), ll drift detected, delta = 43936
    0      1           SA         184451442.031250      5927.239746      src/lib/agent.c:55  	validate(), ll drift detected, delta = 32739
    0      1           SA         184456517.500000      5075.468750      src/lib/agent.c:55  	validate(), ll drift detected, delta = 30187
````
</details>

[dmesg20191125.log](https://github.com/thesofproject/sof/files/3885109/dmesg20191125.log)
[sof_trace.log](https://github.com/thesofproject/sof/files/3885110/sof_trace.log)

lgirdwood, 2019-11-25:
@Liviali155 can you play a sine wave of a fixed frequency and attach the capture. How long does a 5 second wave file take to play (aplay -d 5 -f dat /dev/zero) ?

Liviali155, 2019-11-26:
> aplay -d 5 -f dat /dev/zero

I aplay sine_48k_stereo_440.wav via (0,0) and (0,1)media playback,the capture file are attachments
as the capture show,aplay sine wave via (0,0) takes about 10s,but aplay sine wav via (0,1)mediaplayback takes about 13s
Aplay via (0,1) media playback
[image]

Aplay via (0,0)
[image]

[sine_48k_stereo_440.zip](https://github.com/thesofproject/sof/files/3889630/sine_48k_stereo_440.zip)

[aplay(0,0)capturefile.zip](https://github.com/thesofproject/sof/files/3889621/aplay.0.0.capturefile.zip)
[mediaplaybackcapturefile.zip](https://github.com/thesofproject/sof/files/3889626/mediaplaybackcapturefile.zip)

Liviali155, 2019-11-26:
Bi-sect result: Caused by firmwire commit [ee9283c1](https://github.com/thesofproject/sof/commit/ee9283c159649e4248e4188fe0b47b551e22342c)

Liviali155, 2019-11-26:
On BYT MB nocodec can reproduce this issue
platform: BYT MB with nocodec
Firmware: db7671ac https://github.com/thesofproject/sof/commits/master
Kernel: 58ed3272 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-byt-nocodec.tplg  same as firmware

paulstelian97, 2019-11-26:
@jajanusz Did I do something wrong with my commit? Does BYT not define one of those yet is still expected to use the SSP clock? Or, did it define some other clock as SSP, breaking the assumption I had when I made that change?

paulstelian97, 2019-11-26:
Further reports: BYT defines CLK_SSP but not NOTIFIER_ID_SSP_FREQ, so it goes for the CPU_FREQ notifier ID. If that seems to be incorrect, the following patch may be attempted:

```diff
diff --git a/src/include/sof/lib/notifier.h b/src/include/sof/lib/notifier.h
index 45fc5efda..4514ded49 100644
--- a/src/include/sof/lib/notifier.h
+++ b/src/include/sof/lib/notifier.h
@@ -45,7 +45,7 @@ struct notifier {
        void (*cb)(int message, void *cb_data, void *event_data);
 };
 
-#if defined(CLK_SSP) && defined(NOTIFIER_ID_SSP_FREQ)
+#ifdef CLK_SSP
 #define NOTIFIER_CLK_CHANGE_ID(clk) \
        ((clk) == CLK_SSP ? NOTIFIER_ID_SSP_FREQ : NOTIFIER_ID_CPU_FREQ)
 #else
```

If this works I shall send a PR with this fix. I made the condition more complex because I believed that using NOTIFIER_ID_SSP_FREQ without defining it would be an error.

paulstelian97, 2019-11-26:
The above patch was applied in my pull request #2141. Can you check if it fixes this issue? @Liviali155

lgirdwood, 2019-11-26:
@fredoh9 stange CI did not pick this up ? We should have a test that plays 5 seconds of audio and then checks the result does indeed take 5 seconds to play. @wenqingfu fyi

Liviali155, 2019-11-27:
> The above patch was applied in my pull request #2141. Can you check if it fixes this issue? @Liviali155

I tried with #2141,error not seen

lgirdwood, 2019-11-27:
#2141 now merged.

closed 2019-11-27 by lgirdwood