thesofproject/sof #2136: [BUG] DSP FW load fails across boots on CML

sathyap-chrome, 2019-11-25:
**Describe the bug**
The issue is seen on couple of chrome systems.
What tests was run to reach the error state is unknown.
But at current state, DSP FW load fails and its not recovered even after cold reboot.
Only when we reflash Coreboot, the FW load is successful.

**Reproduction Rate**
Quite rare, but when happens its persistent

**Expected behavior**
DSP FW load failure should not occur. Even when error out, reboot should fix the same.

**Impact**
once issue happens, audio is not usable, so its severity is high

**Environment**
1)  SOF: CML -007 branch

**Screenshots or console output**
Attached the dmesg logs with verbose traces from failed and pass case.

[dmesg_dsp_fail.txt](https://github.com/thesofproject/sof/files/3885547/dmesg_dsp_fail.txt)
[dmesg_dsp_pass.txt](https://github.com/thesofproject/sof/files/3885548/dmesg_dsp_pass.txt)

sathyap-chrome, 2019-11-25:
@mengdonglin checked and saw the below trace hinting at rom not responding:
[    5.816760] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000

Request FW team for more analysis and also next steps that might help debug further

lgirdwood, 2019-11-25:
@sathyap-chrome if reflashing BIOS fixes this then it's possible something is corrupting parts of the coreboot (or other stitched in FWs) responsible for device boot/config (including DSP boot/config needed by DSP ROM).

sathyap-chrome, 2019-11-25:
@lgirdwood yes, we are looking at the same too. Wanted to raise issue to get more debug help.
Is there any other debug we can attempt to point to narrow down the issue.

We are currently suspecting ME being in bad state.

Also we are looking at what sequence can bring to this state.

hsanjuan, 2019-11-25:
I started having a very similar error using sof-cnl-v1.4.1.ri on Kernel 5.4. It seems I cannot revert either to the previous setup on Kernel 5.3 and SOF v1.3. While there seems to be more progress loading the drivers, I get an unending stream of:

```
[  135.380022] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  135.380025] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[  135.380028] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
```

hsanjuan, 2019-11-25:
> I started having a very similar error using sof-cnl-v1.4.1.ri on Kernel 5.4. It seems I cannot revert either to the previous setup on Kernel 5.3 and SOF v1.3. While there seems to be more progress [...]

Nevermind, this seems to have been caused by a mismatch in the topology

mengdonglin, 2019-11-26:
Comments from @sathyap-chrome: Found issue to be related to ME region being in bad state.

sathyap-chrome, 2019-11-28:
Chrome team working with coreboot team . Hence closing this issue.

closed 2019-11-28 by sathyap-chrome