thesofproject/sof #2143: [BUG][BYT][RT5682]Dmesg show "error : DSP panic!" when doing volume change without playback/capture

Liviali155, 2019-11-27:
**Describe the bug**
Run "test-vol.sh" to do  volume change without playback/capture -1000 times
After Done run  "test-vol.sh"  again,Dmesg show "error : DSP panic!"
Then aplay/arecord can not work, Error not seen  after sudo reboot system

 
**To Reproduce**
1.sudo reboot
2.Run "test-vol.sh" to do  volume change without playback/capture -1000 times
3.After done run "test-vol.sh" again

**Reproduction Rate**
5/5

**Expected behavior**
Volume can be changed normally

**Impact**
Dmesg show "error : DSP panic!"

**Environment**
platform: BYT MB with RT5682 in I2S mode
Firmware: d740a3e0 https://github.com/thesofproject/sof/commits/master
Kernel: c6423829 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-byt-rt5682.tplg same as firmware

**Screenshots or console output**
<details>
 <summary>dmesg_error.log</summary>

````
 [  915.224248] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  915.426351] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  915.426540] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  915.518935] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  915.520027] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  915.613007] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  915.613399] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  915.705501] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  915.706073] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  915.907299] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  915.907510] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  915.936002] sof-audio-acpi 80860F28:01: error : DSP panic!
[  915.936021] sof-audio-acpi 80860F28:01: panic: dsp_oops_offset 1329152 offset 1329152
[  915.936086] sof-audio-acpi 80860F28:01: error: can't enter idle
[  915.936097] sof-audio-acpi 80860F28:01: error: trace point 00000000
[  915.936105] sof-audio-acpi 80860F28:01: error: panic at src/lib/agent.c:50
[  915.936112] sof-audio-acpi 80860F28:01: error: DSP Firmware Oops
[  915.936121] sof-audio-acpi 80860F28:01: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060525 SAR     0x00000000
[  915.936130] sof-audio-acpi 80860F28:01: EPC1     0x00000000 EPC2     0x00000000 EPC3     0x00000000 EPC4    0xff2cf79b
[  915.936145] sof-audio-acpi 80860F28:01: EPC5     0xff2cf79b EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  915.936154] sof-audio-acpi 80860F28:01: EPS2     0x00000000 EPS3     0x00000000 EPS4     0x00060120 EPS5    0x00060120
[  915.936163] sof-audio-acpi 80860F28:01: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x000000a2
[  915.936170] sof-audio-acpi 80860F28:01: stack dump from 0xff306d00
[  915.936181] sof-audio-acpi 80860F28:01: 0xff306d00: ff306d20 00000000 00060525 4bcefed8
[  915.936192] sof-audio-acpi 80860F28:01: 0xff306d04: 00000000 00000000 00000032 006acf93
[  915.936206] sof-audio-acpi 80860F28:01: 0xff306d08: 44461300 2e23ef08 398db828 ffff897e
[  915.936217] sof-audio-acpi 80860F28:01: 0xff306d0c: 398db828 ffff897e 00000000 00000000
[  915.936227] sof-audio-acpi 80860F28:01: 0xff306d10: 3243c800 ffff897e 90cbbcc0 ffffffff
[  915.936240] sof-audio-acpi 80860F28:01: 0xff306d14: 00000000 00000000 00000000 00000000
[  915.936250] sof-audio-acpi 80860F28:01: 0xff306d18: 3243c800 ffff897e 90cbbe50 ffffffff
[  915.936260] sof-audio-acpi 80860F28:01: 0xff306d1c: 00000000 00000000 00000000 00000000
[  915.936274] sof-audio-acpi 80860F28:01: error: ipc host -> DSP: pending no complete no raw 0x0
[  915.936282] sof-audio-acpi 80860F28:01: error: mask host: pending yes complete no raw 0x2
[  915.936290] sof-audio-acpi 80860F28:01: error: ipc DSP -> host: pending no complete no raw 0x800008000dead009
[  915.936300] sof-audio-acpi 80860F28:01: error: mask DSP: pending no complete no raw 0x0
[  915.936307] sof-audio-acpi 80860F28:01: error: waking up any trace sleepers
[  915.936333] sof-audio-acpi 80860F28:01: error: trace IO error   
	
````
[... 31 lines cut ...]
    0      1       VOLUME 2.14    645752177.604167         9.843750 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     648960425.989583   3208248.500000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     648960435.677083         9.687500 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     649928239.322917    967803.625000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     649928249.010417         9.687500 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 2.14    652054414.895833   2126166.000000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 2.14    652054424.687500         9.791667 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 2.14    654946535.468750   2892110.750000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 2.14    654946545.312500         9.843750 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    659918990.468750   4972445.000000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    659919000.000000         9.531250 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.3     662628135.572917   2709135.500000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.3     662628145.364583         9.791667 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     669830884.062500   7202738.500000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.2     669830893.750000         9.687500 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    670605824.635417    774930.875000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 3.22    670605834.166667         9.531250 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 1.3     671876351.562500   1270517.375000 udio/volume/volume.c:284 	volume_set_chan: Limited request 0 to VOL_MIN. 
	
````
</details>

[dmesg20191127.log](https://github.com/thesofproject/sof/files/3895574/dmesg20191127.log)
[sof_errortrace.log](https://github.com/thesofproject/sof/files/3895575/sof_errortrace.log)
[sof_trace.log](https://github.com/thesofproject/sof/files/3895576/sof_trace.log)

Liviali155, 2019-11-27:
Bi-sect result: Caused by firmwire commit ee9283c1
@paulstelian97 please check it

paulstelian97, 2019-11-27:
@Liviali155 Same as the other issue I think, can you check with #2141 applied? That should return to the old behaviour on BYT.

lgirdwood, 2019-11-27:
#2141 applied.

closed 2019-11-27 by lgirdwood