thesofproject/sof #2144: [BUG][CFL]DMIC pause-release cause ipc timed out for 0x60060000 size 12

dengyangchao, 2019-11-27:
**Describe the bug**
Run pause-release stress test, and occur ipc timed out for 0x60060000 size 12, pause-release interval time 0.5s. 
when issue occur aplay and paplay can't work only reboot can resume.

**To Reproduce**
1.reboot system via"sudo reboot"
2.wait runtime PM status to suspend
3.run stress test via command ```./sof-test-pause-release.sh -t pcm6c```

**Reproduction Rate**
under 10%

tried 3 times
1st occur at 48th
2nd occur at 34th
3rd occur 43th

**Expected behavior**
pause-release normal

**Impact**
occur ipc timed out

**Environment**
Platform: CFL-S RVP with onboard codec ALC700 in HDA mode
Firmware: d740a3e0 https://github.com/thesofproject/sof/commits/master
Kernel: c6423829 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-4ch.tplg same as firmware

<details>
 <summary>Dmesg</summary>

````
[  168.132466] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  168.132501] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  168.467581] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 3
[  168.467588] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  168.467829] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  168.468368] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[  168.632342] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  168.632364] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  168.956066] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 4
[  168.957049] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[  168.957051] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[  168.957336] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[  169.560076] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 3
[  169.560083] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  170.067786] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60060000 size 12
[  170.067793] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  170.067802] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  170.067972] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  170.067988] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000003 rirb 00
[  170.067991] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  170.068010] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  170.068014] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  170.068590] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[  170.068601]  DMIC48kHz: ASoC: trigger FE cmd: 3 failed: -110
[  170.068611] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 4
[  170.069138] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[  170.069146] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[  170.571788] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60070000 size 12
[  170.571794] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  170.571802] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  170.571975] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  170.571992] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000003 rirb 00
[  170.571995] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  170.572014] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  170.572017] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  170.572022]  DMIC48kHz: ASoC: trigger FE cmd: 4 failed: -110
[  170.572302] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 3
[... 82 lines cut ...]
    0      1           SA         191173239.062500   1251359.375000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 27913
    0      1           SA         192424455.208333   1251216.125000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 27282
    0      1           SA         193674373.125000   1249917.875000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25690
    0      1           SA         193675690.000000      1316.875000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25287
    0      1           SA         194925714.427083   1250024.375000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 27445
    0      1           SA         196176952.395833   1251238.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 27215
    0      1           SA         197426903.697917   1249951.250000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 26274
    0      1           SA         198678246.562500   1251342.875000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 28060
    0      1           SA         199929451.927083   1251205.375000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 27176
    0      1           SA         201179386.822917   1249934.875000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25954
    0      1           SA         202429448.489583   1250061.625000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 27134
    0      1           SA         203681987.187500   1252538.750000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 27894
    0      1           SA         204931887.031250   1249899.875000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25960
    0      1           SA         206181944.479167   1250057.500000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 27056
    0      1           SA         207434490.520833   1252546.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 27956
    0      1           SA         208684391.145833   1249900.625000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 26036
    0      1           SA         209934438.229167   1250047.125000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 26937

````
</details>

[dmesg.log](https://github.com/thesofproject/sof/files/3895646/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3895647/logger.log)
[logger-error.log](https://github.com/thesofproject/sof/files/3895648/logger-error.log)
[sof-test-pause-release.txt](https://github.com/thesofproject/sof/files/3895650/sof-test-pause-release.txt)

dengyangchao, 2019-11-27:
Seems similar with #2044

lgirdwood, 2019-11-27:
@dengyangchao I assume this only happens with DMIC and that SSP/HDA are OK ?

dengyangchao, 2019-11-27:
> @dengyangchao I assume this only happens with DMIC and that SSP/HDA are OK ?

@lgirdwood At present, the issue only observed on CFL-S RVP's DMIC.

lgirdwood, 2019-11-27:
@dengyangchao ok, thanks. @singalsu @tlauda looks like DMIC driver issue on pause release.

singalsu, 2019-11-27:
I don't get IPC timeouts with WHL. Is it happening with other platforms than CFL?

dengyangchao, 2019-11-28:
> I don't get IPC timeouts with WHL. Is it happening with other platforms than CFL?

At present, only reproduced the issue on CFL. Other platforms can't reproduced

dengyangchao, 2019-11-28:
Caused by commit e0491ff , and only use xcc-compiler generate sof version can reproduce the issue.
@tlauda please help to check

lgirdwood, 2019-11-28:
Related to https://github.com/thesofproject/sof/issues/2146 same bisect failure.

sinahuang, 2019-12-02:
Issue still can be reproduced on APL UP2 with nocodec.

**Test recipe**
platform: APL UP2 with nocodec.
Firmware: feb9e2b https://github.com/thesofproject/sof/commits/master
Kernel: 0bdbe73a https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-apl-nocodec.tplg same as firmware

<details>
 <summary>dmesg error log</summary>

````
[  230.358455] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  230.359032] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20240000 successful
[  230.391797] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 4
[  230.392206] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x2024001e successful
[  230.392228] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[  230.898419] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60070000 size 12
[  230.898439] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[  230.898448] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  230.898500] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  230.898517] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000003 rirb 00
[  230.898528] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  230.898540] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  230.898551] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  230.898567]  Port0: ASoC: trigger FE cmd: 4 failed: -110
[  230.898587] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 3
[  230.898597] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  231.402255] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60060000 size 12
[  231.402277] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[  231.402287] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  231.402341] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  231.402358] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000003 rirb 00
[  231.402369] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  231.402383] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  231.402395] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  231.403193] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20240000 successful
[  231.403228]  Port0: ASoC: trigger FE cmd: 3 failed: -110

````
</details>

dengyangchao, 2019-12-02:
Issue also can be reproduce on CML Chrome RT5682 when pause-release arecord via HW:0,0

Test recipe
platform: CML Chrome with onboard codec RT5682 in I2S mode
Firmware: feb9e2b https://github.com/thesofproject/sof/commits/master
Kernel: 0bdbe73a https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-cml-rt5682-max98357a.tplg same as firmware

```
root@sh-cml-hatch-rt5682-1234:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofrt5682 [sof-rt5682], device 0: Port1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 2: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 3: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 4: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 5: Speakers (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
```
   87.866272] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[   87.866545] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[   88.473329] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 3
[   88.473337] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[   88.981061] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60060000 size 12
[   88.981069] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[   88.981080] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   88.981172] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   88.981184] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000003 rirb 00
[   88.981188] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   88.981197] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   88.981200] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   88.981212] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[   88.981220]  Port1: ASoC: trigger FE cmd: 3 failed: -110
[   88.981230] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 4
```

[dmesg.log](https://github.com/thesofproject/sof/files/3910352/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3910353/logger.log)
[logger-error-trace.log](https://github.com/thesofproject/sof/files/3910354/logger-error-trace.log)

dengyangchao, 2019-12-12:
Issue can reproduce on CML notebook with ALC3204 HDA codec, when do DMIC (HW:0,6) pause-release stress test.

Firmware: 98d7ab21 https://github.com/thesofproject/sof/commits/master
Kernel: 6714c4e1 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-2ch.tplg same as firmware

mengdonglin, 2019-12-13:
This issue is to be worked around by https://github.com/thesofproject/sof/pull/2191

Liviali155, 2019-12-13:
> @dengyangchao This issue is to be worked around by #2191 of @slawblauciak

Yes,Tried with  #2191,error not seen on below platforms:
CFL-S RVP with onboard codec ALC700 in HDA mode
APL UP2 with nocodec

Firmware: f5bcc4b https://github.com/thesofproject/sof/commits/master
Kernel: 9c0401ca https://github.com/thesofproject/linux/commits/topic/sof-dev

lgirdwood, 2019-12-16:
#2191 now merged, @singalsu @dengyangchao I've updated issue to reflect new root cause.

dengyangchao, 2019-12-17:
Issue can't be reproduced since #2191 merged, close.

closed 2019-12-17 by dengyangchao