thesofproject/sof #2145: [BUG] Build fail with RF-2016.4 for CNL

singalsu, 2019-11-27:
**Describe the bug**
```
[ 69%] Building C object CMakeFiles/sof.dir/src/audio/pcm_converter/pcm_converter_hifi3.c.o
/home/singalsu/tmp/sof/src/audio/pcm_converter/pcm_converter_hifi3.c: In function ‘pcm_convert_s24_to_s16’:
/home/singalsu/tmp/sof/src/audio/pcm_converter/pcm_converter_hifi3.c:169: error: perhaps a TIE conversion macro (from int to ae_int16x4) is missing
/home/singalsu/tmp/sof/src/audio/pcm_converter/pcm_converter_hifi3.c:169: error: incompatible type for argument 1 of ‘AE_S16_0_XC’
```

**To Reproduce**
Build for CNL with configuration X6H3CNL_2016_4_linux and xt-xcc version RF-2016.4

**Reproduction Rate**
Always

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: N/A
    * SOF: d740a3e068eae57ecf08df215ee19102b54981ee
2) Name of the topology file
    * Topology: N/A
3) Name of the platform(s) on which the bug is observed.
    * Platform: CNL

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

lbetlej, 2019-11-27:
@singalsu Is the issue reproduced with RG-2017.8 compiler ? I recommend you to update toolchain to RG-2017.8.

singalsu, 2019-11-27:
> @singalsu Is the issue reproduced with RG-2017.8 compiler ? I recommend you to update toolchain to RG-2017.8.

I just installed RG-2017.8 and it builds OK.

Should I just close this issue?

lgirdwood, 2019-11-27:
@singalsu is xtensa-build.sh updated to use the new CC ?

singalsu, 2019-11-27:
@lgirdwood Yes, it is up to date.

closed 2019-11-27 by lgirdwood

paulstelian97, 2019-11-28:
@singalsu i.MX8 also is on 2016.4 and I'm not sure I can update. @dbaluta Does NXP have a newer toolchain we can use?

paulstelian97, 2019-11-28:
@lgirdwood Will send scripts update for i.MX8 when the compiler is readily available in NXP.

@mihaix Be aware that this issue affects us and RG-2017.8 fixes it for us as well.

dbaluta, 2019-11-28:
@singalsu why don't you upgrade to a newer version? We seen RI-2019.1 out there.

lgirdwood, 2019-11-28:
@paulstelian97 @dbaluta is it worth adding a compiler.h that will reject building certain platforms if the compiler is a certain version (like Linux does ?)

singalsu, 2019-11-28:
@dbaluta Personally I'd prefer newest compiler too (should have advantage in code speed etc.) but since our CI and official builds use some long time ago decided version I've installed those.

@lgirdwood I've got myself similar build errors too. Build for CNL has been more picky with the types for intrinsics. It should be a simple fix for pcm_converter_hifi3.c to avoid this, possibly just add a manual type cast when compiler does not know what to do with various AE integer types.