thesofproject/sof #2146: [BUG][BYT][RT5682]"ipc timed out for 0x60050000 size 12" when aplay via media playback(0,4)

Liviali155, 2019-11-27:
**Describe the bug**
"ipc timed out for 0x60050000 size 12" when aplay via media playback(0,4)
After that aplay can not work,Recovery after sudo reboot system

It seems like a firmware regression

**To Reproduce**
1.Sudo reboot with headset insert
2.aplay -Dhw:0,4 -fs16_le -c2 -r 48000 -vv -i wav/48California_Dreaming.wav

```
ubuntu@cd-apl-up2-pcm512x-07:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 0: Port5 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 1: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 2: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 3: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 4: Media Playback 4 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
**Reproduction Rate**
5/5

**Expected behavior**
Aplay normally via media playback(0,4)

**Impact**
I/O error,dmesg show "ipc timed out for 0x60050000 size 12"

<details>
 <summary>dmesg_error.log</summary>

````
[   95.573109] sof-audio-pci 0000:00:0e.0: periods:4
[   95.573119] sof-audio-pci 0000:00:0e.0: stream_tag 1
[   95.573128] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   95.573505] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   95.573510] sof-audio-pci 0000:00:0e.0: pcm: stream dir 0, posn mailbox offset is 790528
[   95.588111] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 4 dir 0 cmd 1
[   95.588416] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x14001e successful
[   95.588423] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   95.588824] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   96.699016] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 4 dir 0 cmd 0
[   96.699029] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   97.202566] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60050000 size 12
[   97.202580] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[   97.202590] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[   97.202639] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[   97.202651] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
[   97.202657] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   97.202665] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   97.202672] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[   97.202691] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20140000 successful
[   97.202699]  Media Playback 4: ASoC: trigger FE cmd: 0 failed: -110
[   97.202740] sof-audio-pci 0000:00:0e.0: pcm: free stream 4 dir 0
[   97.202750] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   97.202799] sof-audio-pci 0000:00:0e.0: error: trace IO error
[   97.706450] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60030000 size 12
[   97.706457] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[   97.706463] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[   97.706505] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[   97.706512] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   97.706514] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   97.706518] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   97.706521] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[   97.720180] sof-audio-pci 0000:00:0e.0: pcm: close stream 4 dir 0    
	
	
````
</details>

[dmesg20191127.log](https://github.com/thesofproject/sof/files/3895805/dmesg20191127.log)
[sof_trace20191127.log](https://github.com/thesofproject/sof/files/3895806/sof_trace20191127.log)

**Environment**
platform: APL UP2 with PCM512x in I2S mode
Firmware: d740a3e https://github.com/thesofproject/sof/commits/master
Kernel: c6423829 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-apl-pcm512x.tplg  same as firmware

**Screenshots or console output**

paulstelian97, 2019-11-27:
@Liviali155 I think that is also with my problematic commit (all BYT bugs should suspect it's due to that until #2141 is merged -- I really messed up with that one, sorry). Is it because of that/does my PR fix it?

lgirdwood, 2019-11-27:
@Liviali155 can you try again with master HEAD now that #2141 is merged.

Liviali155, 2019-11-28:
Bi-sect result: Caused by firmwire commit e0491ff7[](https://github.com/thesofproject/sof/commit/e0491ff77838ba59d634105620969b33c4cbfed9)
@tlauda  please check it

Liviali155, 2019-12-02:
Error not seen on:
platform: APL UP2 with nocodec
Firmware: d740a3e https://github.com/thesofproject/sof/commits/master
Kernel: c6423829 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-apl-nocodec.tplg same as firmware

mengdonglin, 2019-12-13:
This issue is to be worked around by https://github.com/thesofproject/sof/pull/2191

Liviali155, 2019-12-13:
> @Liviali155 This issue is to be worked around by #2191 of @slawblauciak

Yes,I tried with #2191,error not seen

Liviali155, 2019-12-17:
Error not seen on  APL UP2 with PCM512x in I2S mode
Firmware: 85e462ad https://github.com/thesofproject/sof/commits/master
Kernel: c5e4a391 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-apl-pcm512x.tplg same as firmware

closed it

closed 2019-12-17 by Liviali155