thesofproject/sof #2149: [BUG][BYT][RT5682]Dmesg show "error: sink SSP0.IN not found" when doing Modules reload

Liviali155, 2019-11-28:
**Describe the bug**
Doing Modules reload tips"sof_insert failed"
Dmesg show "error: sink SSP0.IN not found" when doing Modules reload

After that aplay/arecord can work normally
It seems a sof regression

**To Reproduce**
1.sudo reboot
2.kill sof-logger process
3."sudo ./sof_bootloop.sh" to do Modules reload

**Reproduction Rate**
5/5

**Expected behavior**
Sof insert successfully

**Impact**
Sof insert failed,dmesg show "error: sink SSP0.IN not found"

```Inserting snd_soc_rt5651
Inserting snd_soc_rt5670
Inserting snd_soc_rt5677_spi
Inserting snd_soc_rt5682
Inserting snd_soc_pcm512x_i2c
Inserting snd_soc_wm8804_i2c
Inserting snd_soc_max98357a
Inserting snd_sof_pci
Inserting snd_sof_acpi
Home directory not accessible: Permission denied
sof_insert failed, see boot_fail.log for details
```

<details>
 <summary>dmesg_error.log</summary>

````
[  849.491306] sof-audio-acpi 80860F28:01: sink MIXER1.0 control none source BUF3.2
[  849.491313] sof-audio-acpi 80860F28:01: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[  849.491371] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[  849.491381] sof-audio-acpi 80860F28:01: sink SSP0.IN control none source modem_in
[  849.491386] sof-audio-acpi 80860F28:01: error: sink SSP0.IN not found
[  849.491396] sof-audio-acpi 80860F28:01: ASoC: no sink widget found for SSP0.IN
[  849.491402] sof-audio-acpi 80860F28:01: ASoC: Failed to add route modem_in -> direct -> SSP0.IN
[  849.491408] sof-audio-acpi 80860F28:01: sink modem_out control none source SSP0.OUT
[  849.491411] sof-audio-acpi 80860F28:01: error: source SSP0.OUT not found
[  849.491418] sof-audio-acpi 80860F28:01: ASoC: no source widget found for SSP0.OUT
[  849.491423] sof-audio-acpi 80860F28:01: ASoC: Failed to add route SSP0.OUT -> direct -> modem_out
[  849.491434] sof-audio-acpi 80860F28:01: tplg: 1 hw_configs found, default id: 0!
[  849.491442] sof-audio-acpi 80860F28:01: tplg: config SSP2 fmt 0x4001 mclk 19200000 bclk 2400000 fclk 48000 width (24)25 slots 2 mclk id 0 quirks 0
[  849.491451] sof-audio-acpi 80860F28:01: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  849.491535] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  849.491542] sof-audio-acpi 80860F28:01: ASoC: Parent card not yet available, widget card binding deferred
[  849.491546] sof-audio-acpi 80860F28:01: tplg: complete pipeline PIPELINE.3.SSP2.OUT id 29
[  849.491551] sof-audio-acpi 80860F28:01: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  849.491651] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  849.491656] sof-audio-acpi 80860F28:01: tplg: complete pipeline PIPELINE.2.SSP2.IN id 20
[  849.491661] sof-audio-acpi 80860F28:01: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  849.491719] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  849.491724] sof-audio-acpi 80860F28:01: tplg: complete pipeline PIPELINE.1.SSP2.OUT id 11
[  849.491729] sof-audio-acpi 80860F28:01: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  849.491788] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  849.491798] sof-audio-acpi 80860F28:01: ipc tx: 0x50020000: GLB_COMP_MSG: GET_VALUE
[  849.492043] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[  849.492054] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  849.492072] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: GET_VALUE
[  849.492082] sof-audio-acpi 80860F28:01: ipc tx: 0x50020000: GLB_COMP_MSG: GET_VALUE
[  849.492340] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: GET_VALUE
[  849.492350] sof-audio-acpi 80860F28:01: ipc tx: 0x50020000: GLB_COMP_MSG: GET_VALUE
[  849.492450] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: GET_VALUE
[  849.492459] sof-audio-acpi 80860F28:01: ipc tx: 0x50020000: GLB_COMP_MSG: GET_VALUE
[  849.492556] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: GET_VALUE
[  849.497245] sof_rt5682 sof_rt5682: rt5682-aif1 <-> ssp2-port mapping ok
[  849.497298] sof-audio-acpi 80860F28:01: creating new PCM Low Latency
[  849.497303] sof-audio-acpi 80860F28:01: spcm: allocate Low Latency Playback 0 playback DMA buffer size 0x10000 max 0x10000
[  849.497345] sof-audio-acpi 80860F28:01: spcm: allocate Low Latency Capture 0 capture DMA buffer size 0x10000 max 0x10000
[  849.497373] sof_rt5682 sof_rt5682: snd-soc-dummy-dai <-> Low Latency 0 mapping ok
[  849.497401] sof-audio-acpi 80860F28:01: creating new PCM Media Playback 1
[  849.497406] sof-audio-acpi 80860F28:01: spcm: allocate Media Playback 1 playback DMA buffer size 0x800000 max 0x800000
[  849.501080] sof_rt5682 sof_rt5682: snd-soc-dummy-dai <-> Media Playback 1 mapping ok
[  849.505690] input: sof-rt5682 Headset Jack as /devices/platform/80860F28:01/sof_rt5682/sound/card0/input5
[  849.637623] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  849.639542] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  849.639663] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  849.639994] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  849.640086] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  849.640654] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  849.640770] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  849.641115] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  849.992123] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[  849.992135] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  850.492713] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[  850.492735] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
````
</details>

**Environment**
Environment
platform: BYT MB with RT5682 in I2S mode
Firmware: a62f5623  https://github.com/thesofproject/sof/commits/master
Kernel: 0bdbe73a https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-byt-rt5682.tplg same as firmware

**Screenshots or console output**

[dmesg.log](https://github.com/thesofproject/sof/files/3900286/dmesg.log)
[sof_trace20191128.log](https://github.com/thesofproject/sof/files/3900287/sof_trace20191128.log)
[kmod_scripts20190809.zip](https://github.com/thesofproject/sof/files/3900327/kmod_scripts20190809.zip)

Liviali155, 2019-11-28:
Bi-sect result: Caused by firmwire commit 0bbe40da(https://github.com/thesofproject/sof/commit/0bbe40dad52eae1f2af5ceecfa655136daa4ee21)
@plbossart  please check it

juimonen, 2019-11-28:
So in this commit there's some hardcoded virtual widgets for SSP 2 and 0 which, I guess should be also parametrized...:
VIRTUAL_WIDGET(ssp0 Tx, out_drv, 3)
+VIRTUAL_WIDGET(ssp0 Rx, out_drv, 4)
+VIRTUAL_DAPM_ROUTE_IN(modem_in, SSP, 0, IN, 0)
+VIRTUAL_DAPM_ROUTE_OUT(modem_out, SSP, 0, OUT, 1)

juimonen, 2019-11-28:
so maybe something like: https://github.com/thesofproject/sof/pull/2152

lgirdwood, 2019-11-28:
@Liviali155 this is a kernel issue and not FW. Can you relog for kernel. Thanks. I will close here.

closed 2019-11-28 by lgirdwood