thesofproject/sof #2150: [BUG][CML]Multipipeline aplay&arecord pause release cause ipc timed out for 0x60070000 size 12

dengyangchao, 2019-11-28:
**Describe the bug**
Multipipeline aplay(0,0)&arecord(0,0), pause release arecord cause ipc timed out for 0x60070000 size 12

**To Reproduce**
1. Reboot via "sudo reboot"
2. Wait runtime PM to suspend
3. open a terminal to aplay via "aplay -Dhw:0,0 -r48000 -c2 -fs16_le 48California_Dreaming.wav -vv -i"
4. Open another terminal to arecord via "arecord -Dhw:0,0 -r48000 -c2 -fs16_le test.wav -vv -i"
5. Pause-release arecord 
6. Repeat step 5

**Reproduction Rate**
3/3

**Expected behavior**
Aplay and arecord work normal

**Impact**
1.Arecord show "connection time out" and dmesg show " ipc timed out for 0x60070000 size 12"
2.Aplay show I/O error

**Environment**
Platform: CML Chrome with onboard codec RT5682 in I2S mode
Firmware: a62f5623 https://github.com/thesofproject/sof/commits/master
Kernel: 0bdbe73a https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/sof-cml-rt5682-max98357a.tplg same as firmware

**Screenshots or console output**
[image]

**Dmesg**
```
[   80.691658] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[   80.691919] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[   80.691930] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[   80.872844] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 4
[   80.873070] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[   80.873072] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[   81.381156] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60070000 size 12
[   81.381165] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[   81.381175] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   81.381269] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   81.381282] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000083 rirb 00
[   81.381285] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   81.381297] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   81.381301] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   81.381308]  Port1: ASoC: trigger FE cmd: 4 failed: -110
[   81.381317] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 3
[   81.381324] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
```

<details>
 <summary>logger-trace-error</summary>

````
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1           SA          83327043.281250  83327040.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 26009
    0      1           SA          87515859.687500   4188816.500000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25638
    0      1           SA          97763357.812500  10247498.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25609
    0      1           SA          98475862.760417    712504.937500      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25705
    0      1           SA          99059616.770833    583754.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25752
    0      1           SA          99537103.541667    477486.781250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25536
    0      1           SA         100060860.104167    523756.562500      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25651
    0      1           SA         100535874.114583    475014.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25915
````
</details>

[amixer.txt](https://github.com/thesofproject/sof/files/3900370/amixer.txt)
[dmesg.log](https://github.com/thesofproject/sof/files/3900371/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3900372/logger.log)
[logger-error.log](https://github.com/thesofproject/sof/files/3900373/logger-error.log)

dengyangchao, 2019-11-28:
Caused by commit e0491ff , and only use xcc-compiler generate sof version can reproduce the issue.
@tlauda please help to check

mengdonglin, 2019-12-02:
Related to #2146 same bisect failure.

Liviali155, 2019-12-05:
on APL UP2 nocodec can reproduce this issue  when doing  arecord suspend/resume
arecord show "connection time out" and dmesg show " ipc timed out for 0x60040000 size 12"

***Test recipe***
platform: APL UP2 with nocodec.
Firmware: 65de27d2 https://github.com/thesofproject/sof/commits/master
Kernel: 54c96eae https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-apl-nocodec.tplg same as firmware

<details>
 <summary>dmesg_error.log</summary>

````
[  826.764192] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  826.765204] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  826.765210] sof-audio-pci 0000:00:0e.0: pcm: stream dir 1, posn mailbox offset is 790604
[  826.765308] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 5 dir 1 cmd 1
[  826.766157] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x2024001e successful
[  826.766164] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  826.979827] ata2: SATA link down (SStatus 4 SControl 300)
[  826.979862] ata1: SATA link down (SStatus 4 SControl 300)
[  827.267867] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60040000 size 12
[  827.267879] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[  827.267887] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  827.267983] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  827.268016] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000083 rirb 00
[  827.268025] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  827.268037] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  827.268047] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  827.268060]  Port5: ASoC: trigger FE cmd: 1 failed: -110
[  827.268100] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 5 dir 1 cmd 0
[  827.268110] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  827.771871] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60050000 size 12
[  827.771884] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[  827.771894] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  827.771942] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  827.771954] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000083 rirb 00
[  827.771960] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  827.771968] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  827.771974] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  827.772780] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20240000 successful
[  827.772801]  Port5: ASoC: trigger FE cmd: 0 failed: -110
[  827.773333] sof-audio-pci 0000:00:0e.0: pcm: free stream 5 dir 1
[  827.773345] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  828.275854] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60030000 size 12
[  828.275866] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[  828.275875] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  828.275923] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  828.275934] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
[  828.275939] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  828.275947] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  828.275952] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  828.276225] sof-audio-pci 0000:00:0e.0: pcm: close stream 5 dir 1
[  828.484299] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 5 dir 0 cmd 0
[  828.484309] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  828.987868] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60050000 size 12
[  828.987881] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[  828.987890] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  828.987937] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  828.987948] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
[  828.987953] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  828.987961] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  828.987967] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  828.987980] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20140000 successful
[  828.987988]  Port5: ASoC: trigger FE cmd: 0 failed: -110
[  828.988026] sof-audio-pci 0000:00:0e.0: pcm: free stream 5 dir 0
[  828.988035] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  829.491856] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60030000 size 12
[  829.491869] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[  829.491879] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[... cut ...]
[  829.491953] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  829.491960] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  829.492172] sof-audio-pci 0000:00:0e.0: pcm: close stream 5 dir 0
[  829.749507] r8169 0000:03:00.0 enp3s0: Link is Up - 1Gbps/Full - flow control rx/tx	
````
</details>

[dmesg20191205.log](https://github.com/thesofproject/sof/files/3926070/dmesg20191205.log)
[sof_trace20191205.log](https://github.com/thesofproject/sof/files/3926071/sof_trace20191205.log)

wwittbrx, 2019-12-11:
Could you check if cherry-picking https://github.com/thesofproject/sof/commit/cc4c356d7db464388ca7995c38b93c302cf07868 fixes this issue for you? This is just a quick workaround, not a full fix, so don't close the issue if you can't reproduce.

dengyangchao, 2019-12-12:
@wwittbrx  I added the commit cc4c356  , it works , issue can't reproduce.
Platform: CML Chrome with onboard codec RT5682 in I2S mode

mengdonglin, 2019-12-13:
@dengyangchao This issue is to be worked around by https://github.com/thesofproject/sof/pull/2191

Liviali155, 2019-12-13:
> @dengyangchao This issue is to be worked around by #2191 of @slawblauciak

Yes,tried with #2191,error not seen
***Environment***
Platform: CML Chrome with onboard codec RT5682 in I2S mode
Firmware: f5bcc4bb https://github.com/thesofproject/sof/commits/master
Kernel: 9c0401ca https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/sof-cml-rt5682-max98357a.tplg same as firmware

dengyangchao, 2019-12-17:
Issue can't be reproduced, since #2191 merged, close

closed 2019-12-17 by dengyangchao