thesofproject/sof #2154: [BUG][APL][PCM512x]"error: ipc error for 0x60010000 size 20" when aplay via(0,0)

Liviali155, 2019-12-02:
**Describe the bug**
Aplay via (0,0) can not work,dmesg show "error: ipc error for 0x60010000 size 20"

**To Reproduce**
1.sudo reboot
2.Run "aplay -Dhw:0,0 -f dat -c 2 wav/48California_Dreaming.wav -vv -i" to aplay via (0,0)
```
ubuntu@cd-apl-up2-pcm512x-07:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 0: Port5 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 1: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 2: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 3: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 4: Media Playback 4 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice
```

**Reproduction Rate**
5/5

**Expected behavior**
Aplay via (0,0) normally

**Impact**
Aplay via (0,0) can not work,dmesg show "error: ipc error for 0x60010000 size 20"

```ubuntu@cd-apl-up2-pcm512x-07:~$ aplay -Dhw:0,0 -f dat -c 2 wav/48California_Dreaming.wav -vv -i
Playing WAVE 'wav/48California_Dreaming.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 85000
PERIOD_SIZE: 4080
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 341000
BUFFER_SIZE: 16368
BUFFER_BYTES: 65472
TICK_TIME: 0
ubuntu@cd-apl-up2-pcm512x-07:~$
```

<details>
 <summary>dmesg_error.log</summary>

````
[   41.719205] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x140000 successful
[   41.719223] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
[   41.719228] sof-audio-pci 0000:00:0e.0: period_bytes:0x1080
[   41.719231] sof-audio-pci 0000:00:0e.0: periods:15
[   41.719243] sof-audio-pci 0000:00:0e.0: stream_tag 1
[   41.719253] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   41.719603] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[   41.719615] sof-audio-pci 0000:00:0e.0: error: hw params ipc failed for stream 1
[   41.719621] sof-audio-pci 0000:00:0e.0: ASoC: 0000:00:0e.0 hw params failed: -22
[   41.719631]  Port5: ASoC: hw_params FE failed -22
[... 2458 lines cut ...]
    0      1        MIXER 1.3      41430298.958333      3939.895752    src/audio/mixer.c:189 	mixer_params() error: period_bytes = 0
    0      1         PIPE          41430304.843750         5.885417 src/audio/pipeline.c:393 	pipeline_params() error: ret = -22, host->comp.id = 0
    0      1          IPC          41430310.989583         6.145833    src/ipc/handler.c:298 	ipc: pipe 1 comp 0 params failed -22
    0      1        MIXER 1.3      41432443.958333      2132.968750    src/audio/mixer.c:189 	mixer_params() error: period_bytes = 0
    0      1         PIPE          41432449.947917         5.989583 src/audio/pipeline.c:393 	pipeline_params() error: ret = -22, host->comp.id = 0
    0      1          IPC          41432456.302083         6.354167    src/ipc/handler.c:298 	ipc: pipe 1 comp 0 params failed -22
    0      1        MIXER 1.3      41434622.656250      2166.354248    src/audio/mixer.c:189 	mixer_params() error: period_bytes = 0
    0      1         PIPE          41434628.802083         6.145833 src/audio/pipeline.c:393 	pipeline_params() error: ret = -22, host->comp.id = 0
    0      1          IPC          41434635.104167         6.302083    src/ipc/handler.c:298 	ipc: pipe 1 comp 0 params failed -22
    0      1        MIXER 1.3      41436779.479167      2144.375000    src/audio/mixer.c:189 	mixer_params() error: period_bytes = 0
    0      1         PIPE          41436785.572917         6.093750 src/audio/pipeline.c:393 	pipeline_params() error: ret = -22, host->comp.id = 0
    0      1          IPC          41436791.875000         6.302083    src/ipc/handler.c:298 	ipc: pipe 1 comp 0 params failed -22
    0      1        MIXER 1.3      41438887.604167      2095.729248    src/audio/mixer.c:189 	mixer_params() error: period_bytes = 0
    0      1         PIPE          41438893.593750         5.989583 src/audio/pipeline.c:393 	pipeline_params() error: ret = -22, host->comp.id = 0
    0      1          IPC          41438900.052083         6.458333    src/ipc/handler.c:298 	ipc: pipe 1 comp 0 params failed -22
    0      1        MIXER 1.3      41442925.312500      4025.260498    src/audio/mixer.c:189 	mixer_params() error: period_bytes = 0
    0      1         PIPE          41442931.354167         6.041667 src/audio/pipeline.c:393 	pipeline_params() error: ret = -22, host->comp.id = 0
    0      1          IPC          41442937.656250         6.302083    src/ipc/handler.c:298 	ipc: pipe 1 comp 0 params failed -22
    0      1        MIXER 1.3      41445166.562500      2228.906250    src/audio/mixer.c:189 	mixer_params() error: period_bytes = 0
    0      1         PIPE          41445172.552083         5.989583 src/audio/pipeline.c:393 	pipeline_params() error: ret = -22, host->comp.id = 0
    0      1          IPC          41445178.802083         6.250000    src/ipc/handler.c:298 	ipc: pipe 1 comp 0 params failed -22
    0      1        MIXER 1.3      41447379.010417      2200.208252    src/audio/mixer.c:189 	mixer_params() error: period_bytes = 0
    0      1         PIPE          41447385.104167         6.093750 src/audio/pipeline.c:393 	pipeline_params() error: ret = -22, host->comp.id = 0
    0      1          IPC          41447391.354167         6.250000    src/ipc/handler.c:298 	ipc: pipe 1 comp 0 params failed -22
    0      1        MIXER 1.3      41449573.854167      2182.500000    src/audio/mixer.c:189 	mixer_params()

Liviali155, 2019-12-02:
Bi-sect result: Caused by firmwire commit 74e1bcc7[](https://github.com/thesofproject/sof/commit/74e1bcc7866f22e1ec5c060c5448fa6280ca86fc)
@bkokoszx  please check it

Liviali155, 2019-12-02:
Issue can be reproduced on 

platform: BYT MB with RT5682 in I2S mode
Firmware: feb9e2b2 https://github.com/thesofproject/sof/commits/master
Kernel: 0bdbe73a https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-byt-rt5682.tplg same as firmware

sinahuang, 2019-12-02:
Issue still can be reproduced on APL UP2 with nocodec.
**Test recipe**
platform: APL UP2 with nocodec.
Firmware: feb9e2b https://github.com/thesofproject/sof/commits/master
Kernel: 0bdbe73a https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: tools/topology/sof-apl-nocodec.tplg same as firmware

bkokoszx, 2019-12-02:
@Liviali155 
Do you have reproduction with PR #2156?

Liviali155, 2019-12-03:
> @Liviali155
> Do you have reproduction with PR #2156?

Tried with PR #2156 on APL UP2 with PCM512x in I2S mode,error not seen

PR #2156 (2019-12-04): comp: fetch params from sink buffer
Fetches params (for mux and demux) from sink buffer
(sink buffer params are set in pipeline_set_params() before
mixer/mux_params()). For mux and mixer components some
of source buffer can be unset yet in mixer/mux_params(). 

Fixes #2154
closed 2019-12-04 by plbossart