thesofproject/sof #2157: [FEATURE] Add support for Samus (2015 Pixelbook

cujomalainey, 2019-12-02:
**Is your feature request related to a problem? Please describe.**
A common machine for Google to test on is the pixelbooks, it would be excellent to add samus to the mix

**Additional context**
Uses the bdw-rt5677 machine driver

lgirdwood, 2019-12-16:
@wenqingfu @mengdonglin do we have any Samus HW ?

cujomalainey, 2019-12-16:
I think we found the issue in the said links in another bug, we are patching both samys and buddy to support SOF

cujomalainey, 2019-12-16:
https://github.com/thesofproject/linux/pull/1484

xiulipan, 2019-12-18:
@cujomalainey I have test on one device with https://github.com/thesofproject/sof/pull/2188 and https://github.com/thesofproject/sof/pull/2095
Will try to redo the test after rebase and update thesofproject/linux#1484

plbossart, 2020-07-29:
Samus is well supported now, the only bug is https://github.com/thesofproject/sof/issues/2459 at the firmware level, and we need to sort out suspend-resume + improve the position reports without IPC:
https://github.com/thesofproject/linux/issues?q=is%3Aissue+is%3Aopen+label%3ABDW

lgirdwood, 2020-08-04:
@plbossart do you think the SSP L/R bit is inverted here ? or incorrectly documented ? It seems like the L/R swap fix is probably a SSP bit change (unless you have already tried this) ?

plbossart, 2020-08-05:
> @plbossart do you think the SSP L/R bit is inverted here ? or incorrectly documented ? It seems like the L/R swap fix is probably a SSP bit change (unless you have already tried this) ?

Yeah,  I tried and it's not a polarity issue (happens also with DSP_A/B). I have pictures somewhere.

plbossart, 2020-08-05:
@lgirdwood see screen grabs in https://github.com/thesofproject/sof/issues/2459

lgirdwood, 2020-08-06:
@plbossart thanks got you. Fwiw, I wonder (since swapping is consistent) if we were to write a single 32 bit word to the SSP FIFO as part of trigger for > 16 bit data output to re-order the FIFO output on the link. Alternatively we could fill the FIFO with 16 * 0s  as part of trigger start meaning the FIFO is always in a consistent state at start has send data without underruning (which may mean we drop a sample and invert L/R). Just guessing....

plbossart, 2021-06-10:
Closing, Samus is well supported at the firmware level. The driver is still missing suspend/resume support but that's a different request that will probably never be implemented.

closed 2021-06-10 by plbossart