thesofproject/sof #2158: [FEATURE] TGL SOF fw build with otc 3k key

sathya-nujella, 2019-12-02:
Need SOF fw build with otc 3k key.
Please provide build infra/methodology for TGL.

For TGL, below keys are added.. TGL sof fw build should utilize the keys:
   rimage: keys: add new 3k key   
        ./rimage/keys/
                otc_private_key_3k.pem
                otc_public_key_3k.pem

marc-hb, 2020-10-06:
@sathya-nujella I believe #3187 implemented this, can you confirm so we can close? It seems to work for a number of people, see #3187. Note this require the MEU_TOOL, rimage is not enough.

For TGL support in rimage there is another issue: #1833

lgirdwood, 2020-10-29:
See https://github.com/thesofproject/rimage/pull/33 for rimage CAVS 2.5 signing WIP PR.

cujomalainey, 2020-12-09:
I believe this is complete since the PR is closed

lgirdwood, 2020-12-10:
Yes code is upstream in rimage now.

closed 2020-12-10 by lgirdwood