thesofproject/sof #2164: [BUG] S0ix failure with ipc timed out for 0x40010000 size 76

keqiaozhang, 2019-12-04:
**Describe the bug**
This issue only occurs when the device automatically goes to S0ix and unable to reproduce it with command line.

**To Reproduce**
1. Disable rootfs verification:
    /usr/share/vboot/bin/make_dev_ssd.sh --remove_rootfs_verification --partitions 2
2. reboot
3. mv /etc/init/console-ttyS0.conf /usr/local/
4. reboot
5. Use guest mode to log in
6. create several shell to observe s0ix residency, cstate and dmesg constantly
   * watch -n 5 cat /sys/kernel/debug/pmc_core/package_cstate_show 
   * watch -n 5 cat /sys/kernel/debug/pmc_core/slp_s0_residency_usec 
   * dmesg -w 
7. Create one blank browser page in full screen mode(press the full screen bottom near the refresh on keyboard ) and press Esc to remove mouse cursor
8. Idle the device for more than 60 secs(You don't have to wait for the screen to turn off)
9. Check the dmesg page to see if any ipc timed output.
10. Repeat the idle steps. 

**Reproduction Rate**
The reproduction rate is > 30%. On my device, I can reproduce the issue at every first time.

**Expected behavior**
Device can suspend and resume from S0ix.

**Impact**
Audio failed to resume form S0ix.

**Environment**
1) CML 007 drop of firmware
2) Name of the topology file
* Topology:  sof-da7219-max98357a.m4
3) Name of the platform(s) on which the bug is observed.
* Platform:CML

**Screenshots or console output**
dmesg
```
[   85.604297]  Speakers: ASoC: pre trigger FE Speakers cmd 0
[   85.604310] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[   85.604925] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[   85.604947] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000
[   85.605374] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000
[   85.605382]  SSP1-Codec: ASoC: trigger BE SSP1-Codec cmd 0
[   85.605465]  Speakers: ASoC: hw_free FE Speakers
[   85.605472] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[   85.605481] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000
[   85.605776] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000
[   85.605798]  SSP1-Codec: ASoC: hw_free BE SSP1-Codec
[   85.605836]  SSP1-Codec: ASoC: close BE SSP1-Codec
[   85.618945]  Speakers: ASoC: close FE Speakers
[   85.618951] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[   85.618955]  SSP1-Codec: ASoC: BE SSP1-Codec event 2 dir 0
[   85.619964]  Speakers: ASoC: BE playback disconnect check for SSP1-Codec
[   85.619966]  Speakers: freed DSP playback path Speakers -> SSP1-Codec
[   87.940767] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[   87.940810] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000
[   88.243292] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[   88.243318] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   88.244363] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   88.244395] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[   88.244400] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   88.244424] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   88.244431] sof-audio-pci 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
[   88.244485] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[   88.244498] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[   88.244505] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[   88.244592] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
[   90.820093]  Speakers: ASoC: pop wq checking: Playback status: inactive waiting: yes
```
trace 
```
    0      2          IPC          82173896.250000       821.614563    src/ipc/handler.c:628 	ipc: pm -> restore
    0      2          IPC          82174523.385417       627.135437    src/ipc/handler.c:221 	ipc: comp 34 -> params
    0      2         PIPE 7.39     82174541.041667        17.656250 src/audio/pipeline.c:315 	pipeline_params()
    0      2         HOST          82174546.562500         5.520833     src/audio/host.c:511 	host_params()
    0      2          DMA          82174562.395833        15.833333 intel/cavs/hda-dma.c:462 	hda-dmac: 5 channel 0 -> get
    0      2          DMA          82174568.125000         5.729167 intel/cavs/hda-dma.c:687 	hda-dmac: 5 channel 0 -> config
    0      2       VOLUME          82174575.677083         7.552083 udio/volume/volume.c:247 	volume_params()
    0      2          DAI 7.38     82174581.145833         5.468750      src/audio/dai.c:356 	dai_params()
    0      2         PIPE 7.39     82174590.156250         9.010417 src/audio/pipeline.c:379 	pipeline_prepare()
    0      2         HOST          82174595.729167         5.572917     src/audio/host.c:641 	host_prepare()
    0      2       VOLUME          82174602.291667         6.562500 udio/volume/volume.c:593 	volume_prepare()
    0      2          DAI 7.38     82174611.197917         8.906250      src/audio/dai.c:420 	dai_prepare()
    0      2          IPC          82192934.166667     18322.968750    src/ipc/handler.c:450 	ipc: comp 34 -> trigger cmd 0x40000
    0      2         PIPE 7.39     82192960.989583        26.822916 src/audio/pipeline.c:608 	pipeline_trigger()
    0      2         HOST          82192966.250000         5.260417     src/audio/host.c:291 	host_trigger()
    0      2          DMA          82192971.614583         5.364583 intel/cavs/hda-dma.c:531 	hda-dmac: 5 channel 0 -> start
    0      2          DMA          82192976.666667         5.052083 intel/cavs/hda-dma.c:353 	hda-dmac: 5 channel 0 -> enable
    0      2       VOLUME          82193036.927083        60.260418 udio/volume/volume.c:533 	volume_trigger()
    0      2          DAI 7.38     82193042.083333         5.156250      src/audio/dai.c:483 	dai_comp_trigger(), command = 1
    0      2          DAI 7.38     82193046.927083         4.843750      src/audio/dai.c:494 	dai_comp_trigger(), START
    0      2          SSP          82193211.875000       164.947922 ers/intel/cavs/ssp.c:885 	ssp_trigger() cmd 1
    0      2          SSP          82193217.812500         5.937500 ers/intel/cavs/ssp.c:814 	ssp_start()
    0      2          IPC         106996325.416667  24803108.000000    src/ipc/handler.c:450 	ipc: comp 34 -> trigger cmd 0x50000
    0      2         PIPE 7.39    106996343.541667        18.125000 src/audio/pipeline.c:608 	pipeline_trigger()
    0      2         HOST         106996348.750000         5.208333     src/audio/host.c:291 	host_trigger()
    0      2          DMA         106996354.322917         5.572917 intel/cavs/hda-dma.c:626 	hda-dmac: 5 channel 0 -> stop
    0      2       VOLUME         106996359.739583         5.416667 udio/volume/volume.c:533 	volume_trigger()
    0      2          DAI 7.38    106996364.791667         5.052083      src/audio/dai.c:483 	dai_comp_trigger(), command = 0
    0      2          DAI 7.38    106996369.479167         4.687500      src/audio/dai.c:547 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         106996373.958333         4.479167 src/drivers/dw/dma.c:451 	dw_dma_stop(): dma 0 channel 3 stop
    0      2          SSP         106996389.635417        15.677083 ers/intel/cavs/ssp.c:885 	ssp_trigger() cmd 0
    0      2          SSP         106996449.010417        59.375000 ers/intel/cavs/ssp.c:866 	ssp_stop(), TX stop
    0      2          SSP         106996454.531250         5.520833 ers/intel/cavs/ssp.c:875 	ssp_stop(), SSP port disabled
    0      2          IPC         106997006.197917       551.666687    src/ipc/handler.c:339 	ipc: comp 34 -> free
    0      2         PIPE 7.39    106997023.020833        16.822916 src/audio/pipeline.c:687 	pipeline_reset()
    0      2         HOST         106997028.020833         5.000000     src/audio/host.c:685 	host_reset()
```

[trace.txt](https://github.com/thesofproject/sof/files/3919526/trace.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/3919527/dmesg.txt)

ranj063, 2019-12-04:
@keqiaozhang the logs and the steps to reproduce dont seem to be consistent. You only mention reboot and idle as the steps to reproduce. But there are 5 instances of pcm open in the dmesg. Where are those coming from?

keqiaozhang, 2019-12-04:
@ranj063 
It came form cras, when you type a wrong command and try press ‘’Tab‘’ to complete it, you will hear a "ding" sound, this is the key part to reproduce this issue.  Leave the device idle before you hear the 'ding' sound, you will 100% reproduce this issue. I have updated the steps.

ranj063, 2019-12-04:
@keqiaozhang what do you mean by "make sure PCM is occupied by CRAS"?

keqiaozhang, 2019-12-04:
@ranj063 The "ding" sound is played by cras on speaker as a warning.  So at this point, speaker is occupying by cras.

keyonjie, 2019-12-05:
we have confirmed that the issue is caused by entering runtime S0Ix before audio is runtime suspended, working on solution now.

keyonjie, 2019-12-06:
we already have the fix for this, will provide it soon.

mengdonglin, 2019-12-08:
We need a fix in PCH settings by coreboot: system can enter runtime S0ix only when Audio DSP is off

closed 2019-12-08 by mengdonglin