thesofproject/sof #2166: [BUG][GLK]Playback repetitively via HDMI cause ipc timed out for 0x60040000 size 12

dengyangchao, 2019-12-04:
**Describe the bug**
Playback repetitively  via HDMI cause ipc timed out for 0x60040000 size 12. 
Issue can reproduced via "s2idle&resume with HDMI playback" stress test, fail at 142th.

**To Reproduce**
1.Reboot system via"sudo reboot".
2.Wait runtime PM to suspend.
3.playback HDMI via```./sof-test-playback.sh```

**Reproduction Rate**
3/3
1st: fail at 78th
2nd: fail at 1st
3rd: fail at 144th

**Expected behavior**
Playback via HDMI normal

**Impact**
Cause ipc timed out for 0x60040000 size 12

**Environment**
Platform: GLK Chrome with onboard codec DA7219 in I2S mode 
Firmware: 0205c95a https://github.com/thesofproject/sof/commits/master
Kernel: 709427e6 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-glk-da7219.tplg same as firmware

dmesg
```
[   39.385686] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   39.385689] sof-audio-pci 0000:00:0e.0: pcm: stream dir 0, posn mailbox offset is 790604
[   39.404029] sof-audio-pci 0000:00:0e.0: In hda_link_pcm_trigger cmd=1
[   39.404033] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 6 dir 0 cmd 1
[   39.404418] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x24001e successful
[   39.404420] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   39.908426] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60040000 size 12
[   39.908431] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[   39.908435] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[   39.908477] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[   39.908482] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000181 rirb 00
[   39.908484] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   39.908487] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   39.908489] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[   39.908493]  HDMI2: ASoC: trigger FE cmd: 1 failed: -110
[   39.908500] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 6 dir 0 cmd 0
[   39.908505] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   40.412427] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60050000 size 12
[   40.412433] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[   40.412436] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[   40.412479] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[   40.412484] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000181 rirb 00
[   40.412486] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   40.412489] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   40.412491] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[   40.412497] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20240000 successful
[   40.412501]  HDMI2: ASoC: trigger FE cmd: 0 failed: -110
```

[dmesg.log](https://github.com/thesofproject/sof/files/3920681/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3920682/logger.log)
[sof-test-playback.sh.txt](https://github.com/thesofproject/sof/files/3920683/sof-test-playback.sh.txt)

dengyangchao, 2019-12-05:
Caused by commit e0491ff, and related to #2146 ,same bisect failure @tlauda please help to check

mengdonglin, 2019-12-13:
@dengyangchao This issue is to be worked around by https://github.com/thesofproject/sof/pull/2191 of @slawblauciak

Liviali155, 2019-12-13:
> @dengyangchao This issue is to be worked around by #2191 of @slawblauciak

Tried with #2191，error not seen
***Environment***
Platform: GLK Chrome with onboard codec DA7219 in I2S mode
Firmware: f5bcc4bb https://github.com/thesofproject/sof/commits/master
Kernel: 9c0401ca https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-glk-da7219.tplg same as firmware

dengyangchao, 2019-12-17:
Issue can't be reproduced since #2191 merged, close.

closed 2019-12-17 by dengyangchao