thesofproject/sof #2170: [BUG] Trace line number and file name mismatch

paulstelian97, 2019-12-04:
**Describe the bug**
When I have traces enabled (both etrace and DMA trace) and I have the verbose traces enabled,
the WFE/WFX traces have wrong file names. The file name is src/schedule/task.c, but the line number is the one in src/include/sof/lib/wait.h (34, resp. 40). This makes it confusing to users
of the trace subsystem.

**To Reproduce**
Steps to reproduce the behavior: Enable verbose trace (if the trace class specification is merged, enable the TRACE_CLASS_WAIT verbose traces), then get in the situation where this trace
is emitted and you can see it. The file name will be of the C file which calls the inline function
instead of the actual header file.

The file name may even differ from that due to oddities with inline functions, dependent on
optimization levels (and whether the function is inlined or not).

The line number will in any case not match the specified file name.

If this particular situation is hard to reproduce, pick any other trace that is done within an inline
function inside a header. It will also manifest this bug.

**Reproduction Rate**
Always (the error is static and visible in the generated code).

**Expected behavior**
The file name and line number should match and either be of the call of the inline function
or to have both the file name and line number be of the header itself.

**Impact**
Annoyance

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: (internal build, though irrelevant; should manifest on any as it's strictly firmware)
    * SOF: fc87fdfc
2) Name of the topology file
    * Topology: (not relevant, but it is sof-imx8qxp-cs42888.m4)
3) Name of the platform(s) on which the bug is observed.
    * Platform: (not relevant, but it is IMX8)

**Screenshots or console output**
(I'm not sure if there is a good reason that I get an interrupt every millisecond which creates
the logs themselves, but that's a different issue).

```
    0      2         WAIT         1394022547.929430       992.974487  src/schedule/task.c:40    WFX
    0      2         WAIT         1394022555.243243         7.313814  src/schedule/task.c:34    WFE
    0      2         WAIT         1394023547.794294       992.551025  src/schedule/task.c:40    WFX
    0      2         WAIT         1394023554.984985         7.190691  src/schedule/task.c:34    WFE
    0      2         WAIT         1394024547.758258       992.773254  src/schedule/task.c:40    WFX
    0      2         WAIT         1394024554.834835         7.076577  src/schedule/task.c:34    WFE
    0      2         WAIT         1394025547.956456       993.121643  src/schedule/task.c:40    WFX
    0      2         WAIT         1394025555.075075         7.118618  src/schedule/task.c:34    WFE
    0      2         WAIT         1394026547.965466       992.890381  src/schedule/task.c:40    WFX
    0      2         WAIT         1394026555.147147         7.181682  src/schedule/task.c:34    WFE
    0      2         WAIT         1394027547.755255       992.608093  src/schedule/task.c:40    WFX
    0      2         WAIT         1394027554.930931         7.175676  src/schedule/task.c:34    WFE
```

paulstelian97, 2019-12-04:
@jajanusz It was you who introduced the original changes with RELATIVE_FILE right? Is there any easy way to fix this issue or a more complex solution is required?

I have found an interesting solution [here](https://stackoverflow.com/a/40947954/3897333) but I worry that it may fail to satisfy the
original problem of getting a reproducible build. Could it be used?

lgirdwood, 2019-12-16:
@mmaka1 fyi, looks like this will also need to be fixed for ECL plugins.

jajanusz, 2020-02-05:
RELATIVE_FILE works just like __FILE__ (gives name of compilation unit, but is determmenistic, when __FILE__ can be whatever compiler decides it to be and can change between runs in different locations), what means that they are just misused in cases that you mentioned. I didn't get deeply into issue but if traces are used just-like-that in headers they will never work and it has to be reworked.

paulstelian97, 2020-02-06:
@jajanusz So inline functions should just not use traces? That's quite the limit.