thesofproject/sof #2174: [FEATURE] Add a way to check APL firmware boot log

musiclvme, 2019-12-05:
**How to check the dsp firmware boot log**
I know that sof-logger is a good way to check runtime log, but how to check firmware boot log ?
is there a existed way or not?

**Describe alternatives you've considered**
Does trace_point(TRACE_BOOT_START)  means boot log, how to check this log on APL?
uart would be added to check boot log in the future on APL?

paulstelian97, 2019-12-05:
@tlauda I feel like this issue isn't platform specific. Any input on that?

lgirdwood, 2019-12-17:
@musiclvme this looks like we need a very simple script to parse dmesg for the trace point debug printk on boot failure. This script could additionally also run the gcc preprocessor on trace.h so taht dmesg trace point values could be matched with current values.

plbossart, 2019-12-17:
I don't get the ask. If the firmware does not boot, you will not even get a firmware trace.

paulstelian97, 2019-12-17:
Right now I'm investigating boot issues on my end and no, platform_trace_point on the various Intel platforms just writes some value in a specific memory location to be checked.

I have actual issues with traces which happen during platform init, after the traces themselves are initialized but before boot is complete -- while SOF will write in the mailbox (with mailbox traces enabled) that can't be seen from userspace except manually with memtool (at least on my platform) because the /sys/kernel/debug/sof directory is created only when firmware is fully booted. I wonder, can't at least some of the files (etrace, trace and exception, maybe) be created earlier?

lgirdwood, 2019-12-18:
@plbossart FW boot failure wont give any trace data, but it does give trace_point data (i.e. integers, generic on all platforms) and worst case ROM values (on Intel SKL+)

paulstelian97, 2019-12-18:
@lgirdwood Trace points are *almost* generic, on i.MX the trace_point macro is a no-op. What would the best implementation be? Just storing it in a (fixed) location that kernel code or memtool would be able to pick it up?

lgirdwood, 2019-12-18:
@paulstelian97 yes, just write it to a known offset in your mailbox or shared IP. Something that both easy for your driver and FW.

paulstelian97, 2019-12-18:
Will come back with an implementation early next year (I don't think I have the time this week) for trace points for my platform.

Anyway, back on traces. At least my code does print out some early traces in the probe() functions of various drivers such as EDMA and ESAI. Without an actual serial console (currently I only have a hacky implementation) I cannot see those traces, which is why I at least partly agree with what this issue is saying.

kv2019i, 2025-12-17:
No activity in 5 years, closing. Please reopen if still relevant.