thesofproject/sof #2176: [BUG]DSP panic when change DMIC arecord between HW:0,6 and HW:0,7 quickly

dengyangchao, 2019-12-06:
**Describe the bug**
DSP panic when change DMIC arecord between HW:0,6 and HW:0,7 quickly. 
Can execute a script ```test1.sh``` like below:
```
arecord -Dhw:0,6 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
arecord -Dhw:0,7 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
arecord -Dhw:0,6 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
arecord -Dhw:0,7 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
```

Arecord -l  list
```
**** List of CAPTURE Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 6: DMIC48kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 7: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

**To Reproduce**
1.Reboot system via"sudo reboot".
2.Wait runtime PM to suspend.
3.Execute script via```./test1.sh```
[test1.sh.txt](https://github.com/thesofproject/linux/files/3930864/test1.sh.txt)

**Reproduction Rate**
3/3

**Expected behavior**
Change dmic arecord successfully

**Impact**
Cause dsp panic

**Environment**
Platform: CFL-S RVP with onboard codec ALC700 in HDA mode
Firmware: 65de27d2 https://github.com/thesofproject/sof/commits/master
Kernel: 54c96eae https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-4ch.tplg same as firmware

<details>
 <summary>Dmesg</summary>

````
[  239.891683] sof-audio-pci 0000:00:1f.3: periods:4
[  239.891697] sof-audio-pci 0000:00:1f.3: stream_tag 2
[  239.891700] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  239.891982] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  239.891983] sof-audio-pci 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 790604
[  239.892317] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 7 dir 1 cmd 1
[  239.892765] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[  239.892766] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  239.892994] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[  239.892996] sof-audio-pci 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[  239.892997] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  239.892998] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  239.893132] sof-audio-pci 0000:00:1f.3: error: runtime exception
[  239.893132] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[  239.893133] sof-audio-pci 0000:00:1f.3: error: panic at :0
[  239.893134] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[  239.893135] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  239.893135] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000000 PS       0x00060325 SAR     0x00000000
[... 129 lines cut ...]
    0      2          DAI 11.74   299853923.593750         8.697917      src/audio/dai.c:448 	dai_prepare()
    0      2          IPC         299855052.864583      1129.270874    src/ipc/handler.c:457 	ipc: comp 68 -> trigger cmd 0x40000
    0      2         PIPE 11.75   299855075.677083        22.812500 src/audio/pipeline.c:711 	pipeline_trigger()
    0      2         HOST 11.68   299855080.156250         4.479167     src/audio/host.c:297 	host_trigger()
    0      2          DMA         299855085.208333         5.052083 intel/cavs/hda-dma.c:456 	hda-dmac: 4 channel 1 -> start
    0      2          DMA         299855090.208333         5.000000 intel/cavs/hda-dma.c:321 	hda-dmac: 4 channel 1 -> enable
    0      2       VOLUME 11.69   299855149.947917        59.739582 udio/volume/volume.c:549 	volume_trigger()
    0      2       EQ_IIR 11.70   299855154.739583         4.791667 udio/eq_iir/eq_iir.c:741 	eq_iir_trigger()
    0      2          DAI 11.74   299855159.739583         5.000000      src/audio/dai.c:534 	dai_comp_trigger(), command = 1
    0      2          DAI 11.74   299855164.218750         4.479167      src/audio/dai.c:545 	dai_comp_trigger(), START
    0      2         DMIC         299855168.385417         4.166667 rs/intel/cavs/dmic.c:1429	dmic_trigger()
    0      2         DMIC         299855172.916667         4.531250 rs/intel/cavs/dmic.c:1229	dmic_start()
    0      2         DMIC         299855181.614583         8.697917 rs/intel/cavs/dmic.c:1247	dmic_start(), dmic->fifo_b
    0      2         DMIC         299855186.406250         4.791667 rs/intel/cavs/dmic.c:1274	dmic_start(), mic_a = 1, mic_b = 1, fir_a = 1, fir_b = 1
    0      2         DMIC         299855191.458333         5.052083 rs/intel/cavs/dmic.c:1274	dmic_start(), mic_a = 1, mic_b = 1, fir_a = 1, fir_b = 1
    0      2         DMIC         299855196.718750         5.260417 rs/intel/cavs/dmic.c:1331	dmic_start(), dmic_active_fifos = 1
    0      2  SCHEDULE_LL         299855202.760417         6.041667 edule/timer_domain.c:47  	timer_domain_register()
    0      2         DMIC         299855209.375000         6.614583 rs/intel/cavs/dmic.c:1344	dmic_start(), done

````
</details>

[dmesg.log](https://github.com/thesofproject/linux/files/3930909/dmesg.log)
[logger.log](https://github.com/thesofproject/linux/files/3930910/logger.log)
[logger-error.log](https://github.com/thesofproject/linux/files/3930911/logger-error.log)

xiulipan, 2019-12-06:
The issue can also be reproduced by change the order to.
```
arecord -Dhw:0,7 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
arecord -Dhw:0,6 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
arecord -Dhw:0,7 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
arecord -Dhw:0,6 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
```

dengyangchao, 2019-12-06:
Tried on ICL-U RVP ALC700 HDA mode, use the Nov 1st daily test version still can reproduce.
Recipe as below:
Platform: ICL-U RVP with onboard codec ALC700 in HDA mode
Firmware: fe321d65 https://github.com/thesofproject/sof/commits/master
Kernel: c265f25d https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-4ch.tplg same as firmware

dengyangchao, 2019-12-10:
Issue can't reproduce on **sof-dev(3c391073) + v1.3** and **sof-v5.0(32c6c62) + v1.3**
recipe as below:

Kernel: https://github.com/thesofproject/linux/commits/topic/sof-dev 3c391073
Kernel: https://github.com/thesofproject/linux/commits/release/sof-v5.0 32c6c62
FW: https://github.com/thesofproject/linux-firmware/tree/master/intel/sof/cml/intel   sof-cml-v1.3-0f73628.ri     rename to sof-cml.ri
Platform: CML laptop with ALC3204 in HDA mode

singalsu, 2019-12-10:
I changed for my WHL device the recording to stereo. I don't get DSP panic but an I/O error for the fourth capture: "arecord: pcm_read:2103: read error: Input/output error".

The error trace shows a lot of errors of type "dmic_irq_handler() error: full fifo B or PDM overrrun". It means that FIFO watermark has been exceeded in DMIC HW due to DMA has stopped consuming data. And DMA could stop consuming if the pipeline has stopped. I'll see if I can find out more.

singalsu, 2019-12-10:
When I change capture to four channels, I get the panic too.

mmaka1, 2019-12-11:
Pls attach the core dump content as well anytime a dsp panic is reported.

singalsu, 2019-12-12:
I get no more the panic if I remove IIR from DMIC capture pipes. I'll debug now if there's a memory issue there.

singalsu, 2019-12-12:
Nothing yet found out but some status update here: In verbose trace this is the last seen before panic that happens at trigger start for the pipeline. There's no component copy() operations happening after.

```
   0      2       EQ_IIR 11.70   174874290.989583      1399.791626 udio/eq_iir/eq_iir.c:756 	eq_iir_trigger()
    0      2          DMA         174874306.562500        15.572917 intel/cavs/hda-dma.c:367 	hda-dmac: 4 channel 0 -> copy 0x1380 bytes
    0      2        POWER         174874311.770833         5.208333 src/lib/pm_runtime.c:56  	pm_runtime_put()
    0      2         PIPE         174874376.354167        64.583336 src/audio/pipeline.c:601 	pipeline_comp_trigger(), current->comp.id = 74, dir = 1
    0      2          DAI 11.74   174874380.312500         3.958333      src/audio/dai.c:536 	dai_comp_trigger(), command = 1
    0      2          DAI 11.74   174874390.156250         9.843750      src/audio/dai.c:552 	dai_comp_trigger(), START
    0      2         DMIC         174874394.010417         3.854167 rs/intel/cavs/dmic.c:1429	dmic_trigger()
    0      2         DMIC         174874397.968750         3.958333 rs/intel/cavs/dmic.c:1229	dmic_start()
    0      2         DMIC         174874407.083333         9.114583 rs/intel/cavs/dmic.c:1247	dmic_start(), dmic->fifo_b
    0      2         DMIC         174874411.510417         4.427083 rs/intel/cavs/dmic.c:1274	dmic_start(), mic_a = 1, mic_b = 1, fir_a = 1, fir_b = 1
    0      2         DMIC         174874416.197917         4.687500 rs/intel/cavs/dmic.c:1274	dmic_start(), mic_a = 0, mic_b = 0, fir_a = 0, fir_b = 0
    0      2         DMIC         174874420.520833         4.322917 rs/intel/cavs/dmic.c:1331	dmic_start(), dmic_active_fifos = 1
    0      2  SCHEDULE_LL         174874425.885417         5.364583 edule/timer_domain.c:47  	timer_domain_register()
    0      2         DMIC         174874431.927083         6.041667 rs/intel/cavs/dmic.c:1344	dmic_start(), done
    0      2          DMA         174874436.197917         4.270833 src/drivers/dw/dma.c:215 	dw_dma_start(): dma 1 channel 0 start
    0      2          EDF         174874446.510417        10.312500 edule/edf_schedule.c:65  	edf_scheduler_run()
    0      2          EDF         174874450.937500         4.427083 edule/edf_schedule.c:200 	schedule_edf_task_running()
    0      2          IPC         174874457.031250         6.093750    src/ipc/handler.c:1143	ipc: last request 1610612736 returned 0
    0      2          EDF         174874461.666667         4.635417 edule/edf_schedule.c:214 	schedule_edf_task_complete()
    0      2          EDF         174874467.083333         5.416667 edule/edf_schedule.c:65  	edf_scheduler_run()
    0      2          EDF         174874471.093750         4.010417 edule/edf_schedule.c:200 	schedule_edf_task_running()

```
The IIR has something to do with the crash though. The crash still happens if I comment out the processing

`//cd->eq_iir_func(dev, cl.source, cl.sink, cl.frames);`

But the crash can be prevented by re-programming the EQs with a pass-trough coefficient set

```
$ amixer controls | grep EQ
numid=49,iface=MIXER,name='EQIIR10.0 EQIIR_C48'
numid=51,iface=MIXER,name='EQIIR11.0 EQIIR_C16'
$ sof-ctl -n 49 -s eq/eq_iir_pass.txt 
Control size is 304.
Applying configuration "eq/eq_iir_pass.txt" into device hw:0 control numid=49.
4607827,0,88,50331648,0,0,0,0,88,2,1,0,0,0,0,4294967295,4294967295,1,1,0,0,0,0,0,0,0,0,538145694,0,32690
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 88 bytes
hdr: abi 3:0:0
4607827,0,88,50331648,0,0,0,0,88,2,1,0,0,0,0,4294967295,4294967295,1,1,0,0,0,0,0,0,0,0,538145694,0,32690
$ sof-ctl -n 51 -s eq/eq_iir_pass.txt 
Control size is 304.
Applying configuration "eq/eq_iir_pass.txt" into device hw:0 control numid=51.
4607827,0,88,50331648,0,0,0,0,88,2,1,0,0,0,0,4294967295,4294967295,1,1,0,0,0,0,0,0,0,0,538145694,0,32690
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 88 bytes
hdr: abi 3:0:0
4607827,0,88,50331648,0,0,0,0,88,2,1,0,0,0,0,4294967295,4294967295,1,1,0,0,0,0,0,0,0,0,538145694,0,32690

```

The previous saves from crash with both processing function commented out and not commented. The difference with pass-through response and normal is that with it no filter delay line is allocated. When re-programming ant other normal response like bass boost the crash happens.

singalsu, 2019-12-13:
With bisecting I found that this stops working at commit c7d01494bb74ea628e4a15fab9c0f0656b6749ff "pipeline: fix tasks initialization".

singalsu, 2019-12-13:
@tlauda Do you have idea why this impacts the alternating DMIC devices capture?

mmaka1, 2019-12-13:
@singalsu @tlauda There are also much simpler failures (single device, no switch, no stress) of DMIC capture, see #2202.

singalsu, 2019-12-16:
I'm not able to repeat issue #2202 with WHL. On WHL the panic with repeated capture from different device happens always always for me.

@tlauda Do you have thoughts for the bisect finding above? (https://github.com/thesofproject/sof/commit/c7d01494bb74ea628e4a15fab9c0f0656b6749ff)

singalsu, 2019-12-17:
I'm no more getting DSP panic when PR #2223 by @tlauda is applied.

dengyangchao, 2019-12-18:
Tried on both ICL-U RVP and CML Laptop, after applied #2223 , issue can't be reproduced.

Platforms: 
ICL-U RVP with onboard codec ALC700 in HDA mode
CML laptop with onboard codec ALC3204 in HDA mode

lgirdwood, 2019-12-18:
#2223 pending CI approval prior to merge.

PR #2223 (2019-12-19): eq_iir: eq_fir: set *_delay pointer to NULL when freed
Sets *_delay pointer to NULL when freed to prevent attempts
to free the buffer multiple times.

Fixes #2176
closed 2019-12-19 by tlauda