thesofproject/sof #2180: [BUG] Mixer in capture pipeline - DSP PANIC

zrombel, 2019-12-06:
**Describe the bug**
DSP send Panic Notification after StreamTriggerStart to capture pipeline containing mixer component. Panic occurs in second playback iteration. 
In first iteration stream are created, audio is played and captured correctly, streams are freed. 
In second iteration on same topology (components are not freed after first iteration) new streams are created and after StreamTriggerStart to mixer pipeline DSP Panic notification is send back. 
Bug is audio format agnostic, occurs on all hardware platforms.
 
**Topology**
```
+-------------------------+
| +----+   +----+   +---+ |              +----+
| |Host|-->|Buff|-->|Dai| +------------->|SSPx|-----------------+
| +----+   +----+   +---+ |              +----+                 |
+-------------------------+                                     |
                                                                |
+-------------------------+                                     |
| +----+   +----+   +---+ |              +------+               |
| |Host|-->|Buff|-->|Dai| +------------->|SSPx+1|------------+  |
| +----+   +----+   +---+ |              +------+            |  |
+-------------------------+                                  |  |
                                                             |  |
+-------------------------+                                  |  |
| +----+   +----+   +---+ |              +------+            |  |
| |Host|-->|Buff|-->|Dai| +------------->|SSPx+2|---------+  |  |
| +----+   +----+   +---+ |              +------+         |  |  |
+-------------------------+                               |  |  |
                                                          |  |  |
                                                          |  |  |
                                                          |  |  |
                                                          |  |  |
                                    +----------------+    |  |  |
                                    | +----+   +---+ |    |  |  |
                              +-----+ |Buff|<--|Dai| <----+  |  |
                              |     | +----+   +---+ |       |  |
                              |     +----------------+       |  |
                              |                              |  |
+-------------------------+   |     +----------------+       |  |
| +----+   +----+   +---+ <---+     | +----+   +---+ |       |  |
| |Host|<--|Buff|<--|Mix| <---------+ |Buff|<--|Dai| <-------+  |
| +----+   +----+   +---+ <---+     | +----+   +---+ |          |
+-------------------------+   |     +----------------+          |
                              |                                 |
                              |     +----------------+          |
                              |     | +----+   +---+ |          |
                              +-----+ |Buff|<--|Dai| <----------+
                                    | +----+   +---+ |
                                    +----------------+
```

**To Reproduce**
Python tests with slim driver:
tests/sof/fw_12_mixer/test_01_mixer_capture.py 
--playback_iterations > 1

**Reproduction Rate**
100%

**Environment**
FW: master branch commit 65de27d2cbfe7bdaaf5c45da2677e7c933dcc294 and earlier.  

**Logs**
[MixerCaptureDspPanicLog.zip](https://github.com/thesofproject/sof/files/3932023/MixerCaptureDspPanicLog.zip)

lgirdwood, 2019-12-17:
@zrombel good find. It seems like something on the mixer stop/tear down path is not being correctly handled.

jdrzaszcz, 2020-02-24:
Still occur on master: 1d273c3a9f155624144199e36a14ffafeec345ca commit

lgirdwood, 2020-02-24:
@zrombel we should attach raw topology files when reporting these errors too, useful for other testers and developers.

jdrzaszcz, 2020-02-26:
Issue doesn't occur after setting proper schedule_component_id to all capture pipelines.

closed 2020-02-26 by jdrzaszcz