thesofproject/sof #2181: [BUG] Cannot load firmware on HP Spectre x360

MayurVirkar, 2019-12-08:
**Describe the bug**
Cannot load firmware for Linux kernel version 5.4.*. Everything works fine for version 5.3.*.
As a result, speakers + mic don't work.

**To Reproduce**
Upgrade kernel to 5.4.*

**Reproduction Rate**
10/10

**Expected behavior**
The firmware to be loaded and everything to work fine.

**Impact**
Showstopper: Laptop is basically useless as I cannot make and receive calls or watch videos. etc

**Environment**
All versions of Linux 5.4.*
I havent compiled anything. I download the kernel using Ukuu.

**Screenshots or console output**
```
[    2.822759] snd_hda_intel 0000:00:1f.3: enabling device (0000 -> 0002)
[    2.824409] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, aborting probe
[    2.948148] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    2.948154] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.948277] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    2.954754] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    2.963552] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    2.963553] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    2.985657] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    2.985659] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    2.993500] HDMI HDA Codec ehdaudio0D2: Max dais supported: 3
[    3.068340] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-6cc8d
[    3.068342] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:7:0 Kernel ABI 3:10:0
[    3.068566] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    3.073361] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp1
[    3.073364] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp2
[    3.073365] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp3
[    3.073367] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Analog Playback and Capture
[    3.073368] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Digital Playback and Capture
[    3.073370] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic01
[    3.073372] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic16k
[    3.073655] sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof-tplg/sof-hda-generic-2ch.tplg failed with error -2
[    3.073658] sof-audio-pci 0000:00:1f.3: error: tplg request firmware intel/sof-tplg/sof-hda-generic-2ch.tplg failed err: -2
[    3.073704] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -2
[    3.073731] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -2
[    3.073780] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -2
[    3.073836] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -2
```

bin101, 2019-12-08:
As far as I understood the sof and hda modules cannot be loaded simultaneous.
Tray blacklisting the hda modules:

#### **`/etc/modprobe.d/blacklist.conf`**
``` 
blacklist snd_hda_intel
blacklist snd_soc_skl
```

MayurVirkar, 2019-12-09:
Thank you for your response.
I don't think that's the issue here but gave it a try anyway and no luck.
From what I understand is its either not able to identify my hardware properly or the drivers are not loaded maybe due to some signature issue. BUT I leave it to the experts.

xiulipan, 2019-12-09:
@MayurVirkar can you check do you have the tplp files in you folder `/lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg`?

```
[    3.073655] sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof-tplg/sof-hda-generic-2ch.tplg failed with error -2
[    3.073658] sof-audio-pci 0000:00:1f.3: error: tplg request firmware intel/sof-tplg/sof-hda-generic-2ch.tplg failed err: -2
```

MayurVirkar, 2019-12-09:
@xiulipan Why did I start with that? :(
You are right, I don't have those files. I have 
```
sof-hda-generic.tplg
```

xxxbxxx, 2019-12-09:
hi! 
comparing a boot log from 5.4 and 5.3, I noticed a new line:
`kernel: snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, aborting probe`

a search brought me here, and to that commit:
https://patchwork.kernel.org/patch/11050465/

so maybe the issue is that snd_hda_intel disables itselfs and expect some other module to take over?

nb: I have a spectre x360, but I am not using sof (that I'm aware of)

plbossart, 2019-12-09:
@xxxbxxx did the microphone path work with 5.3 with snd_hda_intel?
We added a dynamic detection of the DMIC (Digital Microphones) not handled by snd_hda_intel, so either the information is incorrect and snd_hda_intel should be used, or snd_hda_intel never fully worked for the capture path and you do need to move to the use of snd-sof-pci.

xxxbxxx, 2019-12-09:
> @xxxbxxx did the microphone path work with 5.3 with snd_hda_intel?

I downgraded back to 5.3 to double check, and indeed no audio input device are detected.  (which I never use, so I didn't really pay attention before..)

I also tried tried to `modprobe snd-sof-pci` which loads the module successfully (in 5.4, doesn't exist in 5.3 - vanilla arch linux)
but nothing happens, no log in dmesg or anything.
I guess I'll have to look deeper into this...

plbossart, 2019-12-09:
> I also tried tried to `modprobe snd-sof-pci` which loads the module successfully (in 5.4, doesn't exist in 5.3 - vanilla arch linux)

the module used to be call sof-pci-dev, it was renamed upstream

xxxbxxx, 2019-12-09:
So...
of course nothing happens when I run modprobe, the module was already loaded during the boot process.

It didn't find the firmware, which indeed wasn't on the disk.  Turns out (on arch-linux at least) it isn't in the core/linux-firmware package.
So I installed it from the sof-firmware package. see https://www.archlinux.org/packages/extra/any/sof-firmware/files/
And now the module finds it!

... Unfortunately, still doesn't work:

```
sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
sof-audio-pci 0000:00:1f.3: use msi interrupt mode
sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:10:0
sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
sof-audio-pci 0000:00:1f.3: firmware boot complete
sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:10:0
sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
sof-audio-pci 0000:00:1f.3: error: invalid kcontrol count 2 for volume
sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 6 name : Dmic0 stream none reply 0
sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget Dmic0
sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
```

bin101, 2019-12-09:
`sof-audio-pci 0000:00:1f.3: error: invalid kcontrol count 2 for volume`

you need at least kernel 5.5-rc1

plbossart, 2019-12-09:
> sof-audio-pci 0000:00:1f.3: error: invalid kcontrol count 2 for volume

you are using a kernel that's older than the topology files.

That said, it's just awful for users to have to track topology changes. @juimonen @ranj063 should we just ignore the second control in that case instead of throwing an error?

xxxbxxx, 2019-12-09:
ok.
so I tried to get a older version of the firmware:
ftp://ftp.alsa-project.org/pub/misc/sof/sof-firmware-1.3.1.tar.bz2
But i still get the same error.
```
sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-5dd9a
sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:7:0 Kernel ABI 3:10:0
...
sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:10:0
....
```

any idea where I can find the correct matching firmware?

plbossart, 2019-12-09:
> any idea where I can find the correct matching firmware?

It's not the firmware binary but the topology binary you need. @juimonen may be able to help here.

xxxbxxx, 2019-12-09:
oh! that's it!
so using the firmware from the archlinux package (which is just: 
ftp://ftp.alsa-project.org/pub/misc/sof/sof-firmware-1.4.1.tar.bz2 )

and overwritting the topology files from
ftp://ftp.alsa-project.org/pub/misc/sof/sof-firmware-1.3.1.tar.bz2
(folder usr/lib/firmware/intel/sof-tplg/ )

works!

xxxbxxx, 2019-12-09:
It's not perfect though:  a recording device is now detected (in pulse audio), but it doesn't seem to record anything at first glance.
And there still are errors in the log (probaly caused by  the firmware being a bit too recent for the kernel?)

```
déc. 09 22:49:25 xbx-spectre kernel: snd_hda_intel 0000:00:1f.3: enabling device (0000 -> 0002)
déc. 09 22:49:25 xbx-spectre kernel: snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, aborting probe
déc. 09 22:49:25 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
déc. 09 22:49:25 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
déc. 09 22:49:25 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: use msi interrupt mode
déc. 09 22:49:25 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
déc. 09 22:49:25 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
déc. 09 22:49:25 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
déc. 09 22:49:26 xbx-spectre kernel: HDMI HDA Codec ehdaudio0D2: Max dais supported: 3
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:10:0
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
déc. 09 22:49:26 xbx-spectre kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp1
déc. 09 22:49:26 xbx-spectre kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp2
déc. 09 22:49:26 xbx-spectre kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp3
déc. 09 22:49:26 xbx-spectre kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Analog Playback and Capture
déc. 09 22:49:26 xbx-spectre kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Digital Playback and Capture
déc. 09 22:49:26 xbx-spectre kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic01
déc. 09 22:49:26 xbx-spectre kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic16k
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: Topology: ABI 3:7:0 Kernel ABI 3:10:0
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
déc. 09 22:49:26 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
déc. 09 22:49:26 xbx-spectre kernel: HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
déc. 09 22:49:26 xbx-spectre kernel: HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
déc. 09 22:49:26 xbx-spectre kernel: HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 7:0
[... cut ...]
déc. 09 22:53:45 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
déc. 09 22:53:45 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
déc. 09 22:54:21 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
déc. 09 22:54:21 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
déc. 09 22:55:08 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
déc. 09 22:55:09 xbx-spectre kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
```

aiChaoSONG, 2019-12-10:
@xxxbxxx the "error: no reply expected, received 0x0" is actually not an error, it's more like a warning, and has no impact on anything.

xiulipan, 2019-12-10:
> you are using a kernel that's older than the topology files.
>
> That said, it's just awful for users to have to track topology changes. @juimonen @ranj063 should we just ignore the second control in [...]

@plbossart I think this would be impossible if we do not have changed the ASoC code for that. We had a discussion about this in https://github.com/thesofproject/sof/issues/2077. It was default set to allow only 1 control, and after our change in v5.5 it will accept 2 control. So you can not let the kernel to do that in the past.

Maybe we can only use the 2-control tplg for the device that need the LED control, and make our default tplg only have one control?
@ranj063 @juimonen

MayurVirkar, 2019-12-10:
Wow, Looks like I missed a lot.
@plbossart @xiulipan The mic did not work in kernel 5.3. On top of that, HP Spectre has 4 speakers, but only 2 worked on kernel 5.3

I downloaded latest SOF firmware from alsa and replaced the files in firmware directory.
This is what i get now

```
[    2.772399] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    2.772742] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    2.772749] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.772904] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    2.773100] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    2.781467] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    2.781468] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    2.796235] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    2.796238] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    2.804334] HDMI HDA Codec ehdaudio0D2: Max dais supported: 3
[    2.899119] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[    2.899122] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:10:0
[    2.899125] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[    2.899337] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    2.902555] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp1
[    2.902556] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp2
[    2.902557] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp3
[    2.902558] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Analog Playback and Capture
[    2.902559] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Digital Playback and Capture
[    2.902560] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic01
[    2.902560] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic16k
[    2.903340] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:10:0
[    2.903341] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    2.903345] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    2.904217] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    2.904219] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    2.904899] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    2.904900] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    2.905577] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    2.905578] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    2.906266] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    2.906267] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    2.906269] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    2.906270] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    2.906271] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    2.906273] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    2.906975] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    2.906977] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    2.907692] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    2.907695] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    2.908411] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    2.908530] sof-audio-pci 0000:00:1f.3: error: invalid kcontrol count 2 for volume
[    2.908533] sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 6 name : Dmic0 stream none reply 0
[    2.908535] sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget Dmic0
[    2.908615] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
[    2.908619] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[    2.908621] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[    2.908635] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -22
[    2.908652] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22
```

MayurVirkar, 2019-12-10:
Update:
Now I am on kernel 5.5-rc1. Speakers work now, but 2/4.
It detects the microphone, but it doesnt seem to record anything, ie no input.

New dmesg
```
[    2.804451] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.818056] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.818068] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    2.818407] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    2.818416] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.818597] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    2.818749] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    2.827580] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    2.827582] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    2.855099] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    2.855101] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    2.862227] HDMI HDA Codec ehdaudio0D2: Max dais supported: 3
[    2.941772] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[    2.941774] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[    2.942027] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    2.944602] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp1
[    2.944604] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp2
[    2.944604] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp3
[    2.944605] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link Analog Playback and Capture
[    2.944606] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link Digital Playback and Capture
[    2.944607] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link dmic01
[    2.944607] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link dmic16k
[    2.945476] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:11:0
[    2.945481] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    2.946328] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    2.946330] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    2.947040] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    2.947041] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    2.947719] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    2.947721] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    2.948413] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    2.948414] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    2.948416] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    2.948418] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    2.948419] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    2.948420] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    2.949126] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    2.949128] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    2.949835] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    2.949836] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    2.950587] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    2.966746] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    2.971568] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[    2.971689] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[    2.971816] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 7:0
[    2.978377] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC285: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    2.978379] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    2.978380] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    2.978381] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    2.978381] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    2.978382] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[    3.022905] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi1 <-> iDisp1 Pin mapping ok
[    3.022910] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi2 <-> iDisp2 Pin mapping ok
[    3.022914] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi3 <-> iDisp3 Pin mapping ok
[    3.022919] skl_hda_dsp_generic skl_hda_dsp_generic: Analog Codec DAI <-> Analog CPU DAI mapping ok
[    3.022925] skl_hda_dsp_generic skl_hda_dsp_generic: Digital Codec DAI <-> Digital CPU DAI mapping ok
[    3.022930] skl_hda_dsp_generic skl_hda_dsp_generic: dmic-hifi <-> DMIC01 Pin mapping ok
[    3.022934] skl_hda_dsp_generic skl_hda_dsp_generic: dmic-hifi <-> DMIC16k Pin mapping ok
[... cut ...]
[    3.025092] input: sof-hda-dsp HDMI/DP, pcm=18 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input28
[    3.025154] input: sof-hda-dsp HDMI/DP, pcm=19 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input29
[    3.025214] input: sof-hda-dsp HDMI/DP, pcm=20 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input30
[   20.399143] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   20.496626] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

juimonen, 2019-12-10:
@plbossart @xiulipan yes the issue is exactly that recording topologies with 2 controls just don't load with older kernels. I'm fine changing the default dmic topology to 1 control, however I don't know how to detect and load correct topology for more than 1 control. I don't get why we had the restriction for 1 control in the first place, really causing us problems now.

@MayurVirkar your latest dmesg looks legit. The issue for you not able to record is probably something with pulseaudio not recognizing the correct name for the microphone, don't know if it has ever recognized the dmic. You could try to record with arecord from the dmic and see if you get anything recorded. In upstream we are moving to use UCM between pulseaudio and alsa to recognize all devices properly.

MayurVirkar, 2019-12-10:
@juimonen Ahmm, need some help here.
```
$ arecord -L
default
    Playback/recording through the PulseAudio sound server
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
sysdefault:CARD=sofhdadsp
    sof-hda-dsp, 
    Default Audio Device
dmix:CARD=sofhdadsp,DEV=0
    sof-hda-dsp, 
    Direct sample mixing device
dmix:CARD=sofhdadsp,DEV=1
    sof-hda-dsp, 
    Direct sample mixing device
dmix:CARD=sofhdadsp,DEV=6
    sof-hda-dsp, 
    Direct sample mixing device
dmix:CARD=sofhdadsp,DEV=7
    sof-hda-dsp, 
    Direct sample mixing device
dsnoop:CARD=sofhdadsp,DEV=0
    sof-hda-dsp, 
    Direct sample snooping device
dsnoop:CARD=sofhdadsp,DEV=1
    sof-hda-dsp, 
    Direct sample snooping device
dsnoop:CARD=sofhdadsp,DEV=6
    sof-hda-dsp, 
    Direct sample snooping device
dsnoop:CARD=sofhdadsp,DEV=7
    sof-hda-dsp, 
    Direct sample snooping device
hw:CARD=sofhdadsp,DEV=0
    sof-hda-dsp, 
    Direct hardware device without any conversions
hw:CARD=sofhdadsp,DEV=1
    sof-hda-dsp, 
    Direct hardware device without any conversions
hw:CARD=sofhdadsp,DEV=6
    sof-hda-dsp, 
    Direct hardware device without any conversions
hw:CARD=sofhdadsp,DEV=7
    sof-hda-dsp, 
    Direct hardware device without any conversions
plughw:CARD=sofhdadsp,DEV=0
    sof-hda-dsp, 
    Hardware device with all software conversions
plughw:CARD=sofhdadsp,DEV=1
    sof-hda-dsp, 
    Hardware device with all software conversions
plughw:CARD=sofhdadsp,DEV=6
    sof-hda-dsp, 
    Hardware device with all software conversions
plughw:CARD=sofhdadsp,DEV=7
    sof-hda-dsp, 
    Hardware device with all software conversions
usbstream:CARD=sofhdadsp
    sof-hda-dsp
    USB Stream Output
```

So which device you want me to try?

Also, no, the pulse has never recognized my microphone till date.

juimonen, 2019-12-10:
@MayurVirkar so something like:
arecord -Dhw:0,6 -c 2 -r 48000 -f S16_LE test.wav

It might be that you get "device busy" because pulseaudio might be reserving the device... then you need to do the following:
1) open /etc/pulse/client.conf -> uncomment/set "autospawn=no" -> save
2) pulseaudio --kill
3) retry arecord

juimonen, 2019-12-10:
@MayurVirkar so dmic alsa devices are usually Dhw:0,6 and Dhw:0,7.

MayurVirkar, 2019-12-10:
@juimonen 
WORKS!! Both, 0,6 and 0,7. 
For the first time in the last 10 months, and I am able to record audio on my laptop on Linux! Thanks, man!

So now, the question is, how to configure PulseAudio to recognize proper input devices permanently?

juimonen, 2019-12-10:
@MayurVirkar cool! and apologies for the annoyance...

We are currently in a transition phase, which makes things quite complicated, so:
Pulseaudio has 2 methods of recognizing the audio devices: 
1) "legacy" paths/profiles 
2) UCM conf files

"legacy" is currently the standard in most distros, but upstream is at this very moment trying to move to UCM based setups. So if you want to try out the near future solution, you might need to 
update alsa-ucm-conf, alsa-lib, alsa-utils and pulseaudio to latest and greatest, probably git master and compile & install yourself. I'm involved in this UCM activity and I have no knowledge or expertise or ready solution to pulseaudio "legacy" configuration.

Hui Wang from Canonical had some solution for "legacy" way back if you want to walk on the wild side:
https://github.com/thesofproject/linux/issues/920

Some features are also dependent on the distribution, so I see little bit different behavior for example in ubuntu and fedora. So for example how different input/output source selection is automatized and so on. So even though you might get pulseaudio to recognize the dmic, there still could be some "funny" situation with mic selection.

bin101, 2019-12-10:
@MayurVirkar You could also tell pulseaudio to bind the device manually in the meantime:

in ~/.config/pulse/default.pa:
load-module module-alsa-source device=hw:0,6

replace the hw id by the device which worked for you. In my case I also had to append channels=4 otherwhise my recorded voice was kind of slow-motion.

You can also test it first before creating the file with i.e.:
pacmd load-module module-alsa-source device=hw:0,6

you should then see a device in pavucontrol or any pulseaudio gui

MayurVirkar, 2019-12-11:
Hello @bin101 
As I have two devices, when I run 
```
pacmd load-module module-alsa-source device=hw:0,6
pacmd load-module module-alsa-source device=hw:0,7
```
and check pavucontrol, it works.
but adding it to the file kills the pulseaudio server.

Secondly, they show up as different devices, any way merge those devices?

bin101, 2019-12-11:
@MayurVirkar your default.pa should only contain this:
```
load-module module-alsa-source device=hw:0,6
load-module module-alsa-source device=hw:0,7
```
Why do you want to merge those devices?
Please see this only as a temporary workaround until the desired adjustments in alsa-ucm-conf, alsa-lib, alsa-utils and pulseaudio were made like @juimonen mentioned.

kv2019i, 2019-12-13:
> That said, it's just awful for users to have to track topology changes. @juimonen @ranj063 should we just ignore the second control in that case instead of throwing an error?

I think we should. Based on our own ABI rules, minor ABI-version changes should be backward-compatible (sof/src/ipc/handler.c).

MayurVirkar, 2020-02-04:
Hello guys,
Sorry for the long silence.
So update from my side, now I am on Manjaro with Linux 5.5.0-1-mainline.
I got to all the original issue discussed here is resolved, but mic works fine, the speakers work fine.
But only 2 out of four speakers work. Top ones they dont work, bottom ones do.

Anyway to fix this issue?

plbossart, 2020-02-04:
@MayurVirkar the speakers should be handled by the Hdaudio codec.
Can you try to use
blacklist snd-sof-pci

and see if the legacy snd-hda-intel driver loads?

I suspect a platform configuration issue, not something related to processing by the SOF firmware.

MayurVirkar, 2020-02-12:
Same result with both.
Only 2 speakers out of four work..

MayurVirkar, 2020-02-21:
How can I debug it further?

lgirdwood, 2020-02-24:
@MayurVirkar I guess you have also tried to open alsamixer and tried to manually adjust the codec settings ? It could be the defaults have some speakers muted ?

xxxbxxx, 2020-02-24:
hi,

not directly related, but to track the issue bubbling up the stack:

On archlinux,  there was an upgrade to the recently released alsa.  
which broke the playback with an error in the log: 
`pulseaudio:` `UCM file does not specify 'PlaybackChannels' ...`

The fix was to compile a pulseaudio package from the git master.

lgirdwood, 2020-02-26:
Thanks @xxxbxxx , there are some updates from UCM and Pulseaudio from @juimonen that are being or have been upstreamed and some kernel patches too. I guess this will all stabilise as the distros update.

kv2019i, 2020-02-27:
@MayurVirkar @lgirdwood As the original issue (firmware loading) is now solve, maybe time to close this. Ok?

@MayurVirkar I understand the 4-speaker problem remains, but as that happens with the legacy HDA driver as well, this seems like a codec driver/mixer-setting issue not related to SOF, so this is better handled via ALSA bugzilla and list (codec driver maintainers do not watch this github).

lgirdwood, 2020-02-27:
We can reopen if needed.

closed 2020-02-27 by lgirdwood

MayurVirkar, 2020-03-03:
Hi,
The issue is back.. No microphone ditected..
I am on kernel 5.6.0-rc4-1, everything works fine on kernel 5.5.*..

`
[    3.429070] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.430402] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.430415] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.430643] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.430804] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.435788] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.564453] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    3.564456] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.567505] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    3.567508] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    3.654604] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
[    3.654613] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:12:0
[    3.654877] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    3.669658] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:12:0
[    3.669666] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    3.670475] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    3.670480] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    3.671299] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    3.671303] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    3.672094] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    3.672097] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    3.672866] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    3.672869] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    3.672870] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    3.672872] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    3.672873] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    3.672875] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    3.673624] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    3.673625] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    3.674378] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    3.674379] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    3.675187] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    3.682300] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    3.752004] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input27
[    3.752377] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input28
[    3.752497] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input29
[    3.752708] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input30
[    3.752874] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input31
**[   12.407109] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0**
[   12.492549] sof-audio-pci 0000:00:1f.3: firmware boot complete
`

kv2019i, 2020-03-03:
@MayurVirkar D'oh, sorry to hear that. Can you post full dmesg and if possible, also output of "alsa-info"? I'm asking this as the above log does not show any fatal errors:

> [   12.407109] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0**

This is harmless. There's a patch going to upstream that suppresses this trace print as it is misleading.

> [   12.492549] sof-audio-pci 0000:00:1f.3: firmware boot complete

This means the firmware has loaded, so the at least this is not a new case of the original
problem you have.

... so the problem is elsewhere.

lgirdwood, 2020-03-03:
@MayurVirkar probably best we create a new issue with full dmesg and alsa-info for the microphone.

plbossart, 2020-03-03:
> @MayurVirkar probably best we create a new issue with full dmesg and alsa-info for the microphone.

right, I've seen other reports that the dmic is not detected so we'd want to track this with a different issue. This no longer a firmware load issue

MayurVirkar, 2020-03-06:
Done :)
https://github.com/thesofproject/sof/issues/2496