thesofproject/sof #2182: [BUG] invalid kcontrol count 2

bin101, 2019-12-08:
**Describe the bug**
Hello,
I reported this issue also to the [arch linux bugtracker](https://bugs.archlinux.org/task/64768) but I don't really think this is a sof-firmware issue but maybe you already know the exact source of this issue and can support to solve it.

My system is an X1 Carbon 7th gen 

**To Reproduce**
install 5.4.x kernel from arch core repo and sof-firmware 1.4.1 from arch extra repo.

**Reproduction Rate**
Every time

**Expected behavior**
soundcard should be initiated

**Impact**
soundcard cannot be initiated

**Screenshots or console output**

```
[    3.863426] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[    3.863438] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:10:0
[    3.863440] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[    3.863659] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    3.868410] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp1
[    3.868413] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp2
[    3.868414] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp3
[    3.868416] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Analog Playback and Capture
[    3.868417] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Digital Playback and Capture
[    3.868418] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic01
[    3.868420] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic16k
[    3.869066] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:10:0
[    3.869068] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    3.869075] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    3.869790] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    3.869794] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    3.870510] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    3.870514] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    3.871185] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    3.871189] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    3.878951] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    3.878955] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    3.878958] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    3.878961] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    3.878964] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    3.878966] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    3.879667] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    3.879671] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    3.882785] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    3.882789] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    3.883522] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    3.883838] sof-audio-pci 0000:00:1f.3: error: invalid kcontrol count 2 for volume
[    3.885658] sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 6 name : Dmic0 stream none reply 0
[    3.888835] sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget Dmic0
[    3.891503] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
[    3.891511] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[    3.894675] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[    3.896435] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -22
[    3.896466] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22
```

juimonen, 2019-12-09:
@bin101 problem is that this kernel version does not support extra control (mute switch) added to dmic path in topology. So either you need newer kernel or older topology file, so there should not be anything wrong with firmware.

bin101, 2019-12-09:
@juimonen so I guess this feature will be introduced with kernel 5.5 and newer? Because currently I am using 5.4.2.

juimonen, 2019-12-09:
@bin101 that is my understanding, the patches for this are in kernel audio tree.

juimonen, 2019-12-09:
@bin101 patches are in linus' tree 5.5-rc1

bin101, 2019-12-09:
@juimonen 
Could you maybe give me a link to the patch/repository?

Because if I compare the kcontrol validation function from 5.5-rc1 and 5.4 it is still the same:
https://github.com/torvalds/linux/blob/v5.4/drivers/staging/greybus/audio_topology.c#L527
https://github.com/torvalds/linux/blob/v5.5-rc1/drivers/staging/greybus/audio_topology.c#L527

To be more precise audio_topology.c wasn't changed at all since 26th of February if I look into the 5.5-rc1 tree. Same for 5.4 tree

closed 2019-12-09 by bin101