thesofproject/sof #2190: [BUG] CML: failed to load DSP firmware after resuming form S3

keqiaozhang, 2019-12-11:
**Describe the bug**
This is the first time we have tested S3 on CML Chrome device, but unfortunately, it failed to boot DSP firmware again after resume.
This error happened in the first cycle.

**To Reproduce**
1. echo 0 > /var/lib/power_manager/suspend_to_idle
2. restart powerd
3. echo deep > /sys/power/mem_sleep 
4. suspend_stress_test -c 1 
5. check the kernel log and sof trace.

**Reproduction Rate**
100%

**Expected behavior**
Device can suspend and resume without errors.

**Impact**
IPC timed out and sof panic

**Environment**
1) OS:12607.47.0 
* CML 007 drop of firmware
2) Name of the topology file
* Topology: sof-rt5682-rt1011.tplg
3) Name of the platform(s) on which the bug is observed.
* Platform:CML

**Screenshots or console output**
[s3-dmesg.txt](https://github.com/thesofproject/sof/files/3948015/s3-dmesg.txt)
dmesg:
```
[  129.137210] sof-audio-pci 0000:00:1f.3: calling pci_pm_resume+0x0/0xc2 @ 4061, parent: pci0000:00
[  129.137219] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[  129.137224] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[  129.137229] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[  129.137371] ieee80211 phy0: calling wiphy_resume+0x0/0x153 [cfg80211] @ 4091, parent: 0000:00:14.3
[  129.141517] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  129.141520] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[  129.141524] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[  129.141527] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[  129.141531] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  129.141534] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[  129.141628] sof-audio-pci 0000:00:1f.3: loading firmware
[  129.141633] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
[  129.141637] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[  129.141645] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[  129.141649] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
[  129.141650] sof-audio-pci 0000:00:1f.3: booting DSP firmware
[  129.141742] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  129.141763] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  129.141766] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
[  129.141768] sof-audio-pci 0000:00:1f.3: periods:1
[  129.141829] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[  129.141837] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
[  129.141840] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[  129.141864] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[  129.142303] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[  129.142307] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010e0e successful
[  129.142311] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x6000021 successful
[  129.142337] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  129.144276] iwlwifi 0000:00:14.3: Applying debug destination EXTERNAL_DRAM
[  129.245276] nvme nvme0: Shutdown timeout set to 8 seconds
[  129.294743] iwlwifi 0000:00:14.3: FW already configured (0) - re-configuring
[  129.299494] ieee80211 phy0: wiphy_resume+0x0/0x153 [cfg80211] returned 0 after 158307 usecs
[  129.349313] usb usb2: usb_dev_resume+0x0/0x14 returned 0 after 207324 usecs
[  129.349341] usb 2-3: calling usb_dev_resume+0x0/0x14 @ 4094, parent: usb2
[  129.349346] usb 2-3: usb_dev_resume+0x0/0x14 returned 0 after 0 usecs
[  129.349364] scsi host1: calling scsi_bus_resume+0x0/0x19 @ 4082, parent: 2-3:1.0
[  129.349370] scsi host1: scsi_bus_resume+0x0/0x19 returned 0 after 0 usecs
[... 196 lines cut ...]
    0      2          IPC         161023236.510417         4.427083    src/ipc/handler.c:899    ipc: pipe 8 comp 46 -> new (type 2)
    0      2          DAI         161023258.229167        21.718750      src/audio/dai.c:157    dai_new()
    0      2         DMIC         161023267.760417         9.531250 rs/intel/cavs/dmic.c:1511   dmic_probe()
    0      2        POWER         161023282.395833        14.635417 avs/lib/pm_runtime.c:107    dis-dmic-clk-gating index 1 CLKCTL c0000006
    0      2          DAI         161023291.979167         9.583333        src/lib/dai.c:70     dai_get(), d = 0xbe08441c, sref = 1
    0      2          DMA         161023297.760417         5.781250        src/lib/dma.c:126    dma_get() ID 0 sref = 5 busy channels 0
    0      2          IPC         161023768.645833       470.885406    src/ipc/handler.c:957    ipc: pipe 8 -> new
    0      2         PIPE         161023816.302083        47.656250 src/audio/pipeline.c:46     pipeline_new()
    0      2          IPC         161024132.135417       315.833344    src/ipc/handler.c:645    ipc: pm core mask 0x1 -> enable
    0      2          IPC         161024643.593750       511.458344    src/ipc/handler.c:896    ipc: hdr 0x30010000 tx (220) > rx (28)
    0      2          IPC         161024648.177083         4.583333    src/ipc/handler.c:899    ipc: pipe 9 comp 48 -> new (type 15)
    0      2      KEYWORD         161024670.208333        22.031250 /audio/detect_test.c:264    test_keyword_new()
    0      2          IPC         161025230.677083       560.468750    src/ipc/handler.c:896    ipc: hdr 0x30010000 tx (200) > rx (28)
    0      2          IPC         161025237.031250         6.354167    src/ipc/handler.c:899    ipc: pipe 9 comp 49 -> new (type 17)
    0      2     SELECTOR         161025259.114583        22.083334 /selector/selector.c:99     selector_new()
    0      2          IPC         161025661.875000       402.760406    src/ipc/handler.c:929    ipc: pipe 9 buffer 50 -> new (0x1400 bytes)
    0      2       BUFFER         161025684.114583        22.239584   src/audio/buffer.c:27     buffer_new()
    0      2          IPC         161026012.343750       328.229156    src/ipc/handler.c:929    ipc: pipe 9 buffer 51 -> new (0x1400 bytes)
    0      2       BUFFER         161026035.625000        23.281250   src/audio/buffer.c:27     buffer_new()
    0      2          IPC         161026476.458333       440.833344    src/ipc/handler.c:957    ipc: pipe 9 -> new
    0      2         PIPE         161026515.833333        39.375000 src/audio/pipeline.c:46     pipeline_new()
    0      2          IPC         161026935.885417       420.052094    src/ipc/handler.c:645    ipc: pm core mask 0x1 -> enable
    0      2          IPC         161027276.927083       341.041656    src/ipc/handler.c:996    ipc: comp sink 2, source 0  -> connect
```
</details>

ranj063, 2019-12-11:
@keqiaozhang is S3 POR on CML chrome? I have a feeling this might need coreboot changes. @sathyap-chrome can you please comment?

sathyap-chrome, 2019-12-11:
@ranj063  S3 is not POR but we always try S3 cycling to ensure both sleep flows works as expected.
Can you elaborate more on the changes you feel needed.

Does S3 issue in TGL have same / similar signature ?

ranj063, 2019-12-11:
@sathyap-chrome I'm not sure what changes are needed. But wasn't there a point when S0ix was not enabled and S3 was the default? Do you know what was.dine to enable S0ix?

sathyap-chrome, 2019-12-11:
S3 or Soix is more decision in powerd and some userspace controls.
And S3 flow is tested on minimal setups every weekly test cycle.

For us the issue happened on only 1 board till now. Since keyiao is able to reproduce, its worth trying to fix if its a genuine error

ranj063, 2019-12-11:
@sathyap-chrome from the logs it doesnt look like the system enter S3 successfully at all no? 
@keqiaozhang can you please try adjusting the min and max value for suspend and see if the system stays in S3 for that long before resuming?

sathyap-chrome, 2019-12-11:
@ranj063  tried setting min 20 seconds in suspend.
suspend_stress_test -c 1 --suspend_min=20 - still issue is seen.

sathyap-chrome, 2019-12-11:
I tested with upstream/cml-007-pm-ctx-save FW too and still see same issue.
let me know if any more data is needed.

mmaka1, 2019-12-11:
The error status 06000021 reports the ROM is waiting for CSE IPC reset sequence to complete. Looks like incorrect timing between CSE and ADSP ROM transition from coming out of S3.

ranj063, 2019-12-11:
> @ranj063 tried setting min 20 seconds in suspend.
> suspend_stress_test -c 1 --suspend_min=20 - still issue is seen.

@sathyap-chrome what do you mean by issue still seen? Do you mean that the system stays in S3 for 20s and then fails to resume after ? Or it doesnt manage to stay in S3 for 20s?

sathyap-chrome, 2019-12-11:
I mean the IPC timeout issue is still seen.
@mmaka1 comment may be true as we are not seeing this on all devices but a few.

let me know what could be next step or debug idea.

aiChaoSONG, 2019-12-12:
Test 2000 times with the procedures mentioned here with keyon’s backport kernel (https://github.com/thesofproject/linux/tree/chrome-4.19-backport) and 008 FW on ChromeOS, not reproduced.

keqiaozhang, 2019-12-12:
@sathyap-chrome 
For 008 recipe, snd_soc_skl must be blacklisted, or you will hit tplg failure like 
```
cml_rt1011_rt5682 cml_rt1011_rt5682: ASoC: CODEC DAI intel-hdmi-hifi1 not registered
```

sathyap-chrome, 2019-12-12:
@keqiaozhang yes. this change will be part of 008 recipe when we push to chrome.

So if we use 008 recipe, you are not seeing the S3 issue ?

sathyap-chrome, 2019-12-12:
@aiChaoSONG @keqiaozhang can you check also if its FW / kernel that might fix this ?

Can we take 007 FW and only move the kernel and see if issue happens or not .

sathyap-chrome, 2019-12-12:
I tested with same (https://github.com/thesofproject/linux/tree/chrome-4.19-backport) but using FW and tplg from 007 - issue is still reproduced. So maybe its the FW fixes of 008 that is helping this fix ?

sathyap-chrome, 2019-12-12:
@aiChaoSONG @keqiaozhang 

 I see the error my device even after using 008 FW and kernel
[   64.367374] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[   64.367395] sof-audio-pci 0000:00:1f.3: error: status = 0x001d003c panic = 0x00000000

Note that the error happens on only a few boards, so we should first reproduce the issue with 007 and on the same board try 008 Fw 7 kernel.

sathyap-chrome, 2019-12-12:
can you try S3 on ubuntu with sof-dev ?

aiChaoSONG, 2019-12-12:
@sathyap-chrome not reproduced this issue on ubuntu with sof-dev + master

keqiaozhang, 2019-12-12:
@sathyap-chrome 
We did some debugging this afternoon and found that this is not a sof related issue but a coreboot regression. we tried the old version(before R80) on same devices and issue is not reproducible. Please try it on your devices.

lgirdwood, 2019-12-16:
@sathyap-chrome please close if confirmed coreboot regession.

sathyap-chrome, 2019-12-17:
@lgirdwood   i have kept it open as i'm not able to see similar results - i also got pushed back to see if we can identify why this could be an issue and how to avoid. We checked that coreboot is not having any relavant fixes in this area.

Will debug further - if we identify root cause, then will close.

keyonjie, 2019-12-18:
the log 'unknown ROM status value 06000021' is valuable, @sathyap-chrome do you see the same on your side?

sathyap-chrome, 2019-12-18:
the error signature is same on my side and @keqiaozhang side for S3. I can share more logs if needed

keyonjie, 2019-12-18:
I think this issue happens only on 007 recipe?
Can you please cherry-pick an upstream commit from https://git.kernel.org/pub/scm/linux/kernel/git/broonie/sound.git/log/?h=for-next and retry?

184fdfca4ba63d57a668 ASoC: SOF: Intel: hda: correct ROM state mask

sathyap-chrome, 2019-12-18:
I will try SOF v0.8 and get back.
But i tried the patch suggested on top of SOF v0.7 stack we have and the issue is still seen on failing CB images.

sathyap-chrome, 2019-12-18:
With SOFv0.8 based patches on chrome tree also this issue is seen on failing . attached logs for both
a. dmesg with SOF v0.7 + only patch taken
b. dmesg with SOF v0.8
[dmesg_sof008.txt](https://github.com/thesofproject/sof/files/3977765/dmesg_sof008.txt)
[dmesg_logs_rom_fix.txt](https://github.com/thesofproject/sof/files/3977766/dmesg_logs_rom_fix.txt)

ClarexZhou, 2020-01-13:
Close for S3 is not supported with ubuntu os. Close this issue and check the case after R81 branch out with latest SOF driver and FW.

closed 2020-01-13 by ClarexZhou