thesofproject/sof #2201: [BUG] topology sof-bdw-codec.m4 (sof-bdw-rt5640.conf)

perexg, 2019-12-12:
It seems that there are two merged compound blocks SectionGraph.SSP0 in sof-bdw-rt5640.conf:

<pre>
SectionGraph."SSP0" {
        index "1"
        lines [
                "SSP0.OUT, , BUF1.3"  
        ]
}
</pre>

<pre>
SectionGraph."SSP0" {
        index "2"
        lines [
                "BUF2.0, , SSP0.IN"
        ]
}
</pre>

The current configuration loader (alsatplg) parses this like:

<pre>
SectionGraph.SSP0 {
        index '2'
        lines {
                0 'SSP0.OUT, , BUF1.3'
                0 'BUF2.0, , SSP0.IN'
        }
}
</pre>

It's definitely wrong, because the compounds must have unique keys, so after I fixed that in alsa-lib, the result is like:

<pre>
SectionGraph.SSP0 {
          index '2'
          lines {
                  0 'SSP0.OUT, , BUF1.3'
                  1 'BUF2.0, , SSP0.IN'
          }
}
</pre>

But the index field is merged (the default parser mode), of course. It looks like an issue.

perexg, 2019-12-12:
https://github.com/alsa-project/alsa-lib/commit/9980e18c3c99da42a90a98b140a0ffe7d0f92cd3

kv2019i, 2019-12-13:
@ranj063 @juimonen  @xiulipan -- can you take a look at this BDW issue?

juimonen, 2019-12-13:
@kv2019i quickly checked yesterday, but was not able immediately to understand the issue... so is the parsing wrong in alsa-lib, but still this topology is working in bdw devices?

perexg, 2019-12-13:
The problem is that 'index' first index value is lost. It's not only a problem for bdw - it seems that you define SectionGraph.SSP0 mutliple times in many configuration files (thus 'index' entries are overwriten while [] array is merged). It may cause problems.

perexg, 2019-12-13:
BTW: A quick question: Is the order of the defined structure important? Or it does not matter? This question is, because if I sort the compound identifiers, the conf file has the same size, but the binary topology file size differs.

perexg, 2019-12-13:
The lines {} block is probably fine for old and new alsa-lib parser, because the index values used internally for the [] array are usually skipped. The problem is only that you lost the first index field initialization (`index '1'` line).

ranj063, 2019-12-13:
@perexg Could you please check if #2207 fixes the issue for you?

perexg, 2019-12-13:
I found another issue in the topology library which explains the different output sizes for the binary files. The fix is here: https://github.com/alsa-project/alsa-lib/commit/2b50b594dcbf69f8dcc6e8cf673748f7063c3c17

lgirdwood, 2019-12-16:
@perexg  IIUC, it looks like the "SSP0" identifier is used to create 2 compound graphs (one is used for playback and the other for capture) but the parser will try and merge this into a single graph (and drop the index). Looks like #2207 should fix this ?

closed 2020-02-04 by lbetlej