thesofproject/sof #2202: [BUG][TGL]Ipc timed out for 0x60050000 size 12 when start arecord via DMIC

dengyangchao, 2019-12-13:
**Describe the bug**
Ipc timed out for 0x60050000 size 12 when start arecord via DMIC.

**To Reproduce**
1.Reboot system via"sudo reboot".
2.Wait runtime PM to suspend.
3.arecord via command"arecord -Dhw:0,7 -r48000 -c4 -fs16_le test.wav -vv -i"

**Reproduction Rate**
2/4
Tried 4 times and issue occur 2 times

**Expected behavior**
Arecord via dmic normal
**Impact**
occur ipc timed out

**Environment**
Firmware: 98d7ab21 https://github.com/thesofproject/sof/commits/master
Kernel: c985ebd2 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-4ch.tplg same as firmware
platform：TGL-U RVP with onboard codec ALC711 in HDA mode

<details>
 <summary>Dmesg</summary>

````
[   63.046385] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[   63.046596] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[   63.046600] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[   63.046665] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[   63.047191] sof-audio-pci 0000:00:1f.3: pcm: open stream 7 dir 1
[   63.047193] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[   63.047194] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[   63.047195] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[   63.047380] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 7 dir 1
[   63.047382] sof-audio-pci 0000:00:1f.3: generating page table for 0000000005cfa122 size 0xf9c0 pages 16
[   63.047387] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   63.047403] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   63.047405] sof-audio-pci 0000:00:1f.3: period_bytes:0x3e40
[   63.047405] sof-audio-pci 0000:00:1f.3: periods:4
[   63.047416] sof-audio-pci 0000:00:1f.3: stream_tag 2
[   63.047420] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   63.047662] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   63.047663] sof-audio-pci 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 790528
[   63.047882] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 7 dir 1 cmd 1
[   63.048134] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[   63.048138] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   63.048326] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   63.556494] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 7 dir 1 cmd 0
[   63.556501] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   64.060374] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[   64.060381] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[   64.060388] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   64.060496] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   64.060505] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000003 rirb 00
[   64.060507] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   64.060514] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   64.060516] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   64.060534] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[   64.060540]  DMIC16kHz: ASoC: trigger FE cmd: 0 failed: -110
[   64.060567] sof-audio-pci 0000:00:1f.3: pcm: free stream 7 dir 1
[   64.060572] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   64.060582] sof-audio-pci 0000:00:1f.3: error: trace IO error
[   64.564416] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[   64.564422] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[   64.564436] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   64.564540] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   64.564549] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   64.564551] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   64.564557] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   64.564559] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   64.564661] sof-audio-pci 0000:00:1f.3: pcm: close stream 7 dir 1

````
</details>

[dmesg.log](https://github.com/thesofproject/sof/files/3958820/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3958821/logger.log)

dengyangchao, 2019-12-13:
Issue also can reproduce via below method:
1.when start DMIC Arecord, use "ctrl+z" and "fg" to switch background and foreground, then repeat.
2.when start DMIC arecord, use "ctrl+c" to stop arecord then re-arecord via dmic, repeat the steps.

dengyangchao, 2019-12-13:
Issue can reproduce on CML laptop with onboard cedec ALC3204 in HDA mode.
Start arecord via headset(HW:0,0), then use "ctrl+z" and "fg" to switch background and foreground

Firmware: f5bcc4bb https://github.com/thesofproject/sof/commits/master
Kernel: 9c0401ca https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-2ch.tplg same as firmware
Platform: CML laptop with onboard cedec ALC3204 in HDA mode

[dmesg.log](https://github.com/thesofproject/sof/files/3959814/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3959815/logger.log)
[logger-error-trace.log](https://github.com/thesofproject/sof/files/3959816/logger-error-trace.log)

mmaka1, 2019-12-13:
Is it a regular stream close that fails or consequence of io error handling after 500ms t/o by arecord?

dengyangchao, 2019-12-16:
> Is it a regular stream close that fails or consequence of io error handling after 500ms t/o by arecord?

When issue occur, arecord shows I/O error

lgirdwood, 2019-12-16:
@dengyangchao can you try with latest master with HiFi3 disabled. See #2191

singalsu, 2019-12-16:
I'm not able to repeat this issue on WHL that I have myself (hw:0,0 hw:0,6 hw:0,7 with repeated ctrl-z and fg) . I tried both gcc and xt-xcc builds. The bug was not filed for WHL though.

dengyangchao, 2019-12-17:
Issue can't reproduced, since #2191 merged.Close

closed 2019-12-17 by dengyangchao