thesofproject/sof #2203: [BUG][CML] Pulse Noise on speaker pipeline in pause/release test

aiChaoSONG, 2019-12-13:
**Describe the bug**
run aplay via speaker pipeline,  and press Space to pause/release playback, you can here pulse noise(sounds like "pa"). check dmesg and sof-logger, no clear errors found.

**To Reproduce**
1. run `aplay -Dhw:0,5 -f dat -i -vvv -c 2 ./longwave.wav`
2. Press SPACE to pause/release playback

***MAKE SURE WAVE FILE CAN OUTPUT SOUND, DO NOT PLAY /dev/zero***

if not reproduced with above procedures, please try the script attached. Because the sound of press space may cover the pulse noise.

**Reproduced Rate**
should be very easy to reproduce on the platform described below

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev {9c0401ca}
    * SOF: master {f5bcc4bb}
2) Name of the topology file
    * Topology: {sof-cml-rt1011-rt5682.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {CML with rt5682 and rt1011 codecs}

**Error Logs**
No clear error seen in logs

[sof-test-pause-release.zip](https://github.com/thesofproject/sof/files/3959180/sof-test-pause-release.zip)
[spk_pulse_noise_pause_release_soflogger.log](https://github.com/thesofproject/sof/files/3959181/spk_pulse_noise_pause_release_soflogger.log)
[spk_pulse_noise_pause_release_dmesg.log](https://github.com/thesofproject/sof/files/3959182/spk_pulse_noise_pause_release_dmesg.log)

lgirdwood, 2019-12-16:
@aiChaoSONG can you paste in audacity screenshots showing the pop (palying sine wave). Please also attach small wavs of pop during sine wave.

aiChaoSONG, 2019-12-17:
> @aiChaoSONG can you paste in audacity screenshots showing the pop (palying sine wave). Please also attach small wavs of pop during sine wave.

@lgirdwood I think it is a little bit hard to loop back for the speaker(not the same SSP as headphone and DMIC), and I tried sof-cnl-nocodec.tplg, there are some error to load this topology. I will ask some veterans for suggestions, and any suggestions from you?

aiChaoSONG, 2019-12-18:
@lrgirdwo @mmaka1 I run some test with nocodec tplg, and the pause/release interval is 0.2s, and there is surely some pulse here, please see the picture and attached wave. seems we lost some samples. but if we change pause/release interval to 0.5s, things are much better, see attached wave.
[image]
[image]
[image]
[headset_interval_0.2s.zip](https://github.com/thesofproject/sof/files/3976832/headset_interval_0.2s.zip)
[headset_interval_0.5s.zip](https://github.com/thesofproject/sof/files/3976833/headset_interval_0.5s.zip)

lgirdwood, 2019-12-18:
@aiChaoSONG potentailly looks like stale data in the DAI FIFO (that's replayed at unpuase). What interface HDA, SSP are you playing from ?
@tlauda it could be we are pausing the DAI prior to FIFO being empty ?

aiChaoSONG, 2019-12-19:
@lgirdwood , it's SSP

lgirdwood, 2019-12-19:
@tlauda SSP trailing bytes ?

tlauda, 2020-03-13:
@ktrzcinx @aiChaoSONG I see some fixes have been merged. Do you still see the issue?

aiChaoSONG, 2020-03-13:
@tlauda I can not hear the noise, but we still have very tiny amount of data lost, if this doesn't matter, we can close this now.
[image]

tlauda, 2020-03-13:
@aiChaoSONG This is expected on SSP. Transfer can end in the middle of the link list item and next one starts from the next item, so some small amount of data is lost.

aiChaoSONG, 2020-03-13:
@tlauda Fine, let's close this.
close for fixed by #2364

closed 2020-03-13 by aiChaoSONG