thesofproject/sof #2208: [SDW][CML]IO errors occurred when do aplay with PA on

YvonneYang2, 2019-12-16:
**Describe the bug**
Aplay cannot work and shows IO error with PA on. Issue cannot be reproduced with PA off.  
Test it with FW second commit cc4c356, aplay can work normally. So seems issue occurred by commit 79c682b

**To Reproduce**
1. Enable the PA and reboot.
2. Aplay via "aplay -Dhw:0,2 -r48000 -c2 -fs32_le long.wav -vv -i"

**Reproduction Rate**
100%

**Expected behavior**
Aplay normally.

**Impact**
Aplay failed and showed IO error.

**Environment**
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: cf2660a 
FW: https://github.com/thesofproject/sof/commits/cml-010-drop-stable commit: 79c682b 
tplg: Same with FW branch, sof-cml-rt711-rt1308-rt715.tplg
platform: CML-U Laptop with codec ALC711, RT1308, RT715 in SDW mode

[kern.log](https://github.com/thesofproject/sof/files/3966145/kern.log)
[logger.log](https://github.com/thesofproject/sof/files/3966146/logger.log)
[error___logger.log](https://github.com/thesofproject/sof/files/3966147/error___logger.log)

YvonneYang2, 2019-12-16:
@slawblauciak Could you please have a look? Thanks ~

slawblauciak, 2019-12-16:
Can you please try this PR? https://github.com/thesofproject/sof/pull/2210

YvonneYang2, 2019-12-16:
Tested it with cml-010-drop-stable(79c682b) + #2210 , issue cannot be reproduced.

YvonneYang2, 2019-12-16:
Since #2210 has been merged into https://github.com/thesofproject/sof/commits/cml-010-drop-stable, aplay can work normally. But with #2210, dmesg will always print same info like :
```
[ 1469.891339] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1469.891492] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1470.391337] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1470.391492] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1470.891181] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1470.891606] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1471.391338] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1471.391505] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1471.891115] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1471.891336] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1472.391165] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1472.391580] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1472.891419] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1472.891850] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1473.391347] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1473.391532] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1473.891133] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
```

[error_trace.log](https://github.com/thesofproject/sof/files/3967097/error_trace.log)
[dmesg.log](https://github.com/thesofproject/sof/files/3967098/dmesg.log)
[logger_full.log](https://github.com/thesofproject/sof/files/3967100/logger_full.log)

YvonneYang2, 2019-12-16:
Since original issue cannot be reproduced, close it.  Track https://github.com/thesofproject/sof/issues/2208#issuecomment-565976490 in https://github.com/thesofproject/sof/issues/2211

closed 2019-12-16 by YvonneYang2