thesofproject/sof #2209: [BUG][ICL] TriggerStart IPC timeout on pipeline with ALH DAI.

zrombel, 2019-12-16:
**Describe the bug**
After creating topology with playback and capture pipelines containing ALH DAI component, when creating and deleting PcmStreams multiple times (topology is not destroyed), when sending IPC TrigStart to playback pipelines FW doesn't reply. Bug occurs only with audio formats 16b/16b where container size is converted to 32b on DAI component. May be connected with PR https://github.com/thesofproject/sof/pull/2029

**Topology**

```
                pipe_plb
                +------------------------+
+----------+    | +------+   +---------+ |     +-------+      +------+
| Host plb |----->|Buffer|-->| Dai ALH | |---->| PDI A | ---> | DP A |
+----------+    | +------+   +---------+ |     +-------+      +------+
                +------------------------+                       ||
                                                                 ||
                pipe_cap                                         ||
                +------------------------+                       ||
+----------+    | +------+   +---------+ |     +-------+      +------+
| Host cap |<-----|Buffer|<--| Dai ALH | |<----| PDI B |<---- | DP B |
+----------+    | +------+   +---------+ |     +-------+      +------+
                +------------------------+
```

**To Reproduce**
Slim Driver python test:
16_01_TestAlhLoopbackVolume48000Hz16b16b2ch with --playback_iterations>=100

**Reproduction Rate**
5%

**Environment**
1) Sof master branch f5bcc commit

Log:
[IclAlhTrigStartIpcTimeout.zip](https://github.com/thesofproject/sof/files/3966607/IclAlhTrigStartIpcTimeout.zip)

lgirdwood, 2019-12-16:
@zrombel #2029 now merged, is this now resolved ?

zrombel, 2019-12-16:
No, this bug is still reproducible and may be caused by PR #2029 .

lgirdwood, 2019-12-18:
@zrombel ah sorry, I mean the subsequent patch to #2029 that disables the HiFi3, but I guess we are suspecting the same issue.

emilchudzik, 2020-02-25:
Double checked issue on two platforms ICL RVP. 
There is no repro on commits: c35e1aa (24.02) and 8b59506 (25.02)

closed 2020-02-25 by emilchudzik