thesofproject/sof #2211: [BUG][CML] Dmesg print same info all the time.

YvonneYang2, 2019-12-16:
**Describe the bug**
With https://github.com/thesofproject/sof/commits/cml-010-drop-stable (c6f349e) , System boot up, then check the dmesg. Dmesg will print same info all the time. 
Function will not be effected.

Checked it with  https://github.com/thesofproject/sof/commits/cml-010-drop-stable (cc4c356 ), issue cannot be reproduced.  

**To Reproduce**
1. System boot up.
2. Check the dmesg.

**Reproduction Rate**
100%

**Expected behavior**
System will not print same info all the time

**Impact**
System print same info all the time.

**Environment**
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: cf2660a 
FW: https://github.com/thesofproject/sof/commits/cml-010-drop-stable commit: c6f349e 
tplg: Same with FW branch, sof-cml-rt711-rt1308-rt715.tplg
platform: CML-U Laptop with codec ALC711, RT1308, RT715 in SDW mode

Dmesg
```
[  386.558299] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  386.558330] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  391.058349] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  391.058381] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  392.058398] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  392.058429] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  392.558330] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  392.558356] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  395.058399] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
```

error trace
```
  0      1           SA          19592782.552083  19592782.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 26228
    0      1           SA           3602732.552083              nan      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25618
    0      1           SA           3610138.020833      7405.468750      src/lib/agent.c:55  	validate(), ll drift detected, delta = 26005
    0      1           SA           3612655.468750      2517.447998      src/lib/agent.c:55  	validate(), ll drift detected, delta = 26366
    0      1           SA           3618873.697917      6218.229004      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25643
    0      1           SA           3622649.739583      3776.041748      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25635
    0      1           SA           3701457.395833     78807.656250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25623
    0      1           SA           3866482.343750    165024.953125      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25638
```

[dmesg.log](https://github.com/thesofproject/sof/files/3967178/dmesg.log)
[error_trace.log](https://github.com/thesofproject/sof/files/3967179/error_trace.log)
[logger_full.log](https://github.com/thesofproject/sof/files/3967180/logger_full.log)

lgirdwood, 2019-12-16:
@YvonneYang2 can you recreate this as a kernel issue. i.e. we need to rate limit the trace IPC debug.

mengdonglin, 2019-12-17:
@Liam, I created kernel feature https://github.com/thesofproject/linux/issues/1632 to implement rate limit the trace IPC debug. So I re-open this issue to track why FW is sending so many trace messages.

mengdonglin, 2019-12-17:
@YvonneYang2 Please help to test tip of latest https://github.com/thesofproject/sof/commits/cml-010-drop-stable, commit https://github.com/thesofproject/sof/commit/5e5f69eb5ad14322a5f4599860a40f8009393d89 (dynamic clock change for waiti, fix for CML/CFL system hang by @jajanusz ). Now it's on top of commit https://github.com/thesofproject/sof/commit/cc4c356d7db464388ca7995c38b93c302cf07868 (disable hifi3 pcm conversion)

YvonneYang2, 2019-12-17:
Tested it with latest  https://github.com/thesofproject/sof/commits/cml-010-drop-stable (top commit: 5e5f69e ), issue cannot be reproduced. 

Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: cf2660a 
FW: https://github.com/thesofproject/sof/commits/cml-010-drop-stable commit: 5e5f69e 
tplg: Same with FW branch, sof-cml-rt711-rt1308-rt715.tplg
platform: CML-U Laptop with codec ALC711, RT1308, RT715 in SDW mode

liam, 2019-12-17:
This Liam isn't interested. Please do some research on how mentions work on
GitHub.

juimonen, 2019-12-17:
@liam yes obviously confusing you with our Liam @lgirdwood. I didn't find quickly a easy way to remove just you from this issue, but I guess you should be able to unsubscribe yourself (from the right sidebar hopefully). Sorry about the extra noise/effort for you and welcome back if audio stuff starts to interest you...

liam, 2019-12-17:
Thanks. I get so many of these mails that they all annoy me. Nothing
personal of course.

lgirdwood, 2019-12-18:
@mengdonglin this is the kernel printing out that it's received a new trace IPC (which indicates lots of trace messages). Only the kernel can rate limit this IPC log (as it's not FW generated trace). I've closed as it's only fixable in the kernel.

closed 2019-12-18 by lgirdwood