thesofproject/sof #2213: [BUG] TriggerStart IPC timeout in bit depth conversion stress test.

jdrzaszcz, 2019-12-16:
**Describe the bug**
After creating topology with playback and capture pipelines with bit depth conversion on DAI, when creating and deleting PcmStreams multiple times (topology is not destroyed), when sending IPC TrigStart to playback pipelines FW doesn't reply. 

**Topology**
+-------------------------------+
| +------+   +------+   +-----+ |  +-----+
| | Host |-->|Buffer|-->| Dai |--->| SSP |
| +------+   +------+   +-----+ |  +-----+
+-------------------------------+    | |
                                                      | |
pipe_cap                                        | |
+-------------------------------+    | |
| +------+   +------+   +-----+ |  +-----+
| | Host |<--|Buffer|<--| Dai |<---| SSP |
| +------+   +------+   +-----+ |  +-----+
+-------------------------------+

**To Reproduce**
Slim Driver python test:
On APL, ICL:
test_13_playback_bitdepth_conversion with --playback_iterations>=100
On APL, ICL:
test_13_playback_bitdepth_conversion with --playback_iterations>=1000

Example:
02_13_TestPtpPlaybackBitDepthConversion16000Hz16b16b4ch
02_13_TestPtpPlaybackBitDepthConversion48000Hz32b32b4ch

**Reproduction Rate**
50 %

**Environment**
Sof master branch 62dbfcff5cd20f0de688d817c8a070ad94e4c3c5 commit

Example log from APL:

[2019_12_16_12_16_50.zip](https://github.com/thesofproject/sof/files/3967792/2019_12_16_12_16_50.zip)
[2019_12_16_11_20_23.zip](https://github.com/thesofproject/sof/files/3967812/2019_12_16_11_20_23.zip)
[2019_12_16_11_48_15.zip](https://github.com/thesofproject/sof/files/3967813/2019_12_16_11_48_15.zip)

slawblauciak, 2020-01-27:
Does this still occur?
I wonder if it might be somehow related to #2298

slawblauciak, 2020-01-27:
Does this still occur on master?

jdrzaszcz, 2020-01-28:
Yes, still occur, but the reproduction rate is lower and playback_iterations has to be over 1000 for all platforms.
I checked it with 0e83f7d094630c3cdbc93a138ccc91efcd4c7ffa commit.

lgirdwood, 2020-01-28:
@slawblauciak @jdrzaszcz fwiw, @mrajwa made an interesting observation https://github.com/thesofproject/sof/pull/2329 that may impact this use case too.

lgirdwood, 2020-02-21:
@jdrzaszcz any update, are we still seeing this ? @tlauda it looks from the tail of the last log that the IPC logic get in a muddle prior to the timeout ?
```
[0m    0      2          SSP          33416300.677083         3.489583 src/drivers/intel/cavs/ssp.c:877 	ssp_trigger() cmd 1[0m
[0m    0      2          SSP          33416304.791667         4.114583 src/drivers/intel/cavs/ssp.c:806 	ssp_start()[0m
[0m    0      2      unknown          33416355.156250        50.364582 src/schedule/timer_domain.c:47  	timer_domain_register()[0m
[31m    0      1           SA          33417142.343750       787.187500      src/lib/agent.c:55  	validate(), ll drift detected, delta = 20277[0m
[31m    0      1          IPC          34046050.677083    628908.312500    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34088050.625000     41999.949219    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34130050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34172050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34256050.625000     84000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34298050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34340050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34382050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34466050.625000     84000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34508050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34550050.729167     42000.105469    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34634050.625000     83999.898438    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34676050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34718050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34802050.677083     84000.054688    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34844050.625000     41999.949219    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34886050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34928050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          34995142.708333     67092.085938    src/ipc/handler.c:1275	ipc: msg hdr for 0x90020000 not found replace 1[0m
[31m    0      1          IPC          35012050.625000     16907.916016    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          35054050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          35096050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          35138050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          35180050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          35264050.625000     84000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          35306050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          35348050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          35432050.625000     84000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          35474050.625000     42000.000000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[31m    0      1          IPC          35511106.822917     37056.199219    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0004 not found replace 0[0m
[31m    0      1          IPC          35516050.572917      4943.750000    src/ipc/handler.c:1275	ipc: msg hdr for 0x600a0000 not found replace 0[0m
[... cut ...]
[0m    0      2          DMA          36443810.208333         3.593750 src/drivers/dw/dma.c:366 	dw_dma_stop(): dma 0 channel 0 stop[0m
[0m    0      2          SSP          36443815.000000         4.791667 src/drivers/intel/cavs/ssp.c:877 	ssp_trigger() cmd 0[0m
[0m    0      2          SSP          36443862.031250        47.031250 src/drivers/intel/cavs/ssp.c:858 	ssp_stop(), TX stop[0m
[0m    0      2      unknown          36443868.072917         6.041667 src/schedule/timer_domain.c:64  	timer_domain_unregister()[0m
```

jdrzaszcz, 2020-02-24:
I checked it with 1d273c3a9f155624144199e36a14ffafeec345ca commit and don't see this issue after updating piplines prioritys.

jdrzaszcz, 2020-02-26:
Can't reproduse on master branch 1d273c3a9f155624144199e36a14ffafeec345ca commit.

closed 2020-02-26 by jdrzaszcz